aqualinkd/AqualinkD #104: Controlling JXi Heater via RS 8 panel

ncpe20012003, 2020-03-04:
I'm able to connect and control JXi heater to RS8 via 485 interface as long as I connect water temp probe, or 22k ohm resistor, across RS8 water temp input. However, AqualinkD does not recognize the JXi. I can put the RS8 panel in service mode and press the heater button and my JXi activates.

I ran serial logger with AqualinkD shutdown, and the JXi uses address 0x68 on the 485 serial bus that's not recognized. What's needed to make control of the JXi possible from AqualinkD? Also, there is a water temp sensor built into the JXi, on responses to the polls does JXi return the water temp and can this data be used?

timelery, 2020-03-06:
I am also very interested in finding out the answer to this question.

sfeakes, 2020-03-06:
AqualinkD turns the heater on/off/temp through the control panel. So as long as the control panel is configured correctly, it should all work.  Can you turn the heater on/off from the control panel? (when it's not in service mode).

The 0x68 not being recognized on serial logger is not relevant from the perspective of heater control.

timelery, 2020-03-06:
> AqualinkD turns the heater on/off/temp through the control panel. So as long as the control panel is configured correctly, it should all work. Can you turn the heater on/off from the control panel? [...]

I do not have aRS controller. I am trying to communicate with the Jxi directly from the PC

ncpe20012003, 2020-03-07:
With the RS panel in Auto mode, not service mode, AqualinkD does not
activate the heater. For testing purposes, it does toggle the filter pump.

sfeakes, 2020-03-07:
It sounds like you need to configure the control panel for the heater.  If you don’t have a RS keypad to do that from, you can use AqualinkD in keypad simulation mode to do that.

ncpe20012003, 2020-03-08:
> It sounds like you need to configure the control panel for the heater. If you don’t have a RS keypad to do that from, you can use AqualinkD in keypad simulation mode to do that.

If the RS Panel is working, what config changes are needed on the panel?

sfeakes, 2020-03-09:
Probably just the heater button.  AqualinkD tells the RS panel to “turn on the heater”, so if that’s not configured correctly it won’t work.

peril99, 2020-08-03:
I just got thru setting these up - there isn't anything special you need to do with the JXi except to make sure that is is configured correctly into the panel -I can send you photos of my config if you need it, but it's documented in the (lengthy) RS setup guide.

https://www.troublefreepool.com/wiki/index.php?title=Jandy_Aqualink_RS

sfeakes, 2020-08-04:
Nothing special, as long as the control panel is controlling the heater, AqualinkD should be fine.