aqualinkd/AqualinkD #105: Communicating directly with Pump - Message Timing and Sequence Issues

timelery, 2020-03-13:
@sfeakes I was hoping to ask for your insight base don all the work you have done here.

I am struggling to figure out why I get inconsistent results when powering off and on my Intelliflow pump. After spending a few days looking at your code I have the following. It works but not consistently and it makes me think my timing or sequence if off or more importantly I am missing a packet that needs to be included.

Do you have any ideas?

To Power On the Pump into Program 3
Packet 1 (Set to Remote) 165,0,96,33,4,1,255
Packet 2 (Set Program 3) 165,0,96,33,1,4,3,33,0,24 -- 175ms after previous message
Packet 3 (Turn Pump On) 1165,0,96,33,6,1,10 -- 175ms after previous message

To Power Off the Pump from Program 3
165,0,96,33,4,1,255]
165,0,96,33,1,4,3,33,0,0 -- 175ms after previous message
165,0,96,33,6,1,4 -- 175ms after previous message

sfeakes, 2020-03-13:
Can you explain a little more on what you have connected & available, and what you're trying to do?
But few pointers that may help 

- Best way is to simply use the serial logger (from this repo) and log what happens when you control the equipment using jandy/pentair controllers.

- You should really be looking at the raw hex from the bus, not the decimal conversion that your posting, that way you can be sure of the actual start / stop packet(s).

- (reason for above) Pentair inteliflow and Jandy protocols are different and will collide / overlap on the RS485 bus, so you can get a Jandy byte sequence in the middle (or overlapping) a Pentair byte sequence so making things very difficult to decipher.

- If you don't have some form of controller to monitor, Jandy has some very old PC based simulators that can control some equipment using a USB2RS458 adapter.  I've decoded (some) equipment I don't actually own by having an old PC with RS485 adapter running Jandy simulator software controlling equipment while also having a Raspberry PI with RS485 adapter running the serial logger within this repo.  The Jandy simulators are really crap and 1/2 don't work well, but it's pointed me in the right direction at times without having to buy equipment.

- If you go through closed threads here and aqualinkd forum and start picking out log files people have posted, you'll probably find everything (or a lot) you need to know on the pump protocol from other peoples systems.  But I have not spent the time to fully documented the Pump protocols yet, I've really only looked / deciphered the status messages.

timelery, 2020-03-13:
@sfeakes  Thank you for the insight. I will spend some time looking at the issue list.

Lastly,

If I have a schedule set on the pump locally and I control it remotely by turning it off or setting a new speed and turning it on I am unable to set the pump back to local schedule mode and have to physically press it on the pump to switch the pump back to local schedule mode.

Are you aware of a command I can send the pump to set it back to local schedule mode?

Thanks,
Tim

sfeakes, 2020-03-13:
I believe the pump works the same way as a SWG, where it will go back to local control when it doesn't get any probe messages. But I do remember seeing something in someones logs that looked suspiciously like a message to set it back to local mode.  I'll see if I can find it.

sfeakes, 2020-03-13:
If you pull the logs from the below thread there are some Pentair commands that I decoded as "go back to local control". Problem is I didn't document what they were.
https://github.com/sfeakes/AqualinkD/issues/95

I *think* one of these are the the go back to local control (bottom two), but you should go over the logs in that thread.
```
Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x60|0x10|0x04|0x01|0xff|0x02|0x19|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x60|0x10|0x04|0x01|0xff|0x02|0x19|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x10|0x60|0x04|0x01|0xff|0x02|0x19|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x10|0x60|0x04|0x01|0xff|0x02|0x19|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x60|0x10|0x06|0x01|0x0a|0x01|0x26|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x60|0x10|0x06|0x01|0x0a|0x01|0x26|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x10|0x60|0x06|0x01|0x0a|0x01|0x26|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x10|0x60|0x06|0x01|0x0a|0x01|0x26|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x60|0x10|0x01|0x04|0x02|0xc4|0x06|0xa4|0x02|0x8a|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x60|0x10|0x01|0x04|0x02|0xc4|0x06|0xa4|0x02|0x8a|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x10|0x60|0x01|0x02|0x06|0xa4|0x01|0xc2|
 Pentair Packet | HEX: 0xff|0x00|0xff|0xa5|0x00|0x10|0x60|0x01|0x02|0x06|0xa4|0x01|0xc2|

```

closed 2020-07-18 by sfeakes