aqualinkd/AqualinkD #107: VSP RPM only shown briefly (like a FOUC)

LogoiLab, 2020-06-13:
Please refer to this video I recorded:
[https://youtu.be/NurcqK32Ayw](https://youtu.be/NurcqK32Ayw)

You can see a brief flash of the RPM before it gets overwritten by "ON"

I just recently updated from a year old version to the latest master branch.

sfeakes, 2020-06-13:
Interesting, that's the status vs device response from the UI.  Can you set logging to debug and post the output from the log please.

LogoiLab, 2020-06-14:
Here is the log:
```
Jun-14-20 10:48:28 AM Notice: Aqualink Daemon v2.0.1
Jun-14-20 10:48:28 AM Notice: Config log_level         = 7
Jun-14-20 10:48:28 AM Notice: Config socket_port       = 80
Jun-14-20 10:48:28 AM Notice: Config serial_port       = /dev/ttyUSB0
Jun-14-20 10:48:28 AM Notice: Config web_directory     = /var/www/aqualinkd/
Jun-14-20 10:48:28 AM Notice: Config device_id         = 0x0a
Jun-14-20 10:48:28 AM Notice: Config extra_device_id   = 0x00
Jun-14-20 10:48:28 AM Notice: Config extra_device_prog = NO
Jun-14-20 10:48:28 AM Notice: Config read_all_devices  = YES
Jun-14-20 10:48:28 AM Notice: Config use_aux_labels    = NO
Jun-14-20 10:48:28 AM Notice: Config override frz prot = NO
Jun-14-20 10:48:28 AM Notice: Config mqtt_server       = (null)
Jun-14-20 10:48:28 AM Notice: Config mqtt_dz_sub_topic = (null)
Jun-14-20 10:48:28 AM Notice: Config mqtt_dz_pub_topic = (null)
Jun-14-20 10:48:28 AM Notice: Config mqtt_aq_topic     = (null)
Jun-14-20 10:48:28 AM Notice: Config mqtt_user         = (null)
Jun-14-20 10:48:28 AM Notice: Config mqtt_passwd       = (null)
Jun-14-20 10:48:28 AM Notice: Config mqtt_ID           = aqualinkd_a0a8cd63
Jun-14-20 10:48:28 AM Notice: Config idx water temp    = -999
Jun-14-20 10:48:28 AM Notice: Config idx pool temp     = -999
Jun-14-20 10:48:28 AM Notice: Config idx spa temp      = -999
Jun-14-20 10:48:28 AM Notice: Config idx SWG Percent   = 0
Jun-14-20 10:48:28 AM Notice: Config idx SWG PPM       = 0
Jun-14-20 10:48:28 AM Notice: Config PDA Mode          = NO
Jun-14-20 10:48:28 AM Notice: Config PDA Sleep Mode    = NO
Jun-14-20 10:48:28 AM Notice: Config force SWG         = NO
Jun-14-20 10:48:28 AM Notice: Config deamonize         = YES
Jun-14-20 10:48:28 AM Notice: Config log_file          = /var/log/aqualinkd.log
Jun-14-20 10:48:28 AM Notice: Config light_pgm_mode    = 0.00
Jun-14-20 10:48:28 AM Notice: Debug RS485 protocol     = NO
Jun-14-20 10:48:28 AM Notice: Read Pentair Packets     = YES
Jun-14-20 10:48:28 AM Notice: Display warnings in web  = NO
Jun-14-20 10:48:28 AM Notice: Ignore SWG 0 msg count   = 20
Jun-14-20 10:48:28 AM Notice: Config BTN Filter_Pump   = label Filter Pump     | VSP ID 0x60 | PMP ID 1 |
Jun-14-20 10:48:28 AM Notice: Config BTN Spa_Mode      = label NONE            | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_1         = label Cleaner         | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_2         = label Waterfall       | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_3         = label Pool Light      | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_4         = label Deck Light      | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_5         = label NONE            | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_6         = label Garden Light    | 
Jun-14-20 10:48:28 AM Notice: Config BTN Aux_7         = label NONE            | 
Jun-14-20 10:48:28 AM Notice: Config BTN Pool_Heater   = label Pool Heater     | 
Jun-14-20 10:48:28 AM Notice: Config BTN Spa_Heater    = label Spa Heater      | 
Jun-14-20 10:48:28 AM Notice: Config BTN Solar_Heater  = label Solar Heater    | 
Jun-14-20 10:48:28 AM Debug:  process_id of child process 19364 
Jun-14-20 10:48:28 AM Notice: Starting web server on port 80
Jun-14-20 10:48:28 AM Notice: Starting MQTT client to (null)
Jun-14-20 10:48:28 AM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jun-14-20 10:48:28 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:28 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:28 AM Debug:  RS received packet of type Message length 24
Jun-14-20 10:48:28 AM Info:   RS Message :- 'Air Temp 62 F'
Jun-14-20 10:48:28 AM Debug:  Getting temp units from message 'Air Temp 62 F', looking at 'F'
Jun-14-20 10:48:28 AM Info:   Temp Units set to 0 (F=0, C=1, Unknown=2)
Jun-14-20 10:48:28 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:28 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:29 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:30 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:30 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:30 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:48:30 AM Debug:  RS received packet of type Status length 12
[... 1006 lines cut ...]
Jun-14-20 10:49:31 AM Debug:  RS received packet of type Message length 24
Jun-14-20 10:49:31 AM Info:   RS Message :- '06/14/20 Sun'
Jun-14-20 10:49:31 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:31 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:32 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:32 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:32 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:32 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:32 AM Debug:  RS received packet of type Status length 12
[... cut ...]
Jun-14-20 10:49:33 AM Info:   RS Message :- '10:48 AM'
Jun-14-20 10:49:33 AM Debug:  time not checked, will check in 3562 secondsJun-14-20 10:49:33 AM Debug:  RS time is accurate '10:48 AM 06/14/20 Sun'
Jun-14-20 10:49:33 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:33 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:34 AM Debug:  RS received packet of type Status length 12
Jun-14-20 10:49:34 AM Notice: Stopping!Jun-14-20 10:49:34 AM Debug:  -- Websocket left
Jun-14-20 10:49:34 AM Notice: Exit!
```

LogoiLab, 2020-06-14:
I also think the Pool/Air Temperature tiles should be setTileOn by default. The transparency makes them hard to read.

sfeakes, 2020-06-14:
```Config extra_device_id   = 0x00```

That's a problem.  please read the wiki for V1 to V2 updating.
V1 depended just on the RS485 pump data protocol as all it could do is read VSP data. V2 can read & write and therefor needs another device ID set to control the pumps through the control panel.    I do need to fix this though, I didn't test using V2 to just read VSP data.

> I also think the Pool/Air Temperature tiles should be setTileOn by default. The transparency makes them hard to read. (I can always open a new issue for this, or bring in a pull request. The design of this application is quite unique I can't say I've ever worked on something quite like this before so probably best for me not to break things.)

The setTileOn is due to the Apple HomeKit UI design I copied for AqualinkD.  I kind-a agree with you, but if you also have homebridge-aqualinkd installed to control through HomeKit, then the two UI's are almost identical. I'll have to think about this request for a bit as it'll have an interesting knock-on effect on some of the other devices.

LogoiLab, 2020-06-15:
Thanks, That fixed it.

sfeakes, 2020-06-15:
Glad that worked.  I've also made the update to have the sensor tiles look like they are on. I'll push that when I've finished the next update.

LogoiLab, 2020-06-17:
Cool, I assume this issue can be closed now.

sfeakes, 2020-06-18:
Yup. Well close this.  Look for update at weekend

sfeakes, 2020-06-21:
I posted v2.1.0 that has an option to highlight value tiles.   You'll need to install v2.1.0 and manually edit the web config as the install scripts won't overwrite configuration files.

add the below to `/var/www/aqualinkd/config.js`
var turn_on_sensortiles = true

Reference below if you're not sure where to put it.
https://github.com/sfeakes/AqualinkD/blob/master/web/config.js

LogoiLab, 2020-08-06:
looks good, thanks again. Going to close this now.

closed 2020-08-06 by LogoiLab