aqualinkd/AqualinkD #108: Getting started

Barrygordon, 2020-06-15:
Hi,  I am just starting with AqualinkD. I believe it installed correctly as it allows me to start and stop the daemon. If I try and use the serial logger I get:

./release/serial_logger: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.23' not found (required by ./release/serial_logger).

I cannot match up modules with what the documentation says.  e.g. there is no AqualinkD in /etc/, but there is a /etc/default/aqualinkd.  In /home/pi there is a directory named software (/home/pi/software) start but it appears to be empty.

I am a little confused.

sfeakes, 2020-06-15:
If '/home/pi/software' is empty what directory are you in when you run './release/serial_logger' ?

IE, Serial logger should be downloaded to '/home/pi/software/AqualinkD/release' if you are logged in as the Pi user when you followed either of the installation methods.

Which installation method did you use?
https://github.com/sfeakes/AqualinkD/wiki#simple-install-trust-everything
or
https://github.com/sfeakes/AqualinkD/wiki#Install

BTW, there is no AqualinkD in /etc/, it's `/etc/aqualinkd.conf`.  (I'll have to search the documentation as there is probably an error).

If serial_logger is giving you a GLIBC error, then AqualinkD probably is as well.  It'll probably be easier to make the binaries on your system rather than any other options.  So in the second link above go to the section about  "**There is a chance the pre-compiled binary will run.**" and use that test, it'll probably fail then continue to the section "**If it faild to run with some form of error on the binary file you'll need to build a new one.**"

Barrygordon, 2020-06-18:
I worked it all out (almost).  I have attached the my configuration, and a screen snap of the web Browser presentation. 

 

If I turn on the SPA Blower it turns on but a message comes up in the top status bar that it will turn off after 30 minutes.  That message should normally be associated with a fill line.

 

The button for the fill line does not work. The fill line is triggered from the solar heater device. Turning on the solar heater should turn on the fill line.  I will probably just move the trigger to aux 4 or 5 where I am sure it will work.  I would like to see the 30 minute message for the fill line.

 

I am planning to integrate the AqualinkD with my home automation system which has iPads wall mounted in each room which I custom programmed. Is there a way to just use TCP  (Sockets). The Autelis people added code to their system to send a UDP Broadcast message whenever something in the Aqualink changed.  Nice, but not necessary. The UDP broadcast updated all the ipads, but really the only iPad that NEEDS to be updated is the one looking at the web page.

 

I assume it (AqualinkD daemon) will auto start on a reboot of the RPi. 

 

Is there a guide to the WEB commands and their format? (I think I saw it in the wiki).

 

All in all a very nice piece of work. 

 

https://github.com/sfeakes/AqualinkD/wiki#simple-install-trust-everything
https://github.com/sfeakes/AqualinkD/wiki#Install

Barrygordon, 2020-06-18:
I have started to interface AqualinkD to my home automation system.  Did you write the web interface as a true server? If so, how many simultaneous connections will it maintain?

 

TIA, 

 

Barry

 

https://github.com/sfeakes/AqualinkD/wiki#simple-install-trust-everything
https://github.com/sfeakes/AqualinkD/wiki#Install

sfeakes, 2020-06-18:
None of your images are loading, maybe more will make sense once they do. But I can't help with the questions as I'm not really understanding what you mean.

Broadcasting over UDP is not a great solution in this day n age, if you want realtime updated I urge you to use MQTT, all home hubs support that and it's a far better option.
Other than that, you can use web sockets for realtime updates, but that's not documented as it's not an interface I want to public version as I change that frequently since the webUI uses it.
All the different API's & MQTT are all documented in the wiki.   There will be a new API V2 in the next few weeks, so if you want to use HTTP, you may want to wait.

The web UI interface is "mobile" app enabled, so you can just load in web browser, hit "save to desktop" on any mobile device and it'll create an app for you.

As for concurrent users on the web UI, it's stateless and uses web sockets once loaded, so limitations are not by "concurrent users" but more by CPU.

Barrygordon, 2020-06-19:
Okay.  I believe I can use the system as is. 

 

I  just ran the system with three browsers connected. I turned on the waterfall it came on and all the systems showed it as on.  From a different browser I turned it off and all the browsers showed it as off. 

From this test I assume you handle the WEB Interface as a true server and send all state changes to all connected browsers.

 

I understand your API for HTTP and I can use that with no trouble.

 

My only concern at this time is how I will handle the pool fill line. The pool fill line is a solenoid controlled (24V DC) fill line

I used the Solar heater relay as my original indoor panel (RS4) only had buttons for Pump, SPA, Blower, Light, Waterfall and three buttons for Heaters (Pool, SPA, Solar).

I forget How I made it only open the fill line for 30 minutes.  

 

I did see while playing around a message that the waterfall will turn off in 30 minutes, but I don’t remember the configuration that did that. 

It might have been the original configuration. I will verify that tomorrow

 

With the current configuration if I turn on the pool fill line it turns off after about 1 second.

 

How does one set the time for the control panel. The AqualinkD shows time as about 40 minutes behind real time, and the date as “1/31/00 mon”  

 

Could you briefly explain the “Extra_Device_ID =0x00”, and the “Extra_Device_Prog=no”  I see in the log file

 

TIA,

 

Barry

******************************************************************************************************

Here is a copy of my current configuration with all the lines that have been commented out removed:

 

./# aqualinkd.conf

#

web_directory=/var/www/aqualinkd/              

log_file=/var/log/aqualinkd.log

log_level=ERROR

display_warnings_in_web=true

socket_port=80

serial_port=/dev/ttyUSB0

override_freeze_protect = no

convert_mqtt_temp_to_c = yes 

convert_dz_temp_to_c = yes

report_zero_spa_temp = no

report_zero_pool_temp = no

device_id=0x0a

read_all_devices = yes

read_pentair_packets = no

force_SWG = no

light_programming_mode=0

light_programming_initial_on=15

light_programming_initial_off=12

use_panel_aux_labels=no

button_01_label=Filter Pump

button_02_label=Spa Mode

button_03_label=Spa Blower

button_04_label=Pool Light

button_05_label=Waterfall

button_06_label=Pool Fill

button_07_label=NONE

button_08_label=NONE

button_09_label=NONE

button_10_label=NONE

button_11_label=NONE

button_12_label=NONE

 

 

 

Here is a copy of the front of the log file from var (NOT from the above configuration)

Jun-18-20 14:47:59 PM Notice: Aqualink Daemon v2.0.1

Jun-18-20 14:47:59 PM Notice: Config log_level         = 5

Jun-18-20 14:47:59 PM Notice: Config socket_port       = 80

Jun-18-20 14:47:59 PM Notice: Config serial_port       = /dev/ttyUSB0

Jun-18-20 14:47:59 PM Notice: Config web_directory     = /var/www/aqualinkd/

Jun-18-20 14:47:59 PM Notice: Config device_id         = 0x0a

Jun-18-20 14:47:59 PM Notice: Config extra_device_id   = 0x00

Jun-18-20 14:47:59 PM Notice: Config extra_device_prog = NO

Jun-18-20 14:47:59 PM Notice: Config read_all_devices  = YES

Jun-18-20 14:47:59 PM Notice: Config use_aux_labels    = NO

Jun-18-20 14:47:59 PM Notice: Config override frz prot = NO

Jun-18-20 14:47:59 PM Notice: Config mqtt_server       = (null)

Jun-18-20 14:47:59 PM Notice: Config mqtt_dz_sub_topic = (null)

Jun-18-20 14:47:59 PM Notice: Config mqtt_dz_pub_topic = (null)

Jun-18-20 14:47:59 PM Notice: Config mqtt_aq_topic     = (null)

Jun-18-20 14:47:59 PM Notice: Config mqtt_user         = (null)

Jun-18-20 14:47:59 PM Notice: Config mqtt_passwd       = (null)

Jun-18-20 14:47:59 PM Notice: Config mqtt_ID           = aqualinkd_b827ebca

Jun-18-20 14:47:59 PM Notice: Config idx water temp    = -999

Jun-18-20 14:47:59 PM Notice: Config idx pool temp     = -999

Jun-18-20 14:47:59 PM Notice: Config idx spa temp      = -999

Jun-18-20 14:47:59 PM Notice: Config idx SWG Percent   = 0

Jun-18-20 14:47:59 PM Notice: Config idx SWG PPM       = 0

Jun-18-20 14:47:59 PM Notice: Config PDA Mode          = NO

Jun-18-20 14:47:59 PM Notice: Config PDA Sleep Mode    = NO

Jun-18-20 14:47:59 PM Notice: Config force SWG         = NO

Jun-18-20 14:47:59 PM Notice: Config deamonize         = NO

Jun-18-20 14:47:59 PM Notice: Config log_file          = /var/log/aqualinkd.log

Jun-18-20 14:47:59 PM Notice: Config light_pgm_mode    = 0.00

Jun-18-20 14:47:59 PM Notice: Debug RS485 protocol     = NO

Jun-18-20 14:47:59 PM Notice: Read Pentair Packets     = NO

Jun-18-20 14:47:59 PM Notice: Display warnings in web  = NO

Jun-18-20 14:47:59 PM Notice: Ignore SWG 0 msg count   = 20

Jun-18-20 14:47:59 PM Notice: Config BTN Filter_Pump   = label Filter Pump     | 

Jun-18-20 14:47:59 PM Notice: Config BTN Spa_Mode      = label Spa Mode        | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_1         = label SPA Blower      | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_2         = label Pool Light      | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_3         = label Waterfall       | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_4         = label Pool Fill       | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_5         = label Aux_5           | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_6         = label Aux_6           | 

Jun-18-20 14:47:59 PM Notice: Config BTN Aux_7         = label Aux_7           | 

Jun-18-20 14:47:59 PM Notice: Config BTN Pool_Heater   = label Pool_Heater     | 

Jun-18-20 14:47:59 PM Notice: Config BTN Spa_Heater    = label Spa_Heater      | 

Jun-18-20 14:47:59 PM Notice: Config BTN Solar_Heater  = label Solar_Heater    | 

Jun-18-20 14:47:59 PM Notice: Starting web server on port 80

Jun-18-20 14:47:59 PM Notice: Starting MQTT client to (null)

Jun-18-20 14:47:59 PM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0

Jun-18-20 14:48:03 PM Notice: AqualinkD set to Service Mode

sfeakes, 2020-06-19:
> Okay. I believe I can use the system as is. I just ran the system with three browsers connected. I turned on the waterfall it came on and all the systems showed it as on. From a different browser I [...]

Now I understand your first comment on web servers.  Traditional Web servers don't actually work like this, they do not send state when it changes, they wait to be polled by the client for information, and sent the state at that point. (So you're limited to how quickly and how many clients are polling the server)
AqualinkD will act just like a traditional web server when you request the page, but to save CPU cycles, after the page loads, it's coded so the web browser opens a web socked back to the server and the server can then send updates over the socket as they happen in real time, this means it doesn't have huge overhead of being poled by the client that a traditional web server has, and events are posted as they happen, not limited to poll cycle time.

> I understand your API for HTTP and I can use that with no trouble. My only concern at this time is how I will handle the pool fill line. The pool fill line is a solenoid controlled (24V DC) fill [...]

This will be specific to your control panel version, you'll probably need to program that at the panel, or use a home automation hub with the AqualinkD API's to set a timer.

> The AqualinkD shows time as about 40 minutes behind real time, and the date as “1/31/00 mon” 

AqualinkD uses system time and will set the Control Panle to that since most panels a terrible at keeping time.  So look at your system time / timezone that's probably what's wrong.

> Could you briefly explain the “Extra_Device_ID =0x00”, and the “Extra_Device_Prog=no” I see in the log file TIA, Barry 

They are described here
https://github.com/sfeakes/AqualinkD/wiki#AqualinkD
and here
https://github.com/sfeakes/AqualinkD/wiki#Version_2

> ****************************************************************************************************** Here is a copy of my current configuration with all the lines that have been commented out removed: .
button_12_label=NONE
Config BTN Solar_Heater  = label Solar_Heater

I think these two lines *may* be an issue with how you've described your "water fill line".  If you use the "solar heater" as the JVA to control your water fill, then you need to use/enable the Solar_Heater button, and you have disabled it.  Also need to edit the config.js for the UI as solar heater is disabled by default.

You also mentioned Autelis previously, out of interest do you have one of those devices & is it still working?

Barrygordon, 2020-06-20:
Thanks for the feedback reply.  At the current time all is working just as I want it (even the fill line) with one remaining issue.  The log file is showing a lot of checksum errors on reads of status.  That may go away with the new control panel.  I would hate to have to run new (better wires) as I am too old to crawl in attics and getting wires from outside my house to the inside is non-trivial. The new panel should be installed next week.

 

I have been researching RS485 (I am an electrical engineer, but have forgotten too much). While RS485 as used by the Aqualink is half duplex, many RS485 to USB converters want a ground reference. Do you know which of the 4 wires (red, black, yellow, green)  if any is ground.  I am currently using the $7.00 converter you suggested which does not use a ground reference

 

I have the Autelis unit which I believe is a brick at this time.  Let me double check it and get back to you. If you want it I will send it to you as “Payment” for AqualinkD and your assistance.

 

I do a lot of work with the RPi but try to stick with Node.js. Many of the devices I use (Philips Pronto line of remotes (now defunct), Command fusion for writing programs for the IOS world) are all programmed in Javascript which I am fairly fluent in. Ergo Node.js was logical for me as the programming language for the RPI.  I can read C and C++, I am just not fluent in it. I use Visual Studio Code on a WIN10 system for my development and test platform. Supports Node.js among other languages.

 

I am aware of how the traditional web servers work and tend to do all my communications coding using web sockets which Node.js handle quite well.  I do not like systems that poll, but rather those rthat are event driven!  When I am developing communicating systems I use Web Sockets. 

 

Again thanks for the AqualinkD and all the assistance

 

https://github.com/sfeakes/AqualinkD/wiki#AqualinkD
https://github.com/sfeakes/AqualinkD/wiki#Version_2

sfeakes, 2020-06-20:
Please check this with a meter, but **_usually_** on Jandy equipment. (power is DC between 8 and 16v)
Ground is Pin4    (Green)
Data - is Pin3.    (black)
Data + is pin2.   (yellow)
Power is Pin1.    (red)

Some people have had issues with Ground reference, some have not, so it seems to be very hit n miss as too many other variables.

I've tested lots of different adapters, I've just started testing the latest FTDI chipset, which seems to work very well on Linux, but drivers can be a PITA on windows.
The one below is the latest that's performed well on my test bench setup (but not tried it against my main equipment yet).  It has the Ground contact that you are after. 
https://www.amazon.com/gp/product/B07SD65BVF/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

If you are using iOS, you may want to look at homebridge and homebridge-aqualinkd, then you can use / access AqualinkD directly from the native Apple HomeKit ecosystem which will also give you easy control outside your home network. homebridge and homebridge-aqualinkd are all Node.js as well, so should be easy for you to modify as needed.

I know Autelis units use the Jandy PC protocol, (copied from the Jandy Serial Adapter 7620). I've decoded and published all the different Jandy control protocols except that one as I've never had access to either Autelis or Jandy Serial Adapter. So do please let me know if you think it's in a usable format, or if you ever come across a Jandy Serial Adapter 7620.

Barrygordon, 2020-06-20:
The Aqualink stopped working.  There is no communications to the control panel.  I am pretty sure it is the control panel that is the problem as it has been flaky with regards to communicating over the RS485 buss.   I guess it just gave up the ghost. It is being replaced next week.  I say there is no communications as the serial logger returns nothing and the test sequence of pulling raw data using sudo –s eval “sty . . . returns nothing. (Both of them  did before). The AqualinkD keeps entering into the log file every 2 – 3 seconds the following:  “<current date/time> Error:  Aqualink daemon looks like serial error, resetting.”

 

I was hoping to do the integration with the AqualinkD this weekend, but oh well, I’ll just write the code.

Regarding the Autelis unit, It is a brick.  If you can make it work it is yours.  If you end up not needing it send it back to me. If you give me a mailing address I will send it to you with its PSU which tested fine. I say it is a brick because I believe it should at least activate the Ethernet link and it doesn’t.

 

I had a 7620 serial interface but don’t know if I sold it or if it is put away somewhere.  I will search for it and if I find I can/will send it to you

 

With regard to the ‘extended_device_ID’  I do not have a variable speed pump and no plans to get one, nor to get a SWG; so I guess I don’t have to worry about it.

 

Looking over the web UI code It looks like there is a lot of stuff for the future.  Is there going to be some way of setting the temperature set points for the pool, SPA and Freeze control? Perhaps using sliders?

 

TIA,

 

Barry

 

https://github.com/sfeakes/AqualinkD/wiki#AqualinkD
https://github.com/sfeakes/AqualinkD/wiki#Version_2

sfeakes, 2020-06-20:
Shoot me an email, sfeakes@gmail.com and we can exchange address.

You can already change setpoints (heaters / freeze / SWG) from the UI along with programming color lights etc, both from the API and the Web UI.   Simply press the tile for a few seconds and an extra menu will popup with options like set points.   If AqualinkD can workout how to change the set points on you panel then the heater tiles will have the temperate in them, (and you can change the setpoint). If they come up as switches, then you can't.  There is a lot to programming the setpoints, so if AqualinkD fails at that, you'll get switches and not thermostats in the UI. If you have a lot of checksum errors that may be why you didn't get thermostats.

the extra_device_id is really for VSP, but there can be a few other advantages to using it depending on how you use the system.  AqualinkD can use the second ID for programming "some" things, which means you're not blocked from turning switches on/off while it's programming like you are with a single ID.  (But really it's not much gain without VSP).

Barrygordon, 2020-06-28:
> Shoot me an email, sfeakes@gmail.com <mailto:sfeakes@gmail.com>  and we can exchange address.

I have everything working except for asynchronous feedback.  I am using sockets (not web sockets) but pure TCP/IP sockets as implemented on an iPad at the basic code level. I notice that whenever I send a command, after the reply returns the AqualinkD closes the connection. My code recognizes this (Socket status =8 , Peer is closing the connection), allows the connection to close and then re-opens it. Why do you close the connection?

 

I had thought you had implemented asynchronous feedback. The only place I use it is with a fill line.    The fill line only runs for 30 minutes and then turns off. I use the asynchronous receipt of the device turning off to see if I want to run it longer (30, 60, 90, 120 minutes).  I can work around it, but I thought you had said Async feedback was in/planned.

 

The Autellis people implemented, at my request,  Async feedback using UDP broadcast messages.  They used a simple protocol I suggested where each message is sent N (for me N was 2) times with the same sequence number embedded in the message. The next message advanced the sequence number by 1. By increasing the repeat count (it was configurable) for a given network, UDP transmission could be made very reliable. 

 

Next week I will be putting up a MQTT broker on an RPi

 

Barry

sfeakes, 2020-06-30:
AqualinkD doesn’t support low level TCP sockets connections, well it does but only to HTTP or Websockets specification. So when you open the socket, if you don’t follow one of those specs, it’ll close the socket as it doesn’t know what to do and wants to free up resources ASAP.  Ultimately there are so options to use/implement either of those, there is no point it writing your own over a TCP socket.  As said previously, websocket is not documented as it does changes between versions and I don’t want people coding to it as that means I need to version control It, which means extra resources consumed. MQTT is the best option for real time updates, as that’s exactly what the protocol it designed for and prittymuch much the defacto standard for this kinda stuff these days.
Async is currently supported.
For your fill line, if you want a notification of when state changes and then act of that change MQTT is the best option as your never miss an update even with a bad TCP/network connection.  But you can also just open a websocket to AqualinkD and you’ll get updates posed over that connection, but if the connection drops and re-opens you’re not guaranteed not to miss an update.

Are you using Objective C or Swift for your iOS development? (Or something else)  If you let me know I’ll see if I can throw up some sample code when I get some time.

sfeakes, 2020-06-30:
Here is a curl example for websocket, (HTTP example is in wiki). You’ll get a JSON reply of complete status, and it’ll keep sending a complete update every time something changes.

curl -i -N -H "Connection: Upgrade" -H "Upgrade: websocket" -H "Host: localhost" -H "Origin: 
http://localhost:80” -H "Sec-WebSocket-Key: SGVsbG8sIHdvcmxkIQ==" -H "Sec-WebSocket-Version: 13" http://localhost:80

Barrygordon, 2020-06-30:
While I use low level sockets, I strictly obey the HTTP protocol so the Aqualink should see “HTTP”.  I use GET and PUT in accordance with the specifications for HTTP.  Now I could have some error but all the other “HTTP”  systems I have worked with have no issue. While the closing of the channel at the end of a transmission is strange, I have coded around it and one cannot tell it is occurring at the user level. As soon as the socket enters the “peer closing” state, I merely reopen the socket and re-send any message that was pending. 

 

As soon as I get some time I do intend to put up a MQTT broker on an RPi.  I have installed the MQTT module on my development iPad and should be testing with it in a week or two. 

 

There is a company called Commandfusion (www.commandfusion.com <http://www.commandfusion.com> ).  The company makes both network automation hardware and software.  Their software provides a Visual Basic like graphics environment coupled with a full Javascript engine for programming an IOS system. It was designed as a graphical user interface (GUI) to any home automation systems.  This was before IOT and voice control.  When I decided to use the iPad (wall mounted) as the user interface system for my home (about 8 years ago, there are 5 of them, one in each major room) I chose their software as the development system.  It consists of a graphics designer and a run time system. The Graphics designer produces an XML output file that drives the run time system.  They started by supporting only IOS, but now have two run time systems, one for IOS, and one for Android. With some simple coding practices the same Javascript code can be used for both an iPad and an IOS based smartphone. I have no experience with their Android implementation.  Obviously the graphics have to change as screens can be quite complex.

 

I tend to constrain my coding to Javascript when possible. At one time I was advising the Philips team with their Pronto remotes which are programmed using a Graphics design tool coupled to a Javascript engine; very similar to what Commandfusion implemented.  They called it Prontoscript, and their remote family was known as the “Pronto’s”.

 

I already put the MQTT code on my iPads, but have not set up the broker yet. I have the RPi ready for it and will do it shortly

Barrygordon, 2020-07-01:
Here is an update on where I am.

 

My iPads are working well using web socket protocol over a TCP/IP socket connection. I had to handle the instance where the aqualinkd closes the connection after every transmission, but that was relatively easy. Since the iPads are normally showing a blank screen (black, or as black as apple allows without turning off the backlight) I only need communications with the aqualinkd when a user selects the “Pool” screen. I then pull full status, and wait for the user to change something or exit the pool screen.  The lack of async feedback is a little bothersome, BUT… You have got me started  with using MQTT.  Thank you, I should have looked at MQTT a long time ago. I will be changing from web sockets over TCP/IP to MQTT as the link between my iPad app and the aqualinkd.

 

I have an MQTT  Broker running on a RPi, and an MQTT  sniffer running on my development (windows) machine. I can clearly see all the messages from the aqualinkd and when I publish according to your MQTT write up in the wiki everything appears to work as it should.  The only issue I see is that the aqualinkd is constantly publishing the following (each with a QOS of 1):

                aqualinkd/swg/enabled: 0

aqualinkd/swg/fullstatus: 255

Since I have no Salt Water generator, How do I stop that traffic?

 

I have attached a copy of my aqualink.conf. 

 

./# aqualinkd.conf
#

# The directory where the web files are stored

# Log to file, comment out if you do not want to log to file
#log_file=/home/pi/software/AqualinkD/release/aqualinkd.log

# The log level. [DEBUG_SERIAL, DEBUG, INFO, NOTICE, WARNING, ERROR]
# Pick the highest level, and all levels below will be sent to syslog.
# your syslog settings may be set to only display messages above a certian level
# in which case make sure you use the log_file settings to capture everything
# you want when debugging
# so, NOTICE also prints WARNING & ERROR
#     DEBUG_SERIAL would print everything possible

#log_level=INFO
Log_level=NOTICE

# Display any ERROR & Warning messages in web interface. 

# The socket port that the daemon listens to
# If you change this from 80, remember to update aqualink.service.avahi

# The serial port the daemon access to read the Aqualink RS8

# If equiptment is in freeze protect mode some commands like pump_off / spa_on are
# ignored.  You can force these to work by setting the below.

# Convert Deg F to Deg C when posting to Domoticz or MQTT.

# default is to use pool water temp as spa water temp when spa is off (and therefor not able to report water temp)
# enable below to report 0 as the spa temp when spa is off.
# This is for MQTT cnnections only, WEB socket and WEB API always report TEMP_UNKNOWN (-999) allowing the consumer to
# decide how to report. 

# default is to not report changes to pool temp when the filter pump is off or in spa mode
# enable below to report 0 as the pool temp when the filter pump is off or when in spa mode.
# This is for MQTT connections only, WEB socket and WEB API always report TEMP_UNKNOWN (-999) allowing the consumer to
# decide how to report.

# mqtt stuff
mqtt_address = 192.168.1.74:1883
mqtt_user = barry    
mqtt_passwd = jillian
mqtt_aq_topic = aqualinkd
#mqtt_dz_pub_topic = domoticz/in
#mqtt_dz_sub_topic = domoticz/out
[... 105 lines cut ...]
#button_06_PDA_label=AUX4

button_07_label=NONE
#button_07_dzidx=43 
#button_07_PDA_label=AUX5

button_08_label=NONE
#button_08_dzidx=NONE 
#button_08_PDA_label=AUX6

button_09_label=NONE
#button_09_dzidx=NONE 
#button_09_PDA_label=AUX7

button_10_label=NONE
#button_10_dzidx=44
#button_10_PDA_label=POOL HEAT

button_11_label=NONE
#button_11_dzidx=56
#button_11_PDA_label=SPA HEAT

button_12_label=Water Fill Line
#button_12_dzidx=NONE 
#button_12_PDA_label=EXTRA AUX

sfeakes, 2020-07-01:
The SWG messages are probably a bug I re-introduced in V2.  There probably isn't much you can do except ignore them for the moment, I'll take a look on that one and get back.

Websockets don't close the connection unless something was sent that AqualinkD didn't like, You can check that with something like Wireshark on any browser in developer mode by looking at the webUI, you can see the connection is constantly open.  So that's probably something on your implementation.

I don't understand the async comment, as you can see you get status of everything over web sockets, and you can also request status for something to be changed.  If by async you mean you want to subscribe to particular event, then as you've seen MQTT is the best options for that, if not can you describe what you're trying to do and I'll see if I can help.  But like the web stuff in previous messages, I think we are using the same terminology to mean different things.

Barrygordon, 2020-07-01:
I have MQTT running now. The only issue I see is the excessive traffic from the SWG. Everything else works fine including asynchronous feedback. I am ignoring the SWQ messages. The only thing I need asynchronous feedback for is the way I fill the pool. If the label “Fill Line” is used then the RS controller will turn of the aux that the label is assigned to after 30 minutes to prevent over filling the pool.  My system sees this feedback and starts the fill line again if the user requested 60, 90 or 120 minutes. It will work fine with MQTT. 

 

I agree with you that I am probably sending something that the aqualinkd does not like, hence the socket shutdown.  That will all be gone when I finish the conversion over to MQTT.  

 

Thanks for all the assistance and the great product.

sfeakes, 2020-07-02:
I can confirm that the multiple SWG messages are a bug. I've inverted some logic in V2, and strangely enough if you force SWG on you won't hit that logic. I'll fix in next update, but for the time being if you change your config from
```
force_SWG = no
```
to
```
force_SWG = yes
```
You'll only get the SWG messages posted to MQTT once.    So bit of a hack, but it'll overcome the issue in this current release.

Barrygordon, 2020-07-02:
My system is working as I want it now.  I have added code to disregard all MQTT messages with a topic of ‘aqualinkd/SWG’. How do you notify users when an update is ready? Or is there someplace I should monitor.

 

TIA,

 

Barry

sfeakes, 2020-07-02:
Updates are posted on the main AqualinkD GitHub page, If you hit the "watch" link at the top you'll get some options of notifications to receive.

Barrygordon, 2020-07-02:
I tried your workaround in the setting of force_SWG with the following results:

 

I always get the following independent of the setting of force_SWG (yes or no)

 

Trace: MQ.process_Data: Has started. feedbackItem=MQTT_Receive, matched String of length =189

PL.processFeedback has started. t=aqualinkd/Alive, v=1

PL.processFeedback has started. t=aqualinkd/Service_Mode, v=0

PL.processFeedback has started. t=aqualinkd/Freeze_Protect, v=0

PL.processFeedback has started. t=aqualinkd/Freeze_Protect/setpoint, v=3.33

PL.processFeedback has started. t=aqualinkd/Freeze_Protect/enabled, v=1

PL.processFeedback has started. t=aqualinkd/Battery, v=1

 

The following comes across ONLY if force_SWG is to NO

 

Trace: MQ.process_Data: Has started. feedbackItem=MQTT_Receive, matched String of length =903

PL.processFeedback has started. t=aqualinkd/Filter_Pump, v=1

PL.processFeedback has started. t=aqualinkd/Filter_Pump/delay, v=0

PL.processFeedback has started. t=aqualinkd/Spa_Mode, v=0

PL.processFeedback has started. t=aqualinkd/Spa_Mode/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_1, v=0

PL.processFeedback has started. t=aqualinkd/Aux_1/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_2, v=0

PL.processFeedback has started. t=aqualinkd/Aux_2/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_2/set, v=0

PL.processFeedback has started. t=aqualinkd/Aux_3, v=0

PL.processFeedback has started. t=aqualinkd/Aux_3/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_3/set, v=0

PL.processFeedback has started. t=aqualinkd/Aux_4, v=0

PL.processFeedback has started. t=aqualinkd/Aux_4/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_5, v=0

PL.processFeedback has started. t=aqualinkd/Aux_5/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_6, v=0

PL.processFeedback has started. t=aqualinkd/Aux_6/delay, v=0

PL.processFeedback has started. t=aqualinkd/Aux_7, v=0

PL.processFeedback has started. t=aqualinkd/Aux_7/delay, v=0

PL.processFeedback has started. t=aqualinkd/Pool_Heater, v=0

PL.processFeedback has started. t=aqualinkd/Pool_Heater/enabled, v=0

PL.processFeedback has started. t=aqualinkd/Pool_Heater/setpoint, v=32.22

PL.processFeedback has started. t=aqualinkd/Spa_Heater, v=0

PL.processFeedback has started. t=aqualinkd/Spa_Heater/enabled, v=0

PL.processFeedback has started. t=aqualinkd/Spa_Heater/setpoint, v=39.44

PL.processFeedback has started. t=aqualinkd/Solar_Heater, v=0

PL.processFeedback has started. t=aqualinkd/Solar_Heater/delay, v=0

PL.processFeedback has started. t=aqualinkd/Temperature/Air, v=34.44

PL.processFeedback has started. t=aqualinkd/Temperature/Pool, v=32.22

 

I am depending on the full status to initialize the state of my process on the iPad. So I will stick with ‘force_SWG=no’ for now.

The traffic is high,  but I have my MQTT client filtering all ‘aqualinkd/SWG’ topics fairly early so it is mostly just network overhead.

sfeakes, 2020-07-02:
AqualinkD can take up to 30~45 seconds to fully initialize and get all information from the control panel. You probably just didn't leave it running long enough with the force_swg=yes option.  You'll get identical information from both options, and you'll still get one set of SWG messages posted to MQTT, but you shouldn't get repeat messages every few seconds with force_swg = yes.

Barrygordon, 2020-07-23:
Has this ‘Bug’ been cleared in the new version (2.2.0) of AqualinkD posted on the GitHub?

sfeakes, 2020-07-24:
Yes, the SWG events constantly posting to MQTT should be fixed.

Barrygordon, 2020-07-25:
I performed the update to the ‘latest version’ For some reason it did not seem to take.  I have attached a screen snapshot from my browser,  A copy of the putty session where I tried to execute aqualinkd, and a copy of the aqualinkd.Conf  from/etc/.

 

The copy of AqualinkD in the folder /home/pi/software/AqualinkD/release  is 384 KB, The older aqualinkd’s were 305 KB or 314 KB. When I monitor what is in stalled I still get the flooding of the SWG MQTT stuff

 

I have also attached the start and stop scripts which I use in putty to start and stop the aqualinkd process.

 

Perhaps I should start with a clean install after running the script to remove everything.

 

Any advice and assistance greatly appreciated.

 

login as: root
root@192.168.1.81's password:
Linux aqualinkD 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jul 25 00:03:51 2020 from 192.168.1.202

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

root@aqualinkD:~# cd /home/pi/software/AqualinkD/release
root@aqualinkD:/home/pi/software/AqualinkD/release# ./stop
-bash: ./stop: Permission denied
root@aqualinkD:/home/pi/software/AqualinkD/release# systemctl stop aqualinkd
root@aqualinkD:/home/pi/software/AqualinkD/release# systemctl status aqualinkd
â aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset
   Active: failed (Result: exit-code) since Sat 2020-07-25 12:56:46 EDT; 18s ago
  Process: 1309 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/
 Main PID: 1311 (code=exited, status=1/FAILURE)

Jul 24 18:34:50 aqualinkD aqualinkd[1311]: Could not select TEMP SET menu
Jul 24 18:34:51 aqualinkD aqualinkd[1311]: Programming: Get Freeze proctect
Jul 24 18:35:23 aqualinkD aqualinkd[1311]: Could not select REVIEW menu
Jul 25 12:56:45 aqualinkD aqualinkd[1311]: Stopping!
Jul 25 12:56:45 aqualinkD systemd[1]: Stopping Aqualink RS daemon...
Jul 25 12:56:46 aqualinkD aqualinkd[1311]: MQTT Connection closed
Jul 25 12:56:46 aqualinkD aqualinkd[1311]: Exit!
Jul 25 12:56:46 aqualinkD systemd[1]: aqualinkd.service: Main process exited, co
Jul 25 12:56:46 aqualinkD systemd[1]: aqualinkd.service: Failed with result 'exi
Jul 25 12:56:46 aqualinkD systemd[1]: Stopped Aqualink RS daemon.
lines 1-16/16 (END)
root@aqualinkD:/home/pi/software/AqualinkD/release# ./start
-bash: ./start: Permission denied
root@aqualinkD:/home/pi/software/AqualinkD/release# systemctl start aqualinkd
root@aqualinkD:/home/pi/software/AqualinkD/release# ./AqualinkD -v
-bash: ./AqualinkD: No such file or directory
root@aqualinkD:/home/pi/software/AqualinkD/release# aqualinkd -v
Error:  Unknown config parameter 'panel_type = RS-8 Combo'
Error:  Unknown config parameter 'device_id=0x0'
Error:  Unknown config parameter 'keep_paneltime_synced = yes'
root@aqualinkD:/home/pi/software/AqualinkD/release# systemctl status aqualinkd
â aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset
   Active: active (running) since Sat 2020-07-25 12:58:02 EDT; 4min 7s ago
[... 42 lines cut ...]
Error:  Unknown config parameter 'panel_type = RS-8 Combo'
Error:  Unknown config parameter 'device_id=0x0'
Error:  Unknown config parameter 'keep_paneltime_synced = yes'
root@aqualinkD:/home/pi/software/AqualinkD/release# systemctl status aqualinkd
â aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-07-25 12:58:02 EDT; 29min ago
  Process: 8258 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
 Main PID: 8259 (aqualinkd)
    Tasks: 1 (limit: 2200)
   Memory: 400.0K
   CGroup: /system.slice/aqualinkd.service
           ââ8259 /usr/local/bin/aqualinkd -c /etc/aqualinkd.conf

Jul 25 12:58:02 aqualinkD aqualinkd[8258]: Config BTN Spa_Heater    = label Spa Heater      |
Jul 25 12:58:02 aqualinkD aqualinkd[8258]: Config BTN Solar_Heater  = label Solar Heater    |
Jul 25 12:58:02 aqualinkD aqualinkd[8259]: Starting web server on port 80
Jul 25 12:58:02 aqualinkD aqualinkd[8259]: Starting MQTT client to 192.168.1.74:1883
Jul 25 12:58:02 aqualinkD aqualinkd[8259]: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jul 25 12:58:03 aqualinkD aqualinkd[8259]: Control Panel B0029221 REV T.2
Jul 25 12:58:04 aqualinkD aqualinkd[8259]: Programming: Get Heater setpoints
Jul 25 12:58:31 aqualinkD aqualinkd[8259]: Could not select REVIEW menu
Jul 25 12:58:32 aqualinkD aqualinkd[8259]: Programming: Get Freeze proctect
Jul 25 12:59:04 aqualinkD aqualinkd[8259]: Could not select REVIEW menu
root@aqualinkD:/home/pi/software/AqualinkD/release#

sfeakes, 2020-07-26:
There are no screen shots or scripts, looks like you might be using email to reply, use githubs web interface for attachments as I don’t thing email works. https://github.com/sfeakes/AqualinkD/issues/108

Since I can’t see anything, I can’t really help.   But few things to check, make sure you only have one aqualinkd running, check the version (written on startup to log/stdout), make sure you updated aqualinkd.conf to the info in the wiki.   Usually aqualinkd doesn’t need config changes between releases, but the last two I’ve had to make large changes.

You should only need a clean install I’d the install.sh script failed / gave warnings.  The install is really basic for upgrade, one binary placed in /usr/bin/aqualinkd and web interface files in /var/www/aqualinkd.   No other files are touched during install Ie /etc/aqualinkd.conf or any of the systemd startup scripts/ configs.

Barrygordon, 2020-07-26:
Okay,  I did not see this before.  There is nothing related to aqualinkd in /usr/bin.  

my previous message (sent via email)

I performed the update to the ‘latest version’ For some reason it did not seem to take.  I have attached a screen snapshot from my browser,  A copy of the putty session where I tried to execute aqualinkd, and a copy of the aqualinkd.Conf  from/etc/.

The copy of AqualinkD in the folder /home/pi/software/AqualinkD/release  is 384 KB, The older aqualinkd’s were 305 KB or 314 KB. When I monitor what is installed I still get the flooding of the SWG MQTT stuff

I have also attached the start and stop scripts which I use in putty to start and stop the aqualinkd process.

Perhaps I should start with a clean install after running the script to remove everything.

Any advice and assistance greatly appreciated.

[PUTTY.TXT](https://github.com/sfeakes/AqualinkD/files/4978871/PUTTY.TXT)

[aqualinkd.conf.txt](https://github.com/sfeakes/AqualinkD/files/4978875/aqualinkd.conf.txt)
[image]
[start.txt](https://github.com/sfeakes/AqualinkD/files/4978885/start.txt)
[stop.txt](https://github.com/sfeakes/AqualinkD/files/4978886/stop.txt)

Barrygordon, 2020-07-26:
OOPs forgot to ask. Would it be possible in the web interface to show the version of aqualinkd just after the word AqualinkD on the top left?  e.g. AqualinkD v2.2

Barrygordon, 2020-07-26:
I am definitely still running 2.0.1!
The web command:
http://192.168.1.81/?command=status
returns:
{"type": "status",
	"status":"Ready",
	"version":"6522 REV I",
	"aqualinkd_version":"2.0.1",
	"date":"06/19/20 FRI","time":"4:25PM",
	"pool_htr_set_pnt":"80",
	"spa_htr_set_pnt":"101",
	"frz_protect_set_pnt":"38",
	"air_temp":"87","pool_temp":"85",
	"spa_temp":" ",
	"temp_units":"f",
	"battery":"ok",
	"leds":{
		"Filter_Pump": "on",
		"Spa_Mode": "off",
		"Aux_1": "off",
		"Aux_2": "off",
		"Aux_3": "off",
		"Aux_4": "off",
		"Aux_5": "off",
		"Aux_6": "off",
		"Aux_7": "off",
		"Pool_Heater": "off",
		"Spa_Heater": "off",
		"Solar_Heater": "off",
		"Freeze_Protect": "enabled"
		}
}

sfeakes, 2020-07-26:
If you click “AqualinkD” in the top left of the web UI, it will show AqualinkD version along with control panel version.

Sorry, I meant /use/sbin, for AqualinkD binary.

You should be running 2.2.0x. (x is a,c, or c) depending on when you pulled it from github.

Your log looks like there may be some errors getting setpoint information, can you post a complete log once you get 2.2.0x 
FYI
You should also uncomment use extended programming from your config.

Barrygordon, 2020-07-26:
I have it all squared away.  I am now running version 2.2.0b according to a pull of status using http.

The only issue I seem to have is that when I start aqualinkd it comes up with the waterfall on.  The waterfall is on AUX_3. 
The filter pump also comes up as on, but it should since the programming is to have the pump on between 8AM and 8 PM.  

I will restart the aqualinkd after 8 PM and see what comes up as on.  

Great piece of work.  Most of my problems were due to my lack of Linux experience

Barrygordon, 2020-07-27:
I am running version 2.2.0b.  

 

I restarted the aqualinkd after 8PM and the filter pump comes up as off as it should,  but the waterfall (aux 3) still comes up as on.  Not a big issue since I normally rarely restart the aqualinkd.  All electronics in the house is on a UPS which is backed up by a whole house generator so a shutdown/restart is very rare. 

 

Regarding ‘use extended programming’, there are two config parameters:

1  #extended_device_id=0x31

2  #extended_device_id_programming = yes

 

Should I uncomment both of them or just the second one.

 

There is no longer a file “/var/log/aqualinkd.log” Has it been moved?

 

There is no copy of aqualinkd in /usr/sbin

 

The SWG problem appears to be fixed.  Thanks

sfeakes, 2020-07-27:
> I am running version 2.2.0b. I restarted the aqualinkd after 8PM and the filter pump comes up as off as it should, but the waterfall (aux 3) still comes up as on. Not a big issue since I normally [...]

Re:- The AUX_3 / Waterfall being on.   What happens when you select that button in the web UI?   Does it turn the correct device on/off?   If not, then check the button part of aqualinkd config

Re:- Extended_device_id.  You need to run the serial logging utility to get the ID's for you system.  (Make sure you run it without AqualinkD running).

Re:- /usr/local/bin/aqualinkd.     That's the binary location. (3rd time lucky..;>)

Re:-  Image of web ui (and logs), it looks like AqualinkD is failing in getting heater setpoints, (Logs have error and UI shows switches not thermostats).  Something needs to be fixed there.  Do you have the right panel type in your config?  Serial Logger will give you that information.

Barrygordon, 2020-07-27:
I have moved ahead and then regressed. I am now running with 2.2.0b.  I thought the latest was 2.2.0c but when I did the install using aqua.sh latest it gave me 2.2.0b

If I tap on the waterfall button (which always reports on)  an aqualinkd restart (and the waterfall is on) the waterfall properly turns off. Perhaps there is an aqualink program turning on the waterfall.  I will look.

I am trying to use aqualinkd with two of my own systems which have different MQTT clients installed. They both behave differently. One of them, 'commandfusion MQTT client'  only receives the first 6 messages on a subscription and never receives any more. That is the one I need for my Home automation iPads.  The second, 'AWS_Proxy nodejs MQTT client', receives all the messages several times and then receives no more.  I am not sure what is going on but it was working better with version 2.0.1.  Is there any way I can reinstall 2.0.1?

I have attached several files as a zip archive. There is the serial_logger output,  the aqualinkd log file, the aqualinkd configuration file, and two files showing the MQTT outputs in the two systems I mentioned.

I do not believe the MQTT problem is yours.

[aqualinkd.zip](https://github.com/sfeakes/AqualinkD/files/4983926/aqualinkd.zip)

The only thing I really need is a way to reinstall 2.0.1 as that will give me a working system with extra SWG messages which I can ignore or just change the SWG setting to Yes as you indicated. I do not believe I need any of the features in 2.2.0

sfeakes, 2020-07-27:
To go back a version, simply download from GitHub and install.  There is no checking that goes on. (You should also change your config appropriately).  Releases are here.
https://github.com/sfeakes/AqualinkD/releases

Looking over your information, something strange is going on.  Are you sure you don't have multiple instances running or something like that?  I don't think going back a version will fix the underlying problems.
I'd expect to see a more output from the serial logger.
```
cd ~
mkdir test
cd test
wget https://github.com/sfeakes/AqualinkD/raw/master/release/serial_logger
chmod 777 serial_logger
sudo systemctl stop aqualinkd
ps -ef | grep [a]qualinkd
--> If the above command gives you an output, then you have multiple instances of aqualinkd running. Stop them before going further <---
sudo ./serial_logger /dev/ttyUSB0
```
Please post the output of that command.

But, most importantly, it looks like AqualinkD is not communicating with your control panel correctly and we need to find out why. Can you set aqualinkd to debug mode, and post the log file.  
Edit /etc/aqualinkd.conf make sure you have the following uncommented and no duplicate lines
```
log_level=DEBUG
log_file=/var/log/aqualinkd.log
```

Start Aqualinkd, let it run for at least 2 minutes.   Stop aqualinkd and post that file.

Barrygordon, 2020-07-29:
A side effect of playing with AqualinkD is that my Linux command line skills are improving.

I have done as you asked and all the relevant files are attached.  I have also attached one file from running with 2.1.0 with a the config parameter  force_SWG  being both yes and no. 

[AqualinkD 2.1.0 feedback.txt](https://github.com/sfeakes/AqualinkD/files/4996324/AqualinkD.2.1.0.feedback.txt)
[AqualinkD 2.2.0b.txt](https://github.com/sfeakes/AqualinkD/files/4996325/AqualinkD.2.2.0b.txt)
[sfeakes requested serial_logger test.txt](https://github.com/sfeakes/AqualinkD/files/4996327/sfeakes.requested.serial_logger.test.txt)
[aqualinkd.conf.txt](https://github.com/sfeakes/AqualinkD/files/4996328/aqualinkd.conf.txt)

sfeakes, 2020-07-30:
Thanks for this.   Can you please also post the file `/var/log/aqualinkd.log`.    That's the file that should be quite large and have all the aqualinkd logging information.

Barrygordon, 2020-07-30:
I'm sorry.  I thought I did.  Here it is.
[aqualinkd.log.txt](https://github.com/sfeakes/AqualinkD/files/4998583/aqualinkd.log.txt)

Barrygordon, 2020-08-03:
Hi,

Did you find anything relative to version 2.2.0b and the MQTT issue I have with that version?

sfeakes, 2020-08-04:
Sorry, I thought I had replied to this before.
On the MQTT issue, From the logs you posted, AqualinkD is getting conformation messages that the MQTT posts were successful, and even getting the messages back.  So I would check your MQTT logs, or use a command like subscription tool like mosquito_sub to monitor the messages.  The below pulled from your logs, you can see AqualinkD sending MQTT message, then receiving conformation from the MQTT server, then getting it's own message back.  This is the same for every MQTT message it's sent.
```
Jul-29-20 12:31:22 PM Info:    NetService:MQTT: Published id=41: aqualinkd/Temperature/Air 35.56
Jul-29-20 12:31:22 PM Debug:   NetService:MQTT: Message publishing acknowledged (msg_id: 41)
Jul-29-20 12:31:22 PM Debug:   NetService:MQTT: Ignore aqualinkd/Temperature/Air 35.56
```

On the thermostat issue, I think I know what the problem is, but I'll need to do some more testing.  Sometimes AqualinkD fails in getting your pool / spa setpoints, so you are not getting thermostats in the UI.  The below settings should help as it will use a different protocol to retrieve that information.
```
extended_device_id = 0x30
extended_device_id_programming = yes
```

closed 2025-11-18 by github-actions[bot]