aqualinkd/AqualinkD #109: Watts reporting truncated

niharmehta, 2020-07-03:
Running 2.10 with VSP support with OneTouch enabled. While monitoring my system in a high speed spillover mode, I noticed that MQTT messages were essentially flipping back and forth between actually Watts and then a truncated reading with the first digit (of 4) removed.  I don't think this is happening at low speeds with 3 digits.    
The debugs seem to indicate that it is reading from the Pentair correctly, but possibly the Onetouch code may be truncating that first digit off? 

`Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 0 = Equipment Status
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 1 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 2 = Intelliflo VS 1 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 3 =      RPM: 3250  
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 4 =    Watts: 2225  
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 5 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 6 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 7 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 8 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 9 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 10 = 
Jul-02-20 22:38:23 PM Info:   OneTouch Menu Line 11 = 
Jul-02-20 22:38:23 PM Debug:  OneTouch Pump Intelliflo VS 1 , Index 1, RPM 3250, Watts 225, GPM 0
Jul-02-20 22:38:23 PM Debug:  RS received ONETOUCH packet of type Status length 12
Jul-02-20 22:38:23 PM Debug:  OneTouch Sending '0x04' to controller`

Logs attached.
[aqualinkd.log.07022240.zip](https://github.com/sfeakes/AqualinkD/files/4867947/aqualinkd.log.07022240.zip)

sfeakes, 2020-07-03:
Thanks for this.  I see the problem and it'll be fixed in next update. You're 100% accurate, when AqualinkD reads the information directly from the pump it's accurate, when it reads from the Onetouch status it's picking up the wrong start digit so it will only get 3 digits and not 4.

sfeakes, 2020-07-18:
I've posted an update that includes this fix.

closed 2020-07-29 by sfeakes