aqualinkd/AqualinkD #110: Heater Setpoint status delayed multiple seconds after changing through http PUT

agprimatic, 2020-07-10:
When changing the Pool Heater set point with the curl command:
`curl -X PUT -d '{"value":"84"}' "aqualink.local/api/Pool_Heater/setpoint/set"`

It takes many seconds before the change is reflected in the status read back by the curl command:
`curl "aqualink.local/api/status"`

If this helps, the actual status change coincides with this message disappearing from the top of the webpage:
Programming: setting heater

Is this a limitation of the Jandy system, or could the reporting of the changed set point be faster?

sfeakes, 2020-07-10:
It’s a limitation of the Jandy system.  AqualinkD has to program the setpoint change, and that can take between 5 and 30 seconds depending on the protocol being used.  API Status always reports the actual status Incase the programming of the control panel fails.

agprimatic, 2020-07-11:
Is there any possible way to support a setpoint increment or decrement?

I am replacing my Jandy Aqualink RS Serial Adapter with AqualinkD, and the Serial Adapter can send SP+ and SP- messages. I am emulating that by reading the status, adding (or subtracting) 1, then sending the new setpoint value. But with the slow updates, this isn't really a viable option.

sfeakes, 2020-07-11:
Not currently.  The RS Serial Adapter uses a protocol I have yet been able to decode. (I've got every other control protocol decoded). And just about to release AqualinkTouch protocol that will speed things like this up tremendously. 

If you would like to send me the RS Serial Adapter, I'll decode the protocol and add anything useful to AqualinkD and send it back to you. (probably take me a few weeks).  Or in a week or so I'm about to release AqualinkD with the AqualinkTouch protocol built in. That will take changing thermostat set points down to about 5 seconds.   I'd still like to decode the RS Serial Protocol, I've just never had the opportunity to do so.

agprimatic, 2020-07-17:
I would like to loan you my RS Serial Adapter, as I am interested in having AqualinkD adjust the temperature set point and increment/decrement faster. Please let me know where to send it. My email is agprimatic@gmail.com.
I would also be interested in getting the change-of-state messages that the Serial Adapter supports working in some fashion over the network interface. Not sure the best way to handle this, but I would rather not do constant polling for status.
Thanks!

sfeakes, 2020-08-28:
Please see V2.2.1 update.  It now supports RS Serial Adapter protocol and can therefor change setpoints instantly.  Also added functionality for Pool_Heater/setpoint/increment.    Again thanks for your help on this.

closed 2025-11-18 by github-actions[bot]