aqualinkd/AqualinkD #111: Temperature set points don’t seem to stay in sync between HomeKit and the panel

adamsinger, 2020-07-11:
Hi there!

I recently upgraded homebridge and AqualinkD to the latest versions to try to fix an unrelated issue, but I’ve started to notice that while I can control most of the pool’s functions via HomeKit, whenever I change temperature set points, they do not sync up with the panel. The sync between the panel and AqualinkD works just fine, but sync between HomeKit and the panel does not work in either direction.

I’ve tried waiting a few minutes before checking, but they never seem to update.

Happy to shoot over any relevant logs you may need

sfeakes, 2020-07-11:
I think there have been some thermostat changes in homebridge recently, did you upgrade homebridge as well?

Look for any errors in wherever you have homebridge log pointing to, (you may also have to play a bit to get anything meaningful from homebridge logging). If you see anything please post it and I'll see if there is a quick workaround.  I'll update my stuff soon as I need to make some changes to homebridge-aqualinkd anyway to upgrade to make use of some new functionality in homekit.  So I may well come across the same issue soon.

On a note, homebridge is absolutely horrible for breaking on new releases, so it's always best to try on a VM / dev / sandbox before making changes to you home system. I honestly refuse to upgrade homebridge on home systems unless I absolutely have to, it's always such a nightmare.

adamsinger, 2020-07-11:
Is there a version of homebridge that is known to work? Before I do, I’ll try to collect some logging, of course, but would be helpful to know if there’s a version that would work properly.

The thing that complicates the logging of homebridge is that when it first starts up, it does give me that getCharacteristic failed error, but that’s also during a time when AqualinkD is still showing heaters as switches. As with a previous issue, they don’t show up as thermostats in AqualinkD until I fiddle with the set points in the iAquaLink app or the control panel first

adamsinger, 2020-07-12:
So I fiddled around with it a bit more. It seems to *sometimes* work. Attached are logs when I attempt to change the setpoint via the Home app.

Here are the homebridge logs I see when it does work

```
  EventedHTTPServer [::ffff:192.168.0.242] HTTP request: /characteristics +6s
  HAPServer [CB:22:5D:E2:CE:31] HAP Request: PUT /characteristics +4ms
  Accessory [Homebridge] Processing characteristic set: [{"aid":27,"iid":26,"value":38.9}] +11ms
  Accessory [Homebridge] Setting Characteristic "Target Temperature" to value 38.9 +4ms
[7/11/2020, 5:50:42 PM] [aqualinkd] Aqualinkd_accessory.setTargetTemperature 'Spa_Heater' 38.9
[7/11/2020, 5:50:43 PM] [aqualinkd] MQTT send topic:'aqualinkd/Spa_Heater/setpoint/set' message:''38.9'
  EventedHTTPServer [::ffff:192.168.0.100] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +32ms
  EventedHTTPServer [::ffff:192.168.0.103] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +12ms
  EventedHTTPServer [::ffff:192.168.0.105] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +9ms
  EventedHTTPServer [::ffff:192.168.0.242] Muting event '27.26' notification for this connection since it originated here. +8ms
  EventedHTTPServer [::ffff:192.168.0.101] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +3ms
  EventedHTTPServer [::ffff:192.168.0.242] HTTP Response is finished +12ms
[7/11/2020, 5:50:50 PM] [aqualinkd] MQTT acting on message for ID:'Spa_Heater'. Topic:'Spa_Heater/setpoint' Message:'38.89' Value used:'38.89'
[7/11/2020, 5:50:50 PM] [aqualinkd] Aqualinkd_accessory.setTargetTemperature 'Spa_Heater' 38.8
  EventedHTTPServer [::ffff:192.168.0.100] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.8}]} +7s
  EventedHTTPServer [::ffff:192.168.0.103] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.8}]} +8ms
  EventedHTTPServer [::ffff:192.168.0.105] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.8}]} +10ms
  EventedHTTPServer [::ffff:192.168.0.242] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.8}]} +11ms
  EventedHTTPServer [::ffff:192.168.0.101] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.8}]} +9ms
```

Here are the logs I see when it doesn't work

```
  Accessory [Homebridge] Processing characteristic set: [{"aid":27,"iid":26,"value":38.9}] +11ms
  Accessory [Homebridge] Setting Characteristic "Target Temperature" to value 38.9 +4ms
[7/11/2020, 5:53:57 PM] [aqualinkd] Aqualinkd_accessory.setTargetTemperature 'Spa_Heater' 38.9
[7/11/2020, 5:53:57 PM] [aqualinkd] MQTT send topic:'aqualinkd/Spa_Heater/setpoint/set' message:''38.9'
  EventedHTTPServer [::ffff:192.168.0.100] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +29ms
  EventedHTTPServer [::ffff:192.168.0.103] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +10ms
  EventedHTTPServer [::ffff:192.168.0.105] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +11ms
  EventedHTTPServer [::ffff:192.168.0.242] Muting event '27.26' notification for this connection since it originated here. +11ms
  EventedHTTPServer [::ffff:192.168.0.101] Sending HTTP event '27.26' with data: {"characteristics":[{"aid":27,"iid":26,"value":38.9}]} +2ms
  EventedHTTPServer [::ffff:192.168.0.242] HTTP Response is finished +10ms
```

It seems as though it sends the request to AqualinkD, and AqualinkD just doesn't always update the set points. Sometimes when I update the temperature in the AqualinkD web portal, it doesn't update the actual set point immediately. Sometimes there is a very long delay. Perhaps this is expected?

Playing with it some more, there does seem to be communication between Homebridge <-> AqualinkD <-> Panel, but it's inconsistent and there are sometimes delays. This only seems to happen with set points. Switches and pumps seem to respond almost immediately.

sfeakes, 2020-07-12:
It takes some time to change set points, as AqualinkD has to put the panel into programming mode where as switches are instant. (5 to 30 seconds depending on protocols that have been configured and the amount of change).
If you use the web ui, you'll see the message and how long it takes.  So that time ~30 seconds will the be time that's reflected in homekit.   In pre AqualinkD V1.x I may have told homekit it had been set before it actually had been to stop this confusion.  I'll check and may make some updates.

adamsinger, 2020-07-13:
That’s helpful.

After fiddling with it some more, I will say there have been times I’ve updated the set point temperature on the panel, and it *never* gets updated in HomeKit. I set the temp today to 88 and several hours later HomeKit is still saying an old number.

sfeakes, 2020-07-14:
Can you confirm that the web UI is displaying the right set point and HomeKit isn’t?   I say that because heater setpoints take a long time to program and if you have rs485 errors then they are the first to fail.   So check AqualinkD log (or syslog depending on your config) for any checksum or setpoint errors.

adamsinger, 2020-07-14:
Correct. HomeKit gives me the wrong temperature. The control panel and the web UI both display the correct temperature. Does that still warrant looking at the logs, or an indication of a different problem?

sfeakes, 2020-07-14:
It’s worth checking the logs for errors.  Also HomeKit uses MQTT, so you can use mosquitto_sub on the AqualinkD topic to monitor what messages are being sent.
Messages are set to hold, so just running ‘mosquitto_sub -t “aqualinkd/#” -v‘ should print out the last messages.   If you can list what the Thermostat states (it’s be in C not F) and if that’s accurate, that will help a lot.

adamsinger, 2020-07-15:
Fiddled some more, and looked at the mosquitto logs:

Example log when setting the temperature from the WebUI, and everything works:

```
aqualinkd/Spa_Heater/setpoint/set 36.7
aqualinkd/Spa_Heater/setpoint 36.67
```

Setting the setpoint from the Home app immediately causes a `aqualinkd/Spa_Heater/setpoint/set` message, but doesn't always get a `aqualinkd/Spa_Heater/setpoint` message, even if I wait 30+ seconds.

Setting a setpoint from either the panel or from the Web UI doesn't always result in a `setpoint/set` message in mosquitto, even if I wait several minutes.

After doing some more testing, I'm pretty convinced the issue has nothing to do with homebridge at all. Whenever mosquitto receives the `setpoint` message, Homekit updates immediately to the correct temp, and whenever I set the temperature from Homekit, it always sends the `setpoint/set` message, again with the correct temp.

Conversely, setting the temperature from the panel sometimes results in the Web UI never updating, and vice versa.

Is 30 seconds the actual maximum time it should take for those to become in sync? Because if so, there's definitely an issue communicating between aqualinkD and my panel. And again: everything else works perfectly fine. Just setting setpoints seems to be problematic.

adamsinger, 2020-07-15:
It's also a little confusing because sometimes it does work, and works several times in a row, but then it sort of gets stuck where several tries in a row will fail, even with waiting minutes in between tries.

sfeakes, 2020-07-16:
Setting the set point from the panel will never reflect in MQTT, as the panel doesn’t send that information, Aqualinkd has to retrieve it.
If you’re setting it in the UI and it doesn’t always work, then this is your problem, AqualinkD if failing to program the panel.   Please look in the logs for errors.

The 30 seconds is simply how long AqualinkD “could” take to program the panel.  There are a lot of variables, sometimes it’ll only take a few seconds.

adamsinger, 2020-07-16:
Sorry, to be clear: the problem seems to be bidirectional between the Web UI (aqualinkd) and the panel. I know that aqualinkd is what sends the MQTT message, but it seems like programming from the panel does not always update aqualinkd (and vice versa).

In any case, is there anything in the logs I should be looking out for?

And on the 30 seconds: is that the “maximum” it could take? Or could it potentially take longer? I’d just want to know how long to wait before I can say that it definitely did not work.

sfeakes, 2020-07-16:
Logs will have error noted to anything important.   You may have to set debug logging to get full details.    If you see any checksum errors, that’s your problem and it’s hardware.  If panel programming errors with no checksum errors, then you’ll probably need to turn on debug logging.
30 sec should be the longest it takes under the worse circumstances.

adamsinger, 2020-07-16:
So with log level set to debug, the only odd thing I've noticed is occasionally I'll get this:

```
Jul-16-20 12:36:19 PM Warning:BAD PACKET To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x72|0x08|0x01|0x02|0x03|0x04|0x05|0x06|0x07|0x20|0x00|0x01|0x00|0x00|0x0b|0x57|0x74|0x72|0x20|0x46|0x65|0x61|0x74|0x75|0x72|0x65|0x00|0x01|0x00|0x00|0x0a|0x50|0x6f|0x6f|0x6c|0x20|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x09|0x53|0x70|0x61|0x20|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x0c|0x46|
Jul-16-20 12:36:19 PM Warning:Serial packet too large
```

I may be imagining it, but it does seem like I will get 3-5 of these bad packet warnings, once every 10 seconds, starting a few seconds after I attempt to change the temperature (in either the WebUI or the panel itself), then they stop until I try to change the temperature again.

I also tried checking for checksum errors both in these logs and using your instructions in the Wiki, and there were none.

sfeakes, 2020-07-17:
I Assume you have an AqualinkTouch control panel.  That’s nothing to worry about, and I fixed it in current release “not yet posted”.  That shouldn’t effect your programming problem, but it is interesting. Are you setting the thermostat from AqualinkD, then playing with the AqualinkTouch control panel constantly checking if it set?

adamsinger, 2020-07-17:
What I've been doing is setting the temperature (with the heater on), from the web UI, waiting 30 seconds, watching logs, then trying again. I am also trying setting the temperature from the panel, and doing the same. I always wait at least 30 seconds before trying again. Sometimes I'll even wait several minutes.

adamsinger, 2020-07-17:
For what it's worth, I have an AquaLink RS6 panel in my bedroom, which is where I usually change settings, and it's where the Raspberry Pi is attached

I've attached photos of my main panel in my pool equipment room. Unsure the exact model number

[image]

[image]

sfeakes, 2020-07-17:
Can you post the complete log file in debug mode.  I should be able to tell what’s going wrong if you post the file (obviously include up to a point it fails)

adamsinger, 2020-07-31:
Hi. Sorry for the delay. I'm attaching one complete log when it does work, and then one complete (~30 second) log when it does not work. Both initiated from the Web UI

```
Jul-31-20 11:25:32 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:32 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:32 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:32 AM Warning:BAD PACKET To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x72|0x08|0x01|0x02|0x03|0x04|0x05|0x06|0x07|0x20|0x00|0x01|0x00|0x00|0x0b|0x57|0x74|0x72|0x20|0x46|0x65|0x61|0x74|0x75|0x72|0x65|0x00|0x01|0x00|0x00|0x0a|0x50|0x6f|0x6f|0x6c|0x20|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x09|0x53|0x70|0x61|0x20|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x0c|0x46|
Jul-31-20 11:25:32 AM Warning:Serial packet too large
Jul-31-20 11:25:32 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:33 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:33 AM Debug:  RS received packet of type Message length 24
Jul-31-20 11:25:33 AM Info:   RS Message :- 'Air Temp 79ﬂF'
Jul-31-20 11:25:33 AM Info:   MQTT: Published id=21990: aqualinkd/SWG 0
Jul-31-20 11:25:33 AM Info:   MQTT: Published id=21991: aqualinkd/SWG/enabled 0
Jul-31-20 11:25:33 AM Info:   MQTT: Published id=21992: aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:33 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21990)
Jul-31-20 11:25:33 AM Info:   MQTT: received (msg_id: 21990), looks like my own message, ignoring
Jul-31-20 11:25:33 AM Debug:  MQTT: Ignore aqualinkd/SWG 0
Jul-31-20 11:25:33 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21991)
Jul-31-20 11:25:33 AM Info:   MQTT: received (msg_id: 21991), looks like my own message, ignoring
Jul-31-20 11:25:33 AM Debug:  MQTT: Ignore aqualinkd/SWG/enabled 0
Jul-31-20 11:25:33 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21992)
Jul-31-20 11:25:33 AM Info:   MQTT: received (msg_id: 21992), looks like my own message, ignoring
Jul-31-20 11:25:33 AM Debug:  MQTT: Ignore aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:33 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:33 AM Info:   MQTT: Published id=21993: aqualinkd/SWG 0
Jul-31-20 11:25:33 AM Info:   MQTT: Published id=21994: aqualinkd/SWG/enabled 0
Jul-31-20 11:25:33 AM Info:   MQTT: Published id=21995: aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:33 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21993)
Jul-31-20 11:25:33 AM Info:   MQTT: received (msg_id: 21993), looks like my own message, ignoring
Jul-31-20 11:25:33 AM Debug:  MQTT: Ignore aqualinkd/SWG 0
Jul-31-20 11:25:33 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21994)
Jul-31-20 11:25:33 AM Info:   MQTT: received (msg_id: 21994), looks like my own message, ignoring
Jul-31-20 11:25:33 AM Debug:  MQTT: Ignore aqualinkd/SWG/enabled 0
Jul-31-20 11:25:33 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21995)
Jul-31-20 11:25:33 AM Info:   MQTT: received (msg_id: 21995), looks like my own message, ignoring
Jul-31-20 11:25:33 AM Debug:  MQTT: Ignore aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:33 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:33 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:33 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:34 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:34 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:34 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:34 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:34 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:35 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:35 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:35 AM Debug:  RS received packet of type Message length 24
Jul-31-20 11:25:35 AM Info:   RS Message :- 'Pump OFF'
Jul-31-20 11:25:35 AM Info:   MQTT: Published id=21996: aqualinkd/SWG 0
Jul-31-20 11:25:35 AM Info:   MQTT: Published id=21997: aqualinkd/SWG/enabled 0
Jul-31-20 11:25:35 AM Info:   MQTT: Published id=21998: aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:35 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21996)
Jul-31-20 11:25:35 AM Info:   MQTT: received (msg_id: 21996), looks like my own message, ignoring
Jul-31-20 11:25:35 AM Debug:  MQTT: Ignore aqualinkd/SWG 0
Jul-31-20 11:25:35 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21997)
Jul-31-20 11:25:35 AM Info:   MQTT: received (msg_id: 21997), looks like my own message, ignoring
Jul-31-20 11:25:35 AM Debug:  MQTT: Ignore aqualinkd/SWG/enabled 0
Jul-31-20 11:25:35 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21998)
Jul-31-20 11:25:35 AM Info:   MQTT: received (msg_id: 21998), looks like my own message, ignoring
Jul-31-20 11:25:35 AM Debug:  MQTT: Ignore aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:35 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:25:35 AM Info:   MQTT: Published id=21999: aqualinkd/SWG 0
Jul-31-20 11:25:35 AM Info:   MQTT: Published id=22000: aqualinkd/SWG/enabled 0
Jul-31-20 11:25:35 AM Info:   MQTT: Published id=22001: aqualinkd/SWG/fullstatus 255
Jul-31-20 11:25:35 AM Info:   MQTT: Message publishing acknowledged (msg_id: 21999)
[... cut ...]
Jul-31-20 11:27:52 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:27:52 AM Debug:  RS SEND cmd '0x00' (programming)
Jul-31-20 11:27:52 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:27:52 AM Debug:  RS SEND cmd '0x00' (programming)
Jul-31-20 11:27:52 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:27:52 AM Debug:  RS SEND cmd '0x00' (programming)
Jul-31-20 11:27:52 AM Debug:  RS received packet of type Status length 12
Jul-31-20 11:27:52 AM Debug:  RS SEND cmd '0x00' (programming)
```

sfeakes, 2020-07-31:
I see the problem.  I'll post a fix (but due to other commitments probably be a few weeks).  But there are some changes in the latest release that *may* actually help this.   What version are you running?

If you pull 2.2.0c and put the following in aqualinkd config it may help.  BTW, these are experimental settings designed to test/help something else, but they may help for this issue
```
thread_netservices = yes
rs_poll_speed = -1
```
If they don't help, simply comment them out.
Please make sure to read the upgrade section of the wiki, since 2.2.0x has some config changes from 2.1.0

adamsinger, 2020-07-31:
I'll give it a try and report back. Thanks!

I just upgraded to 2.2.0c, and also updated my config as instructed.

adamsinger, 2020-07-31:
I just tried it out with those extra settings. Still not syncing properly. Will wait for your update

sfeakes, 2020-08-28:
Posted AqualinkD v2.1.1, if you update and set the `rssa_device_id=` option (details in release notes), that should keep everything in sync for you.

adamsinger, 2020-08-31:
Works perfect now! Thank you

closed 2020-08-31 by sfeakes