aqualinkd/AqualinkD #112: Colored Lights

scroundy, 2020-07-15:
My light remembers the last color it was set to. It doesn't not reset back to the first in a series of show or colors. Am I right in understanding that the AqualinkD programming is based on having the light reset to an initial color?

sfeakes, 2020-07-15:
That depends on what programming mode you are using.  If you just turn the light on from the button, (with out selecting a color) it will default to last mode if your using AqualinkD to program the light. If you’re using the control panel to program the light, then it will do the same, but will be very slow to do that (I’ve already made a change in the next release to fix the speed).

sfeakes, 2020-07-16:
> My light has a memory in it (generic LED purchased off amazon). I can control through the control panel by cycling on/off. If turned on then off it will change to the next setting. If left on for 2 minutes and then turned off, then it will remember the color and when turned back on will be the same color. I don't see how aqualinkd will sync with it since just by turning on/off the light quickly it will switch. Would I need to set light_programming_initial_on to 120?

You either control the on/off pulses through AqualinkD or you configure the panel to do it for you.  Either way AqualinkD needs to know who is sending the pulses.  That is done with the `button_xx_lightmode=` option.   IF you set that option to 0 (ie AqualinkD does the pulsing of the on/off), then you need to configure the next options.

light_programming_mode=light_programming_initial_on=
light_programming_initial_off=

Set mode to 0 it’s the safest, you’ll have to read the light manual for the initial on and off settings. You need the settings for above to put the light into programming mode.
The 2 minute you're talking about is ignored.  

If you set button_xx_lightmode= to something other than 0, then the Panel is pulsing the light on and off, and you need to set it their.

closed 2020-08-28 by sfeakes