aqualinkd/AqualinkD #113: Overload

nmoscone, 2020-07-15:
I am not sure if related, however I configured and installed this a few weeks ago; it worked great I had no issues at all.  Very nice job on the coding!

However all of a sudden the main boards in the controller fired, a chip on the back shorted out (if such I think it’s the relay driver chip).   I am 100% sure it was not a wire short in the panel, I was not using the app, we were in the pool and all of a sudden the pumps shut off.  Also it’s not the glass fuse I tried that it just pops again that chip melted / has burn mark.

Have you ever heard of this issue occurring?   I ordered another board to replace it, I am thinking maybe it’s just due to the age of the controller.   

But I wanted your opinion before I try to hook this back up to the new board.  Do you think in any way this could have over heated the board to cause this?   Maybe to many checks, would it have mattered if I left the site open by mistake?   Or is there any settings I may have overlooked when I did the config?

Thank you

sfeakes, 2020-07-18:
I've never heard of anything like this before on a control panel, there are lots of similar issues with specific Jandy SWG's, as they made some errors in spec's on something in the transformer. Also I don't see any way AqualinkD stresses the Panel CPU any more than any other Jandy keypad, as it uses those protocols and same timing.  Hope this helps.

nmoscone, 2020-07-18:
Thank you for getting back to me.  I don’t think so either, figured it was my bad luck that I just got it hooked up and 3 weeks later the system died; but wanted to double check to see what your thought was before I reconnect it to the new board.  It’s a huge pain to run the pumps and valves manually the last thing I want to do it fry another board!

Again great job on the coding.

closed 2020-08-28 by sfeakes