aqualinkd/AqualinkD #114: Crash on v2.2 when using AqualinkTouch mode.

niharmehta, 2020-07-19:
Updated to v2.2 . Without config changes, it seemed to start fine.  Came back a bit later and made the recommended config changes for AqualinkTouch support instead of OneTouch, Panel Type, and Time Sync.   Restarting is causing AqualinkD to crash a few seconds after starting.  Verified with serial_logger that I am using a preferred ID and even changed it from 0x31 to 0x30 with the same result.  Actually moved it back to 0x41 and still hand a problem, but a couple of minutes I restarted it and it seemed to work.  I have attached serial debug log of the crash. 

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/4943005/aqualinkd.log)

sfeakes, 2020-07-19:
Thanks very much for loading this so quickly and providing logs.   The first two times it looks like it didn’t receive a probe on ID 0x30 or 0x31 (the two IDs you tried), so AqualinkD gave up / timed out.   but kinda looks like on the 3rd try It receives a probe.    I may have to increase the timeout on probes.   I’ll take a further look at you logs tomorrow and maybe post an update.

These are the message 
Jul-18-20 16:51:08 PM Error:   AqualinkD: No probe on '0x30', giving up! (please check config)
Jul-18-20 16:51:08 PM Debug:   RS Serial: Closed serial port

This is the first time I’ve posted an update that I’ve not been able to test on a real panel. My control panel doesn’t support aqualink touch protocol, but I’ve managed to get a few versions of the Jandy Panel firmware Running in a custom simulator and that‘s what this was developed against.

sfeakes, 2020-07-19:
Actually, can you post the output from serial_logger if/when you get some time?

niharmehta, 2020-07-19:
Here is the output from serial_logger. Let me know if you think getting remote access to my system can help until you get enough donations to get your own RS8. 

AqualinkD serial_logger V1.3
Notice:  RS Serial: Logging serial information!
Please wait.|

Notice:  RS Serial: Jandy Control Panel Model   :  RS-8 Combo
Notice:  RS Serial: Jandy Control Panel Version :  REV T
Notice:  RS Serial: Jandy ID's found
Notice:  RS Serial: ID 0x33 is in use  <-- Aqualink (iAqualink / Touch)
Notice:  RS Serial: ID 0xa3 is in use  <-- Unknown Device
Notice:  RS Serial: ID 0x50 is in use  <-- Salt Water Generator (Aquarite mode)
Notice:  RS Serial: ID 0x0a is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x60 is not used  <-- can use for Aqualinkd (PDA mode only)
Notice:  RS Serial: ID 0x0b is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x40 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x41 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x42 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x43 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x30 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x31 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x32 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x08 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x09 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: 

Notice:  RS Serial: Pentair ID's found
Notice:  RS Serial: ID 0x60 is in use  <-- Pentair VSP
Notice:  RS Serial: ID 0x10 is in use  <-- Pentair Master (Probably Jandy RS Control Panel)
Notice:  RS Serial:

lehighkid, 2020-07-19:
I also have this exact issue.  I am able to run aqualinkd w/o issue (if I set my `RS4-Only` to `RS8-Only`) from the command line over ssh
`pi@aqualinkd:~/software/AqualinkD $ sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf`

Using systemctl results in a timeout or a crash
`pi@aqualinkd:~/software/AqualinkD $ sudo systemctl status aqualinkd`
 
timeout:
```
pi@aqualinkd:~ $ sudo systemctl status aqualinkd
● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sun 2020-07-19 09:12:07 EDT; 2min 46s ago
 Main PID: 7006 (code=exited, status=0/SUCCESS)

Jul 19 09:11:56 aqualinkd aqualinkd[7005]: AqualinkD: Config BTN Solar_Heater  = label NONE            |
Jul 19 09:11:56 aqualinkd aqualinkd[7006]: NetService:Starting web server on port 80
Jul 19 09:11:56 aqualinkd aqualinkd[7006]: NetService:Starting MQTT client to XXX.XXX.XXX.XXX:1883
Jul 19 09:11:56 aqualinkd aqualinkd[7006]: AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jul 19 09:11:56 aqualinkd aqualinkd[7006]: AqualinkD: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Jul 19 09:11:56 aqualinkd aqualinkd[7006]: AqualinkD: Waiting for Control Panel probe
Jul 19 09:11:56 aqualinkd aqualinkd[7006]: AqualinkD: Got message but no probe on '0x30', did we start too soon? (waiting for probe)
Jul 19 09:12:07 aqualinkd aqualinkd[7006]: AqualinkD: No probe on '0x30', giving up! (please check config)
Jul 19 09:12:07 aqualinkd aqualinkd[7006]: NetService:MQTT Connection closed
Jul 19 09:12:07 aqualinkd systemd[1]: aqualinkd.service: Succeeded.
```
or a seg fault:
```
pi@aqualinkd:~ $ sudo systemctl status aqualinkd
● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sun 2020-07-19 09:15:38 EDT; 1s ago
  Process: 7056 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
 Main PID: 7057 (code=killed, signal=SEGV)

Jul 19 09:15:23 aqualinkd aqualinkd[7056]: AqualinkD: Config BTN Spa_Heater    = label NONE            |
Jul 19 09:15:23 aqualinkd aqualinkd[7056]: AqualinkD: Config BTN Solar_Heater  = label NONE            |
Jul 19 09:15:23 aqualinkd aqualinkd[7057]: NetService:Starting web server on port 80
Jul 19 09:15:23 aqualinkd aqualinkd[7057]: NetService:Starting MQTT client to XXX.XXX.XXX.XXX:1883
Jul 19 09:15:23 aqualinkd aqualinkd[7057]: AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jul 19 09:15:23 aqualinkd aqualinkd[7057]: AqualinkD: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Jul 19 09:15:23 aqualinkd aqualinkd[7057]: AqualinkD: Waiting for Control Panel probe
Jul 19 09:15:34 aqualinkd aqualinkd[7057]: AqualinkD: Starting communication with Control Panel
Jul 19 09:15:38 aqualinkd systemd[1]: aqualinkd.service: Main process exited, code=killed, status=11/SEGV
Jul 19 09:15:38 aqualinkd systemd[1]: aqualinkd.service: Failed with result 'signal'.
```
here is my serial logger output for reference:
```
pi@aqualinkd:~/software/AqualinkD $ sudo ~/software/AqualinkD/release/serial_logger /dev/ttyUSB0
AqualinkD serial_logger V1.3
Notice:  RS Serial: Logging serial information!
Please wait.|

Notice:  RS Serial: Jandy Control Panel Model   :  RS-4 Only
Notice:  RS Serial: Jandy Control Panel Version :  REV T.2   
Notice:  RS Serial: Jandy ID's found
Notice:  RS Serial: ID 0x33 is in use  <-- Aqualink (iAqualink / Touch)
Notice:  RS Serial: ID 0xa3 is in use  <-- Unknown Device
Notice:  RS Serial: ID 0x50 is in use  <-- Salt Water Generator (Aquarite mode)
Notice:  RS Serial: ID 0x60 is not used  <-- can use for Aqualinkd (PDA mode only)
Notice:  RS Serial: ID 0x08 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x09 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x0a is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x0b is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x40 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x41 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x78 is in use  <-- Jandy VSP ePump
Notice:  RS Serial: ID 0x42 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x43 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x30 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x31 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
[... 54 lines cut ...]
Notice:  AqualinkD: Waiting for Control Panel probe
Notice:  AqualinkD: Starting communication with Control Panel
*****  iAqualink Touch STARTUP Message ******* 
Notice:  AqualinkD: Programming: Get iAqualink AUX Labels
**** AUX 1 = 'Pool Lights No '
**** AUX 2 = 'Aux2 No '
**** AUX 3 = 'Aux3 No '
**** AUX 0 = ' Aux Extra Aux No '
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
Notice:  AqualinkD: Programming: Get iAqualink Touch Setpoints
Notice:  iAQ Touch: IAQ Touch Freeze Protection setpoint 38
Notice:  RS Allbtn: Control Panel B0029226 REV T.2
```

sfeakes, 2020-07-19:
@lehighkid    Crap, sorry about that, I didn't even know an RS-4 existed, please use RS-6 for the moment, I'll fix that and post update
@niharmehta   Working on a fix for the startup probe as well.

sfeakes, 2020-07-19:
V2.2.0a just posted.  Should fix both of these issues.

niharmehta, 2020-07-19:
Upgraded to  V2.2.0a . At least in my case, it is actually worse.  I am unable to get AqualinkD started at all.  Seems like in the UI and logs that its connecting to control panel, then crashes.  I have tried various combinations of the extended device ID, (31,30,32,41). I have disabled the RS string, extended device programming,  NTP,  and just about any new feature and the same problem is occurring.  It wont start.   I have attached logs with each file a different attempt to start the service.  I have even rebooted my PI as well as power down the panel to see if something  is on the Jandy side.  Hopefully attached logs help.  Next step I will try is wiping my AqualinkD directory and doing a fresh download. 

[aqualinkd.logs.zip](https://github.com/sfeakes/AqualinkD/files/4944455/aqualinkd.logs.zip)

lehighkid, 2020-07-19:
Thank you.  I upgraded to V2.2.0a but it didn't fix the issue w/ the service but did address the RS4-Only panel type.

systemclt service status:
```
pi@aqualinkd:~/software/AqualinkD $ sudo systemctl status aqualinkd
● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sun 2020-07-19 15:01:22 EDT; 20s ago
  Process: 7705 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
 Main PID: 7706 (code=killed, signal=SEGV)

Jul 19 15:00:56 aqualinkd aqualinkd[7705]: AqualinkD: Config BTN Solar_Heater  = label Solar Heater    |
Jul 19 15:00:56 aqualinkd aqualinkd[7706]: NetService:Starting web server on port 80
Jul 19 15:00:56 aqualinkd aqualinkd[7706]: NetService:Starting MQTT client to XXX.XXX.XXX.XXX:1883
Jul 19 15:00:56 aqualinkd aqualinkd[7706]: AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jul 19 15:00:56 aqualinkd aqualinkd[7706]: AqualinkD: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Jul 19 15:00:56 aqualinkd aqualinkd[7706]: AqualinkD: Waiting for Control Panel probe
Jul 19 15:01:06 aqualinkd aqualinkd[7706]: AqualinkD: Starting communication with Control Panel
Jul 19 15:01:11 aqualinkd aqualinkd[7706]: AqualinkD: Programming: Get iAqualink AUX Labels
Jul 19 15:01:22 aqualinkd systemd[1]: aqualinkd.service: Main process exited, code=killed, status=11/SEGV
Jul 19 15:01:22 aqualinkd systemd[1]: aqualinkd.service: Failed with result 'signal'.
```
Here is the updated command line output w/ RS4-Only:
```
pi@aqualinkd:~/software/AqualinkD $ sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
Notice:  AqualinkD: Aqualink Daemon v2.2.0a
Notice:  AqualinkD: Panel set to RS-4 Pool/Spa Only 
Notice:  AqualinkD: Config log_level         = 5
Notice:  AqualinkD: Config device_id         = 0x0a
Notice:  AqualinkD: Config extra_device_id   = 0x30
Notice:  AqualinkD: Config extra_device_prog = YES
Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
Notice:  AqualinkD: Config socket_port       = 80
Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
Notice:  AqualinkD: Config read_all_devices  = YES
Notice:  AqualinkD: Config use_aux_labels    = YES
Notice:  AqualinkD: Config override frz prot = NO
Notice:  AqualinkD: Config mqtt_server       = XXX.XXX.XXX.XXX:1883
Notice:  AqualinkD: Config mqtt_dz_sub_topic = (null)
Notice:  AqualinkD: Config mqtt_dz_pub_topic = (null)
Notice:  AqualinkD: Config mqtt_aq_topic     = aqualinkd
Notice:  AqualinkD: Config mqtt_user         = aqualinkd
Notice:  AqualinkD: Config mqtt_passwd       = aqualinkd
Notice:  AqualinkD: Config mqtt_ID           = aqualinkd_b827ebd1
Notice:  AqualinkD: Config idx water temp    = -999
Notice:  AqualinkD: Config idx pool temp     = -999
Notice:  AqualinkD: Config idx spa temp      = -999
Notice:  AqualinkD: Config idx SWG Percent   = 0
Notice:  AqualinkD: Config idx SWG PPM       = 0
Notice:  AqualinkD: Config force SWG         = YES
Notice:  AqualinkD: Config deamonize         = NO
Notice:  AqualinkD: Config log_file          = (null)
Notice:  AqualinkD: Config light_pgm_mode    = 0.00
Notice:  AqualinkD: Debug RS485 protocol     = NO
Notice:  AqualinkD: Read Pentair Packets     = NO
Notice:  AqualinkD: Display warnings in web  = NO
Notice:  AqualinkD: Keep panle time in sync  = YES
Notice:  AqualinkD: Ignore SWG 0 msg count   = 20
Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | VSP ID 0x78 | PMP ID 1 |
Notice:  AqualinkD: Config BTN Aux_1         = label Pool Lights     | Light Progm | CTYPE 2  |
Notice:  AqualinkD: Config BTN Aux_2         = label Aux 2           | 
Notice:  AqualinkD: Config BTN Aux_3         = label Aux 3           | 
Notice:  AqualinkD: Config BTN Pool_Heater   = label Temp1 Heater    | 
Notice:  AqualinkD: Config BTN Spa_Heater    = label Temp2 Heater    | 
Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
Notice:  NetService:Starting web server on port 80
Notice:  NetService:Starting MQTT client to XXX.XXX.XXX.XXX:1883
Notice:  AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Notice:  AqualinkD: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Notice:  AqualinkD: Waiting for Control Panel probe
Notice:  AqualinkD: Starting communication with Control Panel
Notice:  RS Allbtn: Control Panel B0029226 REV T.2
Notice:  AqualinkD: Programming: Get iAqualink AUX Labels
**** AUX 1 = 'Pool Lights No '
**** AUX 2 = 'Aux2 No '
**** AUX 3 = 'Aux3 No '
**** AUX 0 = ' Aux Extra Aux No '
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
**** AUX 0 = ''
Notice:  AqualinkD: Programming: Get iAqualink Touch Setpoints
Notice:  iAQ Touch: IAQ Touch Freeze Protection setpoint 38
Warning: NetService:WEB: Old stanza, using old method to action
Warning: NetService:URI request: '/?command=homebridge'
Error:   JandyDvce: Ignoring set SWG device to state '0x82', state is unknown
```

sfeakes, 2020-07-19:
@niharmehta 

I don't think you are giving AqualinkD enough time to start up before restarting.  It is actually crashing / ie process died, or are you restarting it?  I'm not seeing any crashes in the logs.  Try starting it and leave it for a few minutes.  Then look in the logs / try the UI. 

aqualinkd.log.10 
All looks to be working, but as you said, you have no extended device id set.  I can see from the logs, it's posting to MQTT and accepting connections from HomeBridge.  There was request from web browser though.

aqualinkd.log.9
Started communicating on extended device id 0x41 got some pump VSP info and only JUST started communicating on 0x0a before it was reset.

aqualinkd.log.7
Started communicating with 0x31 (Aqualink touch), then just started on 0x0a before shutdown

sfeakes, 2020-07-19:
@lehighkid  Looks like there is a bug in reading aux labels on the Aqualink Touch protocol.  I'll fix that, but in the mean  time if you set `use_panel_aux_labels=no`, I'm hoping that will work for you.

lehighkid, 2020-07-19:
Yes, sir - thank you!
```
pi@aqualinkd:~ $ sudo systemctl status aqualinkd
● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-07-19 15:44:52 EDT; 1min 31s ago
  Process: 7979 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
 Main PID: 7980 (aqualinkd)
   Memory: 348.0K
   CGroup: /system.slice/aqualinkd.service
           └─7980 /usr/local/bin/aqualinkd -c /etc/aqualinkd.conf

Jul 19 15:44:52 aqualinkd aqualinkd[7979]: AqualinkD: Config BTN Solar_Heater  = label NONE            |
Jul 19 15:44:52 aqualinkd aqualinkd[7980]: NetService:Starting web server on port 80
Jul 19 15:44:52 aqualinkd aqualinkd[7980]: NetService:Starting MQTT client to XXX.XXX.XXX.XXX:1883
Jul 19 15:44:52 aqualinkd aqualinkd[7980]: AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jul 19 15:44:52 aqualinkd aqualinkd[7980]: AqualinkD: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Jul 19 15:44:52 aqualinkd aqualinkd[7980]: AqualinkD: Waiting for Control Panel probe
Jul 19 15:44:59 aqualinkd aqualinkd[7980]: AqualinkD: Starting communication with Control Panel
Jul 19 15:45:08 aqualinkd aqualinkd[7980]: RS Allbtn: Control Panel B0029226 REV T.2
Jul 19 15:45:13 aqualinkd aqualinkd[7980]: AqualinkD: Programming: Get iAqualink Touch Setpoints
Jul 19 15:45:28 aqualinkd aqualinkd[7980]: iAQ Touch: IAQ Touch Freeze Protection setpoint 38
```

niharmehta, 2020-07-19:
@sfeakes . The aqualinkd process stops running and is no longer in the running process list.  I will scrub my config and start from scratch .

niharmehta, 2020-07-19:
rolled back to v2.10 and it i working as before.

sfeakes, 2020-07-19:
@lehighkid    I just posted v2.2.0b.   Please try that.

@niharmehta Still looking into this.  If you get chance, can you try v2.2.0b, and post your config file and complete log file from start to where it dies.

Both, very sorry for the problems in this release.

lehighkid, 2020-07-19:
Thank you!  I have updated, reverted config to `use_panel_aux_labels=yes` and restarted the service and seems to be successful.
```
pi@aqualinkd:~/software/AqualinkD $ sudo systemctl status aqualinkd
● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-07-19 17:28:50 EDT; 1min 49s ago
  Process: 8353 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
 Main PID: 8354 (aqualinkd)
   Memory: 360.0K
   CGroup: /system.slice/aqualinkd.service
           └─8354 /usr/local/bin/aqualinkd -c /etc/aqualinkd.conf

Jul 19 17:28:50 aqualinkd aqualinkd[8354]: AqualinkD: AqualinkD is using use IAQ TOUCH mode for programming (where supported)
Jul 19 17:28:50 aqualinkd aqualinkd[8354]: AqualinkD: Waiting for Control Panel probe
Jul 19 17:28:59 aqualinkd aqualinkd[8354]: AqualinkD: Starting communication with Control Panel
Jul 19 17:29:08 aqualinkd aqualinkd[8354]: RS Allbtn: Control Panel B0029226 REV T.2
Jul 19 17:29:13 aqualinkd aqualinkd[8354]: AqualinkD: Programming: Get AUX labels
Jul 19 17:29:21 aqualinkd aqualinkd[8354]: RS Allbtn: AUX ID Aux_2 label set to 'Pool Lights'
Jul 19 17:29:21 aqualinkd aqualinkd[8354]: RS Allbtn: AUX ID Aux_3 has no control panel label using 'NONE'
Jul 19 17:29:22 aqualinkd aqualinkd[8354]: RS Allbtn: AUX ID Pool_Heater has no control panel label using 'NONE'
Jul 19 17:29:27 aqualinkd aqualinkd[8354]: AqualinkD: Programming: Get iAqualink Touch Setpoints
Jul 19 17:29:38 aqualinkd aqualinkd[8354]: iAQ Touch: IAQ Touch Freeze Protection setpoint 38
```

niharmehta, 2020-07-20:
Updated to 2.2.0b.  Tried it once, and started and remained running just fine. 

However, it did surface the ongoing intermittent problem I have had where it seems to compete with the iAqualink .  The stats on the iAqualink  and MQTT reports start going all over the place with Watts reporting in the 47000 range,  pump showing offline, and other odd issues until I shut down the aqualinkd service. Never figured out root cause of this, but generally resolves when I completely shut everything down and i sequence the startup of the iaqualink and AqualinkD  

Attached config and logs.  

[aqualinkd_log_conf.zip](https://github.com/sfeakes/AqualinkD/files/4945109/aqualinkd_log_conf.zip)

sfeakes, 2020-07-20:
@niharmehta There is a new option in v2.2.x to try to catch this kind or error.  It's not documented as I don't want people using it. Can you try setting `serial_readahead_b4_write = yes` in your config.  You'll get some nice errors stating "SERIOUS RS485 ERROR" if it catches anything, but I hope it may stop your equipment going funny.

niharmehta, 2020-08-11:
Still on 2.2.0b .  This weekend, I had to reboot my AqualinkD Pi. (Just dropped offline for some reasons likely unrelated to AqualinkD.   When it came back up, it started the intermittent problems I have been reporting with the iAqualink.   CHecking the logs showed no entries with "Errors", "ERRORS", "SERIOUS", "Serious" or related strings. I confirmed that the `serial_readahead_b4_write = yes` was set in the config.  After shutting down my RPi, and waiting a few minutes then plugging it back in, everything came up and remains fine. 
  So whatever problems or compatibility issues with the iAqualink, does not seem to be related to the errors that the command is supposed to catch.

closed 2025-11-18 by github-actions[bot]