aqualinkd/AqualinkD #115: v2.1.0 crashes after a few days of continuous web status commands

agprimatic, 2020-07-22:
I am using the web interface to check status once per second from a control system that emulates the curl command:

<code>curl http://upboard.local/api/status</code>

I turned on all debug logging. When the unit reaches steady-state, a cycle on the debugging log looks like this:

<pre><code>Jul-19-20 11:49:06 AM Info:   URI request: '/api/status^M
^M
^B'
Jul-19-20 11:49:06 AM Info:   Message body:
''
Jul-19-20 11:49:06 AM Debug:  API: URI Request 'status^M
^M
'
Jul-19-20 11:49:06 AM Debug:  Served WEB request
Jul-19-20 11:49:07 AM Info:   URI request: '/api/status^M
^M
'
Jul-19-20 11:49:07 AM Info:   Message body:
''
Jul-19-20 11:49:07 AM Debug:  API: URI Request 'status^M
^M
'
Jul-19-20 11:49:07 AM Debug:  Served WEB request
</code></pre>

As it runs, the Message body debug message keeps growing over time. It looks like the status requests aren't being cleared and they seem to bunch up. Also, the URI request ends in a 'G', which is maybe from a previous GET? The last cycle that was captured in the log before it crashed looked like this:

<pre><code>PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}GET /api/status

GET /api/status

'
Jul-22-20 00:00:02 AM Debug:  API: URI Request 'status

'
Jul-22-20 00:00:02 AM Debug:  Served WEB request
Jul-22-20 00:00:03 AM Info:   URI request: '/api/status

G'
Jul-22-20 00:00:03 AM Info:   Message body:
'GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}GET /api/status

GET /api/status

PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}'
Jul-22-20 00:00:03 AM Debug:  API: URI Request 'status

'
Jul-22-20 00:00:03 AM Debug:  Served WEB request
Jul-22-20 00:00:03 AM Info:   URI request: '/api/status

G'
Jul-22-20 00:00:03 AM Info:   Message body:
'GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}GET /api/status

GET /api/status

PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}GET /api/status

'
Jul-22-20 00:00:03 AM Debug:  API: URI Request 'status

'
Jul-22-20 00:00:03 AM Debug:  Served WEB request
Jul-22-20 00:00:03 AM Info:   URI request: '/api/status

G'
Jul-22-20 00:00:03 AM Info:   Message body:
'GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

GET /api/status

PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}GET /api/status

GET /api/status

PUT /api/Aux_5/set HTTP/1.1
Content-Length: 13

{"value":"0"}GET /api/status

GET /api/status

'
Jul-22-20 00:00:03 AM Debug:  API: URI Request 'status

'
Jul-22-20 00:00:03 AM Debug:  Served WEB request
Jul-22-20 00:00:03 AM Info:   URI request: '/api/status

G'</code></pre>

Once this happens, I can restart aqualinkd, and it works for a few days before crashing again.

sfeakes, 2020-07-22:
First thing, I'd urge you not to do a http request every second simply to get status.  If you want / need realtime info use MQTT not HTTP, HTTP isn't designed for that level of granularity, where MQTT is.

That being said, it shouldn't crash, but will create other issues for you on the Pi. Can you post the complete log file?   Some things in their are not making sense, so I'd like to see more of it.

agprimatic, 2020-07-22:
I understand your suggestion to run MQTT. I think I can run it on the same platform as aqualinkd. It was just easier for me to switch from the Serial Interface to HTTP requests in the short term.
I think I fixed my issue. I was doing a PUT and then immediately doing a GET for status. The mongoose web server didn't have time to respond to the first PUT and then was appending the GET to the same message. The buffer would eventually grow too big and then crash the application. I'm guessing this is a bug in mongoose.
By delaying the GET for 100ms, the issue seems to be resolved. I still have a lot of debug messages on (including the mongoose web server), so I'll let you know if there are any issues.

sfeakes, 2020-07-22:
I did find an issue, where I am returning more information on the /api/status than I should. BUT the content length is correct, so any good HTTP library will ignore the extra.  I've fixed that and will post update soon.

sfeakes, 2020-07-23:
Rather than posting HTTP requests every second, you could use a web socket if you don't want to go down the MQTT path just yet.  Websocket will just hold connection open and post updates too it.  Kind-a like a TCP socket connection.

Run the below and you'll get the status JSON posted every time something changes.
```
curl -i -N -H "Connection: Upgrade" -H "Upgrade: websocket" -H "Host: localhost" -H "Sec-WebSocket-Key: SGVsbG8sIHdvcmxkIQ==" -H "Sec-WebSocket-Version: 13" http://localhost:80/
```

agprimatic, 2020-07-23:
Thanks for the suggestion. Unfortunately, when I run the above command, I get continuous status output (not just on changes):

<pre><code>$ curl -i -N -H "Connection: Upgrade" -H "Upgrade: websocket" -H "Host: localhost" -H "Sec-WebSocket-Key: SGVsbG8sIHdvcmxkIQ==" -H "Sec-WebSocket-Version: 13" http://upboard.local:80/
HTTP/1.1 101 Switching Protocols
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Accept: qGEgH3En71di5rrssAZTmtRTyFk=

?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~
                              {"type": "status","status":"Ready","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}?~{"type": "status","status":"AQUAPURE OFF","version":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","s [... cut ...] sion":"8157 REV K","aqualinkd_version":"2.1.0","date":"07/23/20 THU","time":"9:30 AM","pool_htr_set_pnt":"85","spa_htr_set_pnt":"99","frz_protect_set_pnt":"38","air_temp":"74","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}^C
</code></pre>

sfeakes, 2020-07-23:
That's the expected result.   It will keep posting the status json to that socket.  It will slow down when things don't change, but due to how control panel works every time it sends AqualinkD a status, that will get posted, even if it's a duplicate. This may change in the future.  But there is WAY less overhead to this that posting a HTTP request every second.

sfeakes, 2020-07-23:
I've also had your first issue configured in postman and running for 24 hours now, and I can't duplicate it.

sfeakes, 2020-07-26:
After a few days, I can't duplicate this, using postman & AdvancedRestClient hitting server every second with duplicate requests. Are you sure your code is correct?

I've posted update 2.2.0c that fixes the issue of posting information passed the "content length" as highlighted in 2 posts above, but nothing in that release to address this specific problem.

closed 2020-07-29 by sfeakes