aqualinkd/AqualinkD #117: PDA Stuck in INIT phase "retrieving setpoints"

tomtokic, 2020-08-16:
Hey buddy, firstly let me say thank you for your efforts here.  This is super useful to have this written.  I had it working with a Pi 4 on a PDA control panel.  It was working fine on an older version of your code but I needed to rebuild the Pi recently and it now seems to have an issue.

From what I can gather it's looping in its initialisation phase when trying to get the setpoints.  

I have attached a debug as a starting point.  Let me know what else you might need.  I am technical with a moderate level of coding experience and can help on this side if needed.

Thanks in advance.

Tom
[Debug Log.txt](https://github.com/sfeakes/AqualinkD/files/5081527/Debug.Log.txt)

sfeakes, 2020-08-18:
I can see the issue, but need to see more information from the startup of aqualinkd before I can figure out what's happening.
Can you please set logging to serial debug and log to file in AqualinkD's config.  Then start Aqualinkd, wait a minute of so, then post the entire file.

tomtokic, 2020-08-18:
No worries. 
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/5093004/aqualinkd.log)
Here you go.

sfeakes, 2020-08-18:
I see the problem.  I *think* I've fixed it and it'll be posted in next release, but I have no way to actually test the fix works as I can't simulate a PDA panel for the freeze protect settings (and that's what's failing).  In the mean time it looks like pre-release 2.2.0 introduced the problem, so you should be able to pull any prior release and hopefully you won't see the problem.  Releases are here.
https://github.com/sfeakes/AqualinkD/releases

tomtokic, 2020-08-19:
Ok, awesome.  I'll try that.  I'm happy to do any testing for you if you want/need as well prior to releasing the next version.

tomtokic, 2020-08-20:
Hi Shaun,  I can confirm that I am up and running on v2.1.0.

sfeakes, 2020-08-20:
Thanks for the conformation.  I’ll let you know when I post the next update, so you can test that it’s fixed.

tomtokic, 2020-08-20:
Perfect.

sfeakes, 2020-08-28:
I've posted update V2.2.1.   Hope PDA is back working again.  Please try and report back when you get time.

tomtokic, 2020-08-30:
Yep, working fine now.  Thanks.

Did you happen to see my feature request on the forum "aqualinkd PDA always on setting".  It's very responsive in that mode with the browser always running, but would be great not to need to rely on the browser.

Cheers,

Tom

sfeakes, 2020-08-30:
Great, glad that fixed it. I must have missed that request.  I'll look for it and reply.  But `pda_sleep_mode = no` in your config is what you want.  But in that mode it means that the actual Jandy PDA won't work (or the two will conflict).  Stupid thing with what I've seen so far on PDA only panels is they only support one PDA device. Other panels support 4 of each type of device.  So it means that AqualinkD has to share that ID with the real PDA.

tomtokic, 2020-08-30:
Ahh, yes, my apologies.  You are right. I have turned off sleep mode and we are now working great.

Yes, it is annoying the way the PDA system can only have a single unit connected, but with Aqualinkd, I now have it integrated into my home automation system, and with voice control from an Apple Watch it renders the PDA unit "mostly" redundant anyway.

Thank you again, what you have done here with Aqualinkd is amazing!!!