aqualinkd/AqualinkD #118: Not Communicating with board

awshifts, 2020-08-24:
Good Afternoon, 

I had a quick question - and my apologies ahead of time as I am a bit new to this.   I ran through your setup, and even tried to reinstall it a few times, but I keep coming to the same issue.

I can get the AqualinkD web browser up but none of the commands work.  So I am assuming it's not communicating from the 
RS485 to the control panel. I have a snippet of a few items below, and one of the things Ive noticed is that the bunch of text never displays any info other than junk.

Any assistance would be greatly appreciated. 

`Aqualink

Error:  Unknown config parameter 'device_id=0x0'
Notice: Aqualink Daemon v2.1.0
Notice: Config log_level         = 7
Notice: Config device_id         = 0x0a
Notice: Config extra_device_id   = 0x00
Notice: Config serial_port       = /dev/ttyUSB0
Notice: Config rs_panel_size     = 8
Notice: Config socket_port       = 80
Notice: Config web_directory     = /var/www/aqualinkd/
Notice: Config extra_device_prog = NO
Notice: Config read_all_devices  = YES
Notice: Config use_aux_labels    = NO
Notice: Config override frz prot = NO
Notice: Config mqtt_server       = (null)
Notice: Config mqtt_dz_sub_topic = (null)
Notice: Config mqtt_dz_pub_topic = (null)
Notice: Config mqtt_aq_topic     = (null)
Notice: Config mqtt_user         = (null)
Notice: Config mqtt_passwd       = (null)
Notice: Config mqtt_ID           = aqualinkd_b827ebf8
Notice: Config idx water temp    = -999
Notice: Config idx pool temp     = -999
Notice: Config idx spa temp      = -999
Notice: Config idx SWG Percent   = 0
Notice: Config idx SWG PPM       = 0
Notice: Config PDA Mode          = NO
Notice: Config PDA Sleep Mode    = NO
Notice: Config force SWG         = NO
Notice: Config deamonize         = NO
Notice: Config log_file          = (null)
Notice: Config light_pgm_mode    = 0.00
Notice: Debug RS485 protocol     = NO
Notice: Read Pentair Packets     = NO
Notice: Display warnings in web  = NO
Notice: Ignore SWG 0 msg count   = 20
Notice: Config BTN Filter_Pump   = label Filter Pump     |
Notice: Config BTN Spa_Mode      = label Spa Mode        |
Notice: Config BTN Aux_1         = label Cleaner         |
Notice: Config BTN Aux_2         = label Waterfall       |
Notice: Config BTN Aux_3         = label Spa Blower      |
Notice: Config BTN Aux_4         = label Pool Light      |
Notice: Config BTN Aux_5         = label Spa Light       |
Notice: Config BTN Aux_6         = label NONE            |
Notice: Config BTN Aux_7         = label NONE            |
Notice: Config BTN Pool_Heater   = label Pool Heater     |
Notice: Config BTN Spa_Heater    = label Spa Heater      |
Notice: Config BTN Solar_Heater  = label Solar Heater    |
Notice: Starting web server on port 80
Notice: Starting MQTT client to (null)
Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Debug:  ++ Websocket joined
Debug:  WS: Message - Key 'uri' Value 'devices'nDebug:  WS: URI Request 'devices'
Debug:  WEB: homebridge used 2348 of 5120Debug:  ++ Websocket joined
Debug:  WS: Message - Key 'uri' Value 'devices'nDebug:  WS: URI Request 'devices'
Debug:  WEB: homebridge used 2348 of 5120Info:   URI request: '/ '
Debug:  Served WEB request

 sudo -s eval 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0'
^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^BM-^A^@M-^S^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P                                        ^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P                                        ^C^P^Bh^@z^P^C^P^BM-^@^@M-^R^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C                                        ^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^BM-^A^@M-^S^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P                                        ^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P                                        ^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^BM-^@^@M-^R^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C                                        ^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^BM-^A                                        ^@M-^S^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P                                        ^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^BM-^@^@M-^R^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C                                        ^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b                                        ^P^C^P^BX^@j^P^C^P^BM-^A^@M-^S^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P                                        ^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^BM-^@^@M-^R^P^C^P^B3^@E^P^C                                        ^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r                                        ^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^BM-^A^@M-^S^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P                                        ^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^BM-                                        ^@^@M-^R^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E                                        ^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^BM-^A^@M-^S^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P                                        ^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^                                        @E^P^C^P^BM-^@^@M-^R^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b                                        ^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^BM-^A^@M-^S^P^C^P^B`^@r^P                                        ^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^                                        @b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^BM-^@^@M-^R^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r                                        ^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^BM-^A^@M-^S^P                                        ^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^                                        @r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^BM-^@^@M-^R^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E                                        ^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B                                        `^@r^P^C^P^Bh^@z^P^C^P^BM-^A^@M-^S^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^                                        @E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^BM-^@^@M-^R^P^C^P^B8^@J^P^C^P^BP^@b                                        ^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B                                        3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^BM-^A^@M-^S^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^                                        @b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P^B3^@E^P^C^P^B8^@J^P^C^P^BP^@b^P^C^P^BX^@j^P^C^P^B`^@r^P^C^P^Bh^@z^P^C^P     `

awshifts, 2020-08-24:
Also If it Helps this is my board.
[](url)
[image]

awshifts, 2020-08-24:
Just in case this helps. A few images of the board.

[https://photos.app.goo.gl/du232FVuBeLivDCu5](url)

sfeakes, 2020-08-24:
First error (below) is the main issue.  you need to change that to something valid. So you need to edit the config.
Error: Unknown config parameter 'device_id=0x0'.  

But first, I think you need to run the serial_logger to get a valid id.  By the looks of the picture in the 2nd post, you have a PDA only board, so that ID will be probably 0x60. But run the serial logger to make sure.  Also if you do have a PDA only board, you should read the PDA notes in the wiki.

awshifts, 2020-08-24:
Wow that was a fast response.   I am checking now, I thought it was a PDA; but what confused me was the board diagram says its RS (in the google images) and well who knows. :)

sfeakes, 2020-08-24:
Jandy (and many pool equipment manufactures) seem to try to confuse people with their products & names...:).  But from the board (label on the chip), that's a PDA only board.   One thing to be aware of, the latest pre-release v2.2.0x has a bug with PDA mode, you're currently using v2.1.0 so you should be fine.  But don't upgrade until v2.2.1 has been posted.

awshifts, 2020-08-24:
Thanks - it figures, looks like the 600$ route lol

scroundy, 2020-08-24:
FYI - my board is close to the same - PS4 Combo and made to work with both
PDA and iAqualink. See below for a picture of mine. I've never configured
AqualinkD for PDA (although I tried once unsuccessfully).

AqualinkD works (almost without issues) on mine. Currently running 1.3.9c
without issue. But did have bouncing SWG settings on 2.2.x.

[image: image.png]

awshifts, 2020-08-24:
> FYI - my board is close to the same - PS4 Combo and made to work with both PDA and iAqualink. See below for a picture of mine. I've never configured AqualinkD for PDA (although I tried once [...]

for whatever reason, I cant see the image but I was thinking of Purchasing the JANDY R0468501 RS8 P&P CPU REPLACEMENT REV: R - AL8 HEX E0260700-D NEW   not sure if that's the right one so I am asking Jandy now

awshifts, 2020-08-24:
> FYI - my board is close to the same - PS4 Combo and made to work with both PDA and iAqualink. See below for a picture of mine. I've never configured AqualinkD for PDA (although I tried once [...]

Thanks I cant see the image but I am looking at getting this below - not sure if it works with my panel so I am checking with handy, but I am sure it will. 
JANDY R0468501 RS8 P&P CPU REPLACEMENT REV: R - AL8 HEX E0260700-D NEW

scroundy, 2020-08-24:
https://photos.app.goo.gl/v4s1nXcNYuCGPJ53A

[image]

sfeakes, 2020-08-24:
If you can make it work without PDA mode, I highly recommend going that route.  Again, serial_logger will tell you what ID's the panel is probing for, and if anything other than 0x60 pops up, use it.

awshifts, 2020-08-24:
@scroundy thanks - one last question as I am much closer (its starting to read temperatures - but does not control anything as of yet).   Did you have to remove the PDA to do it?

@sfeakes thanks I might just buy the CPU board for 299 - if you think its worth the switch.

sfeakes, 2020-08-24:
You have to be careful on the version / revision board you are buying.  $299 is **probably** an older rev# and while will work perfectly with AqualinkD, (better than PDA only) it may not support things like Variable Speed Pumps etc.
Again Jandy is not very helpful, but this document, while old is the latest official anything from Jandy that I can find that has this kind of information.
https://docs.google.com/viewer?url=https://github.com/sfeakes/AqualinkD/wiki/Jandy-Aqualink-RS-Revisions.pdf

awshifts, 2020-08-24:
@sfeakes  thanks again I will take a look at it - I got it all working under PDA mode for now - obviously im limited to choosing my color sequence with my lights but its working with basic on off. This leads me to my final question - if I make the switch to the RS board will I be able just to choose the light sequence I want? For instance, Mardi Gras is one of them will I be able to just select that as a mode?

sfeakes, 2020-08-25:
@awshifts Yes you should.  Again depending on panel support. AqualinkD can either use the panel to program the lights (if the panel supports the lights you have, and that will be version# specific), or it can bypass the panel completely and cycle the light to the appropiate light mode directly.  IT does take a little configuration and it's not the most intuitive since I try to support as much as possible.  But yes, you should be able to cycle any large manufacture (Jandy/pentair/haywood etc) LED or regular multicolor light mode.

scroundy, 2020-08-25:
You don't have to remove it. But we haven't used our pda for a very long
time and I can't remember if I disconnected it or not.

Looks like you got it working. That's great!

awshifts, 2020-08-27:
@sfeakes last question I promise :) ... if I change boards will it all of the equipment turns on instantly? With the PDA mode there is a delay but I read that, and with that Note, I will assume the state of each device changes almost instantly when using the RS board. 

I tend to have to toggle the switch twice in both the AauqlinkD web interface and Hubitat - so that alone would drive me to update CPUs ... call me crazy but its the little things.

sfeakes, 2020-08-27:
Yes turning anything on/off is instant with all RS board.  Changing heater set points / SWG % / VSP speed can take some time though.  The amount of time will depend on what protocols the board supports.

awshifts, 2020-10-01:
Okay so I changed my bored.   Everything is working like a charm.  I did have one follow up question on the lights.  I was not understanding correctly in the light section of the read me.    I have the colored lights set to be controlled by the remote, but what if i just want them to be controlled buy aqualinkD,  that's the config part im trying to figure out now.   Say I have 17 light modes, how does one config those.

sfeakes, 2020-10-02:
First you must tell the panel it's a normal light if you want Aqualinkd to to the programming.
Then configure the option of how to control the light in /etc/aqualinkd.conf
Then configure the actual mode names (for the Web UI) in /var/www/aqualinkd/config.js

If the panel supports the particular colored light, then you can also let the panel do the programming, and configure aqualinkd as such. (then all Jandy keypads can also change the mode), aqualinkd can still change to the particular mode/color you want.  Reason to use AqualinkD for programming is it supports any light and Jandy panel only supports a few.

awshifts, 2020-10-06:
@sfeakes thanks Ill give it a try and see if I can figure it out :).   I have a supported light set, but prefer to do it via mobile or tablet devices, or google.

closed 2025-11-18 by github-actions[bot]