aqualinkd/AqualinkD #119: Filter Pump status of "flash"

mdetweiler, 2020-09-06:
I'm sure this is not an AqualinkD issue, but I ran into it after having an odd series of events.  

We had a power blip and my wired Jandy controller started showing an image that had mostly dots across the screen with a firmware message across the top only.  All indications online were that there was probably a loose wire so i assumed I caused this when I set up AqualinkD a few weeks ago.  After a bunch of debugging I found that it was resolved by simply restarting the raspberry pi that was running AqualinkD.  Interestingly, AqualinkD worked fine the whole time.  The only lingering issue is that my filter pump returns a status of "flash" now and I don't know what that means or how to fix it.  The pump seems to work fine and AqualinkD does not seem to care, but it is at least an annoyance to my other automation.  Does anyone how to fix it?

Thanks

sfeakes, 2020-09-06:
Flash means the pump is enabled but not on. ie pool/spa JVA is changing between the two and the panel has turned the pump of while the JVA is turning and will turn the pump back on after the JVA has switched modes.

Are you seeing flash on the AqualinkD webUI or in your home automation hub?

This flash status is what the panel is telling AqualinkD, so you should also see that mode in the Jandy panel.  If it's in your home automation hub ONLY and not AqualinkD webUI then your home hub is out of sync, if it's in AqualinkD webUI you probably have a bigger issue.
I'd restart AqualinkD in either case and see what happens.

mdetweiler, 2020-09-06:
Ok..  well I upgraded and restarted and all looks good.  Next time I will try a restart of services before bugging you.  Hopefully that was just a bump in the road.  Thanks for the reply and explanation as always!

sfeakes, 2020-09-06:
Glad it's sorted itself out.

closed 2020-09-06 by sfeakes