aqualinkd/AqualinkD #121: config.js changes don't seem to change

mdetweiler, 2020-09-07:
I've tried to customize my interface by following the instructions here https://github.com/sfeakes/AqualinkD/wiki#aqualinkd-web-ui-configuration, but I do not see any changes in the UI.  I've made changes in the /var/www/aqualink directory as well as the source to see if that helped, but nothing changes after a restart of the service or reboot of the system?

sfeakes, 2020-09-07:
AqualinkD tells the web browser to use Very heavy caching, my guess is the browser is using a cached version for config.js.  So delete browser cache and/or put browser into developer mode (so it doesn’t cache) while your playing with that file.

I’ll make a note of that the wiki, I should have noted that.

mdetweiler, 2020-09-07:
Yup...  that worked!  Sorry for the alarm.   Interesting that the caching does not affect the other dynamic data, only the layout.

sfeakes, 2020-09-07:
Dynamic content is sent over a websockets, so not your normal http connection (which is why updates are instant and not polled). And browser can’t cache websockets data. So that’s the main reason why.  But also Anything on the file system (/var/www/AqualinkD), AqualinkD tells browser to cache heavily anything its creates on the fly It tells the browser not to cache.

closed 2025-11-18 by github-actions[bot]