aqualinkd/AqualinkD #126: After upgrading to 2.2.1, one light gets stuck on

adamsinger, 2020-09-17:
After upgrading to 2.2.1, everything seemed to be working correctly, except one weird issue.

The light for my water feature gets stuck on. When I turn it off via the Home app, the app says it's off, and even the control panel will say it's off, but it's not off. Then, the only way to actually turn it off is to run the ALL OFF command on the control panel or iAquaLink. It seems to only affect this one light for some reason.

When I disconnect my Raspberry Pi from the control panel, the light functions as normal.

I've attached logs as I turn the light on, and then attempt to turn it off again via the Home app:

```
Sep-16-20 19:03:35 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:35 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:35 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:36 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:36 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:36 PM Debug:   NetService:MQTT: topic aqualinkd/Aux_4/set 1
Sep-16-20 19:03:36 PM Debug:   NetService:MQTT: URI Request 'Aux_4/set': value 1.00
Sep-16-20 19:03:36 PM Info:    NetService:MQTT: received 'ON' for 'Aux_4', turning 'ON'
Sep-16-20 19:03:36 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:36 PM Debug:   AqualinkD: RS received packet of type PDA Clear length 9
Sep-16-20 19:03:36 PM Info:    RS Allbtn: RS Received unknown packet, 0x09
Sep-16-20 19:03:36 PM Debug:   AqualinkD: RS received packet of type Message length 24
Sep-16-20 19:03:36 PM Info:    RS Allbtn: RS Message :- 'Fountn Light ON'
Sep-16-20 19:03:37 PM Info:    NetService:MQTT: Published id=100: aqualinkd/Display_Message Fountn Light ON
Sep-16-20 19:03:37 PM Debug:   NetService:MQTT: Message publishing acknowledged (msg_id: 100)
Sep-16-20 19:03:37 PM Debug:   NetService:MQTT: received (msg_id: 100), looks like my own message, ignoring
Sep-16-20 19:03:37 PM Debug:   NetService:MQTT: Ignore aqualinkd/Display_Message Fountn Light ON
Sep-16-20 19:03:37 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:37 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:38 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:38 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:38 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:38 PM Debug:   AqualinkD: RS received packet of type Message length 24
Sep-16-20 19:03:38 PM Info:    RS Allbtn: RS Message :- '09/16/20 Wed'
Sep-16-20 19:03:39 PM Debug:   NetService:MQTT: topic aqualinkd/Aux_4/set 0
Sep-16-20 19:03:39 PM Debug:   NetService:MQTT: URI Request 'Aux_4/set': value 0.00
Sep-16-20 19:03:39 PM Info:    NetService:MQTT: received 'OFF' for 'Aux_4', already 'OFF', Ignoring
Sep-16-20 19:03:39 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:39 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:39 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:39 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:40 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:40 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:40 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:40 PM Debug:   AqualinkD: RS received packet of type Message length 24
Sep-16-20 19:03:40 PM Info:    RS Allbtn: RS Message :- '7:03 PM'
Sep-16-20 19:03:40 PM Debug:   AqualinkD: time not checked, will check in 994 seconds
Sep-16-20 19:03:40 PM Debug:   RS Allbtn: RS time is accurate '7:03 PM 09/16/20 Wed'
Sep-16-20 19:03:40 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:40 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:41 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:41 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:41 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:41 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:41 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:42 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:42 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:42 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:42 PM Debug:   AqualinkD: RS received packet of type Message length 24
Sep-16-20 19:03:42 PM Info:    RS Allbtn: RS Message :- 'Air Temp 80 F'
Sep-16-20 19:03:42 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:43 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:43 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:43 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:44 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:44 PM Debug:   AqualinkD: RS received packet of type Message length 24
Sep-16-20 19:03:44 PM Info:    RS Allbtn: RS Message :- 'Pump OFF'
Sep-16-20 19:03:44 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:44 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:45 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:03:45 PM Debug:   AqualinkD: RS received packet of type Status length 12
[... 85 lines cut ...]
Sep-16-20 19:04:00 PM Info:    RS Allbtn: RS Message :- 'Air Temp 80 F'
Sep-16-20 19:04:01 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:01 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:01 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:01 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:01 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:02 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:02 PM Info:    RS SAdptr: Queue device update requests
Sep-16-20 19:04:02 PM Debug:   RS SAdptr: Added to message queue, position 0 0x00|0x01|0x05|0x05
Sep-16-20 19:04:02 PM Debug:   RS SAdptr: Added to message queue, position 1 0x00|0x01|0x07|0x05
Sep-16-20 19:04:02 PM Debug:   RS SAdptr: Pull from message queue, length 2 0x00|0x01|0x05|0x05
Sep-16-20 19:04:02 PM Debug:   RS SAdptr: Remove from message queue, length 1
Sep-16-20 19:04:02 PM Info:    RS SAdptr: Pool SP 90
Sep-16-20 19:04:02 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:02 PM Debug:   AqualinkD: RS received packet of type Message length 24
Sep-16-20 19:04:02 PM Info:    RS Allbtn: RS Message :- 'Pump OFF'
Sep-16-20 19:04:02 PM Debug:   RS SAdptr: Pull from message queue, length 1 0x00|0x01|0x07|0x05
Sep-16-20 19:04:02 PM Debug:   RS SAdptr: Remove from message queue, length 0
Sep-16-20 19:04:02 PM Info:    RS SAdptr: Spa SP 102
Sep-16-20 19:04:02 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:03 PM Debug:   AqualinkD: RS received packet of type Status length 12
Sep-16-20 19:04:03 PM Debug:   AqualinkD: RS received packet of type Status length 12
```

Also FWIW, I'm the one with the setup that was in this issue: https://github.com/sfeakes/AqualinkD/issues/111

sfeakes, 2020-09-17:
Is it configured as a programmable light?   If so can you comment out **_rssa_device_id=_** in the config, restart AqualinkD. just to see if it works correctly again?  I realize that commenting that out will slow down the setpoints, but I just like to see if the problem go away.  If it’s not configured as programmable light then that setting will have no effect as to how AqualinkD controls the light and it’s something else.

adamsinger, 2020-09-17:
I'm not sure what you mean by "configured as a programmable light". If you mean having a `button_xx_lightMode` set, then no it is not configured as a programmable light. I'm not sure if it needs to be? It's just a simple on-off light with no color modes or anything.

adamsinger, 2020-09-17:
I commented out the `rssa_device_id=` setting, and it did indeed fix the problem with the light