aqualinkd/AqualinkD #130: Background URL is not working from Camera feed

vziukas, 2020-10-02:
Hi,

I got latest AqualinkD version working. Thanks for such great work !

I tried to update background_url to fetch my backyard camera snaps but can not get it working. 

Here is documentation which says how to get URL. I see image if I paste that URL to browser, but I do not see it as background on AqualinkD. Just gray background. 

https://support.reolink.com/hc/en-us/articles/360007011233-How-to-Capture-Live-JPEG-Image-of-Reolink-Cameras-via-Web-Browsers

Example URL would be:
http://192.168.2.119/cgi-bin/api.cgi?cmd=Snap&channel=0&rs=wuuPhkmUCeI9WG7C&user=admin&password=123456

Thanks for any tip !

sfeakes, 2020-10-02:
Just checking.
`/var/www/aqualinkd/config.js` should have the following in it.
```
var background_url = "http://192.168.2.119/cgi-bin/api.cgi?cmd=Snap&channel=0&rs=wuuPhkmUCeI9WG7C&user=admin&password=123456";
```

But I *think* I see the problem.   Can you try posting this into your browser and see what happens
```
http://192.168.2.119/cgi-bin/api.cgi?cmd=Snap&channel=0&rs=wuuPhkmUCeI9WG7C&user=admin&password=123456?1601648880263
```
If you get the same gray image, then I know what's happening.

vziukas, 2020-10-02:
I find out that the code tries to append time stamp in form as `?`
but URL already has `?` in it. 
I make a change to check for existence of `?` in the url and use `&` in that case to append time stamp and it works! I can issue PR if you want.

sfeakes, 2020-10-02:
Thanks, I just did the same when I saw the code.

closed 2020-10-02 by sfeakes