aqualinkd/AqualinkD #132: Wifi Interface

parchmaf, 2020-10-23:
My system is an RS8 that has the wifi interface.  Is it possible to use aquaklinkd with the IP address of the iQ20-RS?

sfeakes, 2020-10-23:
No is it not.  The iQ20RS is a very basic (and restricted) HTTP interface to your control panel.  I believe there are other tools out their that will link the IQ20RS to things like home automation hubs etc.

parchmaf, 2020-10-24:
Thanks for responding.  The only thing I care about is getting it connected to HomeKit.  Any suggestions?

Fred Parchman
fred@parchman.net

sfeakes, 2020-10-24:
Due to the limitations of the iQ20RS I doubt there are any good HomeKit implications, but I’ve seen some interfaces to that unit during searching, but can’t remember the names of them now.

niharmehta, 2020-10-29:
There are a couple of projects utilizing an API of some sort from iAqualink/Jandy.  I think HomeAssistant has integrated this already as on option. Its possible that can be fed in to HomeBridge to make available to HomeKit.

parchmaf, 2020-10-29:
Thanks.  I don’t know much about home assistants. Can it run on a Mac like homebridge?  Aqualinkd looks great on my he Apple home app.  Would love to get that going somehow if possible.

Will you keep developing this?  Let me know if I can help in some way, maybe as a tester or something.   Thanks

Fred Parchman
fred@parchman.net

closed 2025-11-20 by github-actions[bot]