aqualinkd/AqualinkD #140: Unable to set timers with MQTT or API

shbatm, 2020-12-09:
Following the guidelines in the Wiki, you should be able to do the following:

```
Timers can be set on anything with on/off
/api/Filter_Pump/timer/set    (value in minutes, will turn device on wait x min turn device off)
```

However, when I publish a message on those topics, I get the following error in the log:

```shell
Dec  8 17:57:04 aqualink-rps user.warn aqualinkd: NetService:MQTT: ignoring, unknown URI Aux_1/timer/set
Dec  8 18:04:15 aqualink-rps user.warn aqualinkd: NetService:MQTT: ignoring, unknown URI Filter_Pump/timer/set
```

Searching through the source, I can't find where it accepts "timer" as a URI for either the MQTT or web API. Was this a feature that was never implemented?

sfeakes, 2020-12-09:
Sorry about that, the timer code has not been released yet.  I was forced to push a document update before the code update, due to some BS. The code that supports timers will be in the next release once I finish the testing.

shbatm, 2020-12-09:
No problem. I was thinking I was missing something (or losing my mind, very possible). Looking forward to the new release!