aqualinkd/AqualinkD #142: Cant get the Air Blower Assigned...

drevilb, 2021-04-20:
Hi!  Any help would be appreciated.  I have tried so many things...but I cannot get the Air Blower for the spa to work at all.  On the one touch panel it says its the RS4 Combo.  When I set that, everythign else seems to work fine...  but there is no mapping for the Spa Blower at all... It is on the Extra Aux chanel I believe (At least thats what it says on the panel).  When I turn on the air blower, there is no button that lights up in the Web display showing its mapped to something different.  I changed the conf to 16 and showed all the buttons... turned on the air blower on the outside control panel, and nothing "lights up
" or activates on the Web display indicating it being mapped to something....

Any help in what I need to be doing to figure this out?

It looks like it is mapped to, what on some other units is the SOLAR button... does that make a difference?

[image]

sfeakes, 2021-04-21:
The solar is the key here.   The web UI has solar turned off (Ie don’t display) by default.  If you edit config.php as per instructions in wiki to edit web ui settings, you should see the solar button show up.  Then if you turn it on / off via keypad, I’m hoping you’ll see the solar key light up in the web ui.  Then just rename it to spa blower.

drevilb, 2021-04-21:
Thanks.  I did that but it appears the .js file isnt reloading. I cleared cache and still pulling the old file.  Going to reboot the RasPi and see if that does anything.

sfeakes, 2021-04-21:
Make sure you edited the correct file.  In /var/www/aqualinkd,  not the one in /home.

drevilb, 2021-04-21:
GREAT SUCCESS!  I have another question, but will post it in a new issue so I dont clutter this thread.   THANK YOU... THIS IS AMAZING.  wOOt!

closed 2021-04-21 by drevilb