aqualinkd/AqualinkD #143: Password Protect the Web UI .htaccess?

drevilb, 2021-04-21:
Is it possible to do a basic site authentication using .htaccess?  Not sure what web server its running for configurations.  Sorry if this is a really basic question.  Just wanting to allow the page access outside of my network and dont want to leave it unprotected is all.

sfeakes, 2021-04-22:
There is no security, and .htaccess files will be ignored.  You could add very very basic security to AqualinkD, but it would require code changes, and quite frankly would never be all that great / secure.  This is a limitation of a library I use called mongoose. If you do want security, then I would strong suggest installing a real web server, like Apache or nginx and using that to proxy the requests to AqualinkD, and put the security in the real web server. There is no way you can beat/compete with the security those services will offer.

drevilb, 2021-04-22:
Thank you!  I sent you a paypal for a coffee.  I appreciate your help.

closed 2021-04-22 by drevilb

JimO67, 2023-05-12:
Brian,

Did you ever implement Apache or other so you could have secure access?
"proxy the requests" with Apache.  

Thanks

drevilb, 2023-05-12:
No I never did.  Went down that road but had some issues that I didn’t have time to figure out.  I just keep it in my local net, so I guess it didn’t matter.  If you figure it out, would love to know!- Brian

sfeakes, 2023-05-12:
There are multiple ways to do this.  Example VirtualHost config using specific location for Apache would look something like this
```
       <Location /AqualinkD>
              ProxyPass http://ip.of.aqualinkd:80
              Allow from all
               AuthName "Secure Area"
               AuthUserFile /etc/apache2/passwd
               AuthGroupFile /etc/apache2/group
               AuthType Basic
               require group browse-users
       </location>
```

example VirtualHost at root level would be something like
```
    <VirtualHost *:443>
        ServerAdmin webmaster@localhost
        ServerName <my domain name>
        ServerAlias <my domain name alias>

        ProxyPass http://ip.of.aqualinkd:80

        <ADD SECURITY SETTINGS>

        SSLEngine on
        SSLCertificateFile /etc/letsencrypt/live/<my domain name>/cert.pem
        SSLCertificateKeyFile /etc/letsencrypt/live/<my domain name>/privkey.pem
        SSLCertificateChainFile /etc/letsencrypt/live/<my domain name>/fullchain.pem

    </VirtualHost>
```

Above are simple examples.  There are many ways to do this all well documented on the web.

JimO67, 2023-05-12:
Simple things are difficult when you don’t even have the vocabulary to discuss it.  “How to set up a proxy server to virtual host incoming web requests through a password protected server”?   I know what I want but don’t know how to say it.