aqualinkd/AqualinkD #144: Raspi inactive disconnect

drevilb, 2021-05-02:
Hi. Have a raspi 3 built in Wi-Fi. Everything works fine… until it sits for like 30-45 minutes Unaccessed.  Then when I try to access it via web or ssh it times out…. If I keep trying it then comes back after a minute or 3.  Then works fine unless I let it sit in active. 

I disabled power saving mode in the Wi-Fi.  I setup a ping in from every 5 minutes.  I also tried stopping and starting the aqua link daemon every 10 minutes. None has helped yet. 

Are there any timeouts in the daemon that I need to set? Or any other ideas?  Thank you so much!!

sfeakes, 2021-05-03:
There are no time out's / sleep set in the daemon. With all the other installs out their, this is the first time I've heard about this issue. So this is either an OS issue with what ever distribution you are using, or something on your WiFi access point / router.  If you have another machine that's up 24/7, leave ping up and running on the raspberry 3, so it's pinging out to some other system.  See if that helps. BTW, you can slow ping down if you are worried about that, `ping -I 5 machine` will only do a ping every 5 seconds.  BTW, a ping into the raspberry will not keep a connection up if the wifi is sleeping, only a ping out will do that.

drevilb, 2021-05-05:
Ok.   Perhaps I am looking in the wrong spot...  I noticed that I can pull up the interface on my PC.  Then when I pull it up on my phone, it times out and cannnot connect.  If I try for a few minutes, it will finally respond, on the phone.  Its like its timing out on the phone after a bit, but not on the PC.  

Not exactly sure where to go with that.  Also happens on my wifes phone (Both IOS).  At first I was thinking it was only allowing one browser at a time, but when it comes up on the phone, I can still use it on the PC (Using Chrome).  It's almost as if there is a filtering going on, and it takes a bit for the filter to allow the phone devices.  I have turned off all firewall related items to the Pi, but it still is acting weird. 

It obviously appears to be some sort of networking issue... but Im at a loss now.  

Thank you.

drevilb, 2021-05-05:
Ok. Its not just mobile.  It seems whenever I use a device that hasn't connected to the raspi in some amount of time...  Seems like the phone its about 30-45 minutes.  And my laptop I connected yesterday, and had issues today.  I get this message for about 2-3 minutes.  

This site can’t be reached192.168.0.98 took too long to respond.
Try:

Checking the connection
Checking the proxy and the firewall
Running Windows Network Diagnostics
ERR_CONNECTION_TIMED_OUT

I can still access it on other devices at this time, so it is up and running.  Something is weird that it seems like it has to register the device for some amount of time.. then it responds fine.  

Perhaps I should try installing apache and proxying the requests?  

Im a little stumped.

sfeakes, 2021-05-05:
Do you have other web resources on your network that you can check / ping?  I’d assume it was your local dns or mdns, but the above shows you are using IP, not machine name.  Are you sure your firewall / router isn’t causing some problems?  iOS recently introduced this BS where it changes the MAC address on the fly, and that’s created a lot of issues for home networks. (But you said it’s not just mobile, so it’s probably not that). But that’s “private WiFi address” setting on iOS device, you should turn that off for your home network.
It kinda sounds like your router / ap has some form of ip filtering going on.  Can you describe your network topology?

drevilb, 2021-05-06:
Thank you.

 

Yes. I can connect to other resources no issues.

I already turned off the private address garbage

 

I have an ASUS WRT5200 Wifi router (That is just acting as an access point), and I have a Firewalla gold (https://firewalla.com/products/firewalla-gold) router that is connected to a cable modem for internet.  I turned off the firewall for the mobile device and the raspi, but it still experiences the issues.  Its almost like it is taking time to spin up another “thread” or something.  

 

Not too sure..Im sorta at a loss of what else to check.  Perhaps it’s the raspi itself… 

 

Brian

drevilb, 2021-06-08:
Ok. FINALLY got it figured out.  So..Just in case others have this issue.

Behavior:
- Raspi attached to WiFi
- Seen on network fine.  
- connect from a different computer and it cant see it.  
- Try and connect a few more times... after 2 - 5 minutes, its visible again until there is about 30 minutes of inactivity.  But on other machines that are connected, you can still see it and connect.

What I tried and DIDNT work:
- Power Save to NO for the wifi.  
- Adding a Ping to Cron
- Turnging off firewall completely
- Tried a different Raspi
-
What the issue REALLY was:
- On the Router, There is a "professional" setting for ROAMING and handoff.  I changed the Channel to a specific number as opposed to AUTO, INCREASED the becon interval from 100 to 50, and turned off Roaming assistance (I only have one AP so not needed).

AND BOOM.  Snappy and always on.  Works like a charm now.  

Hope this helps someone else.

Thanks!

closed 2021-06-08 by drevilb