aqualinkd/AqualinkD #147: System datetime is ignored and set to UTC

igorek24, 2021-05-26:
System datetime is ignored by AqualinD (on Debian) and if keep_paneltime_synced set to yes, it sets RS panel to UTC time.

sfeakes, 2021-05-27:
Have you set the time zone correctly?   It uses system time, so if you don’t have that set correctly in the OS it defaults to UTC.

igorek24, 2021-05-27:
System time and zone is set correctly and confirmed with timedatectl or date.

sfeakes, 2021-05-27:
Make sure you have it accurate for the user running AqualinkD process.
https://community.intersystems.com/post/setting-tz-environment-variable-linux
AqualinkD uses a gnu localtime() function to read time, that depends on the systems TZ setting.

since you compiled this yourself, did you modify the Makefile?

igorek24, 2021-05-27:
Everything seems to be correct, here is the output from the terminal while running as root and as you can see, it's way off ( not even UTC time:
`# timedatectl status
               Local time: Wed 2021-05-26 23:53:10 PDT
           Universal time: Thu 2021-05-27 06:53:10 UTC
                 RTC time: Thu 2021-05-27 06:53:11
                Time zone: America/Los_Angeles (PDT, -0700)
System clock synchronized: yes
              NTP service: inactive
          RTC in local TZ: no

# hwclock -r
2021-05-26 23:53:17.388520-07:00

# ./aqualinkd -d -c /etc/aqualinkd.conf

Notice:  Panl Prog: Programming: Get iAqualink Touch Setpoints, 31
Notice:  RS Allbtn: RS time is NOT accurate '1:46 AM 05/14/21 Fri', re-setting on controller!
Notice:  iAQ Touch: IAQ Touch Freeze Protection setpoint 38
Notice:  Panl Prog: Programming: Set iAqualink Set Time, 37
Notice:  iAQ Touch: Set date to 05/26/21
Notice:  iAQ Touch: Set time to 11:54`

igorek24, 2021-05-27:
This can be closed. After re-applying the TZ settings and rebooting the server (Pine64 LTS) time is showing correctly now.

closed 2021-05-27 by igorek24