aqualinkd/AqualinkD #152: serial_logger not running on PC install of Aqualink

john1158way, 2021-11-08:
First let me say thanks again for your work in progressing AqualinkD.    My Raspberry Pi was knocked out in a lightning storm and I decided to reinstall on an unused laptop I happened to have.   I loaded up the latest version of the Raspberry Pi OS that you can use for an x86  PC and I installed AqualinkD.    I used the Make option to install and while I received several warning messages all seems to work fine.      Except I cannot get it to run the serial_logger program.   I suspect it is because it is compiled for an ARM architecture and even though I am running a version of Raspberry Pi OS it is a different architecture.   I looked for source code to compile a version of serial logger on my PC to see if that would solve the problem but it was not obvious that it was on GIT Hub.   If you could point me to the source code or provide an insight to how to work around the problem I would be grateful.

sfeakes, 2021-11-08:
Raspberry Pi OS has some very specific USB modifications to overcome the limitations on the Broadcom usb chip in the Pi 1,2 & 3. (and probably Pi 4 soon). So using that as a base distribution on anything other that a Raspberry doesn't surprise me that there are some serial/usb issues.  AqualinkD can easily be compiled on any Linux distribution, so you'll probably be better off starting with a base Debain distribution on your PC.  Other distributions are just as good, but Debain is the base of Raspbain (now Raspberry OS) so you may be more familiar with it.

First thing I would do is spend more time looking at why your VSP can't be controlled by your panel, as you might find that the simplest option.  
First find your Control Panel version#, then search the web for "jandy RS software and PDB versions pdf" that should get you a PDF listing the versions and what's supported.  Then if your Control Panel supports VSP, see what protocol, (Jandy only or Jandy & Pentair).  if it does support VSP, but only ePump (ie Jandy), and your Harris VSP only uses Pentair, then you can buy a converter box to match the protocols up (about $60).

As for writing something yourself, on the pump, that 4 wire cable is RS485 connection and you are correct, you should be able to control that via computer with a USB2RS485 converter.  Few things to remember on the RS485 protocol, there can only be one master on the physical network and that has multiple slaves, the slaves can ONLY do bi-directional communication to the master not each other. The physical control panel has to be the master, and everything else is a slave (including AqualinkD). So if the control panel is NOT communicating with your pump correctly, you HAVE to have a separate physical RS485 network running for this to work, as you will need to make the software work as a master. 

I created a fork of Aqualinkd many years ago to do this for SWG, (Called AquapureD) so you could also look at that code.  The Jandy & Pentair RS485 protocols are not difficult to understand (and your pump will use one of those) and the serial logger will tell you all the information you need to know.  I did have a lot of documentation on these protocols, but had to remove it, but following the code posted in GitHub will show everything as well.

Option 2 will lead you down a really difficult path, but first how do you intend on reading the display of the VSP? Using the pi to close circuits (pressing buttons) is easy, but reading the display would be the stumbling block.

john1158way, 2021-11-08:
thank you for the rapid response.    I was not assuming that the pump had a 485 bus and would use a protocol from one of the major manufacturers that the serial logger could be used on it.   I will connect to that and revert back.  to be clear the small control panel mounted on the motor does not have any connectors other than the one plug that goes to the motor's variable speed package.      I was assuming they used a 2 device serial connection with a proprietary protocol and not a 485 bus with a known protocol.   

From systemctl status I get that the Aqualink control panel is version 6521 REV F  the manual only covers 2 speed motors with no mention of variable speed,  I will do the searches you suggest.

thanks again
John

sfeakes, 2021-11-08:
Rev O & P are where VSP pumps start to get real support from the Jandy Control Panel.  So you could consider a chip upgrade to get you panel to support it, as with rev F you won't be able to control any VSP from your panel.  2 speed pump control is a very different type of control than VSP, so I don't think you will get far on that option.

I would be surprised if that interface is not RS485, (but can't say for sure as I don't have a Harris VSP).  It's always a crap shoot talking to pool manufactures (and even dealers/distributers), but you might want to ask them "how to implement remote control" if the manual doesn't mention it.  I did a few quick web searches, but didn't get much info on the Harris VSP's.

john1158way, 2021-11-09:
thanks for the help.   

I hooked up the pump control panel to the 485 adapter.      the serial logger would start but not complete.   I also started aqualinkd with log level serial debugger.   The log stopped after boot saying the adapter blocked attributes.      I ran the command sudo -s eval 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0' > serial_feed to capture the traffic and it did show the adapter was reading characters but no words or anything what you get mixed in running that command when hooked up to the Jandy control card.   what I was getting was   basically a repeating pattern of ^@M-+^@^A^@^B^@^A^B^@^B^@M-+^@M-+^@M-+^@M-+^@^A^@^B^B^@^A^@^B^@^A^B^@^B^A^@M-+^@M-+^@^@^B^@^BM-+^@M-+^AM-+^@^@M-+^@^@^B^@^A^@^BM-+^@^A^.      This was with the pump control board powered up and not getting any response from the motor.       

As a next step I am going to create access to the Tx and Rx connections when the pump control panel is back connected to the motor and see if once the motor is responding the stream of characters makes any more sense.   I will repeat patterns of turning the pump off and on and changing speeds to see if I can find patterns.  

If that does not do it I will try a direct serial connection  without the adapter  and repeat the exercise.   

thanks again

PS I had already asked the manufacturer and they  said the pump did not support any remote control and were unwilling to share protocols or design info.

tomtokic, 2021-11-09:
Are you running the right baud rate?

john1158way, 2021-11-09:
I have not considered baud rate yet. thank you for raising it as a variable to consider.  I  am only running the default baud rate on the adapter as it picked up the Jandy signals correctly out of the box.  I do not have access to an oscilloscope to look at the signal but may investigate changing the rate on the 485 adapter and seeing if it changes anything.   If there are any programs that cycle through baud rate settings on adapters and look for valid character strings that would be helpful.

tomtokic, 2021-11-09:
No worries.  When Inhave seen that sort of stuff in the past it has generally been when the baud rate was different.   It seems like you have comms but just not the right parameters set.

sfeakes, 2021-11-09:
Interesting what the manufacturer told you since their documentation clearly talks about remote control and the serial wiring, but I could not find real details on that.
The serial logger is designed around the Jandy & Pentair protocols and uses their termination packets as I have never come across a piece of RS485 pool equipment that doesn’t use one of those protocols. There is a raw mode that might help, but most equipment won’t really start to talk until it gets a probe signal from the master, and since you don’t have any master in this case (pump and computer only connected), you may have to start sending out probe signals  to every address (and both protocol) and see if the VSP responds.
There are some other options, but would require some additional hardware, I’ll have a think and see if there are other options. BTW how many USB2RS485 adapters do you have?

john1158way, 2021-11-09:
a few thoughts - 

How do i run serial_logger in raw mode?   

 to the comment I do not have a master,  The only electrical connection between the control panel on the motor and the variable speed drive package on the motor is the 4 wire connection.   So if the 4 wire is power and a 485 bus then wouldn't one of the two parts have to be a master for the pump control panel to control the motor?

in this case wouldn't the control panel on the motor be the most likely choice for the master?

I have 2 USB to RS 485 adapters.   

with the labeling of the 4 wire adapter Tx and Rx I still wonder if it is straight serial and not a 485 bus

If it was straight serial would my 485 adapter still read it?  Does the fact that I do get characters mean it is 485?

The characters I get with monitoring the raw feed from the adapter 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0' are always a consistent pattern  do you think this implies the baud rate is ok as mismatched baud rates would cause inconsistent dropping of bits?

my Harris pump is a few years old and it has been two years since my interaction on remote control.   Perhaps they have added to their new models.

thanks again for support

john1158way, 2021-11-19:
just an update for anyone interested.   I have done a few things and made some progress but it is clear this is a learning experience for me and while I may gain remote control of my pump in the end it will be a process.  Perhaps useful to others wanting to integrate odd equipment to AqualinkD

I removed the control board completely from its casing and found connection holes marked V, Gnd, A, B.  and took a guess this might be a 485 connection.    So I hooked up my adapter to those pin outs    I did not get any response and the no leds on the pump local control board lit up.   i reverted to the hook up I had used successfully before using the V Gnd Tx and Rx pins and had the previous behavior, I could receive data echoing the dev/ttyUSB0 

I then creating a harness that would allow me to access the V, Gnd, Tx & Rx connections while the local pump control panel was connected to the pump motor.  I then connected my 485 adapter to the Tx & Rx connections but did not connect V or Gnd as power was supplied to the local control board from the motor.    With my adapter connected to Tx and Rx  the pump operated normally but I did not get any output.  I added the ground connections but still no data.   Using a meter I found that the motor only supplied the local pump control with circa 2V  and suspect the pin voltage when the devices are connected is too low to drive the adapter.   It is possible that the load of the volt meter is depressing the voltage and the board is operating at 3.3v  Regretfully I did not check the voltage of the connection with the A, B, designation when I had the access.  When I was accessing the local control board by itself I was powering it with 5v from the adapter.   I am fortunate it seems no harm done.

I have purchased an inexpensive ($10) logic analyzer to help and it did capture data with the local control panel connected to the motor.   Using the visualizations of the logic data from open source Pulse View by sigrok I have determined that the local control communicates with the motor at 9600 baud with the Tx and Rx lines operating independently not differential voltage as found in 485  

I pulled several samples of data and I am analyzing them now.     What I started with was observing a condition of power on to the control system and pump off with no buttons pushed for 3 minutes.   The control board had a repeating pattern of 9 data / idle pairs.  8 of the data messages were identical but the 9th and final one was slightly different.   all 9 data messages were the same length 5 bytes  with 1 through 8 being AB, 00, 00, 00, 00 in hex and 9 being AA, 82, 92, 00, 00 but the idle periods in between varied.  The pattern of the idle period length was consistent from cycle to cycle.  The motor did not send any communication to the motor during this capture

I have created sample where I only cycle the motor on and off.  This does create messages from the motor and new messages from the local control panel that I am reviewing now.    My go forward plan is to capture the command exchange and program a PI Zero to mimic the local controller commands and mimic a 485 device to AqualinkD    I am assuming the PI will replace the local control panel.    Once I have a few command / response elements identified I will cross reference with Jandy/Pentair protocols but I am suspecting the manufacturer as opted for as common of a variable speed control as possible and not made any allowances for a pool application.  

Please advise if these updates are of any interest or should be posted somewhere else.   This is definitely not an AqualinkD issue and I have appreciated your advice and support.

coolrazor007, 2021-11-19:
I'm interested as I'm curious in where this ends up going for you.  I also have a Pentair pump and it stopped communicating recently with my Jandy PDA chip.  I think it is due to old wiring that may have failed and I am replacing the Pentair cable soon.  In any case, since your situation is similar-ish to mine I'd like to know what you end up doing.

john1158way, 2021-11-29:
An update on my approach and progress.   Comments and suggestions requested.  @shaun - I am more than happy to move this to the new equipment sub-forum if you prefer.  As always thank you for a great project and I hope this can end up being some contribution to AqualinkD.

I am using a $10 FX2LP based 8 channel 24 mhz logic analyzer from Amazon and PulseView, open-source software by sigrok.  It comes with a protocol analyzer for UART communication that is decoding the communication between the pump control and motor variable speed package.

A word of caution on inexpensive logic analyzers.  On the one I purchased channel 0 had a defect and led me to some incorrect conclusions about the motor not responding to every control message before I started using different channels. I realize I have reached at least 1 incorrect conclusion about the communication.  That one I have corrected but there is a good chance there will be more before the work is complete.

I like to think I am using an agile approach of creating an incremental way forward, getting a result, refining the way forward.  I have not decided at what point I should try to configure and execute a prototype controller.  Prototype could do motor data collection to aid in decoding.   Any experiences / suggestions appreciated.    At a high level my current approach is a variation on the RS485 logging approach defined by Shaun on the logging and new equipment sub-forum:

Establish a base set of conditions to monitor
For each condition capture a period of communication using logic analyzer
	Rx on Channel 2
	Tx on Channel 3
	Ground connected
Execute any control changes at a specific point in time from start of capture
Use PulseView UART decoder on captured messages
	1 start bit, 1 stop bit, 8 data bits, no parity, 9600 baud rate, invert signals (these settings eliminated all  error/warning 
         messages from UART decoder)
Refine data collection condition set 
Build prototype controller replacement for local control
Test prototype control
Refine data collection conditions set
repeat capture and analysis
Expand prototype control command set
Add RS 485 functionality to prototype
Integrate with AqualinkD

**Controller functions and planned condition sets for data capture**
The local pump controller has 3 speed presets which are configurable by RPM
The local pump controller has 2 programmable modes for setting run time and speed combinations
The local pump controller has a automatic priming feature where the pump runs on the high-speed RPM for 5 minutes no matter which speed is selected if the pump has been off for more than 10 minutes
Current condition definition  set
Condition 01 - Pump off no buttons pushed 82 second data capture
Condition 02 - Pump on no buttons pushed 82 second data capture 
Condition 03 - Pump on 5 min high speed priming time out complete starting condition push stop 50 sec into 82 second data capture
Condition 04 - Pump off prime completed starting condition pushed high-speed start 3100 RPM  50 sec into 82 second data capture 
Condition 05 - Pump off prime completed starting condition pushed low-speed  start 1400 RPM 50 sec into 82 second data capture
Condition 06 - Pump off prime completed starting condition pushed medium-speed  start 2400 RPM 50 sec into 82 second data capture
Condition 07 - Pump on prime completed starting condition high speed 3100 RPM reduce high-speed RPM in 50 RPM increments to 2950 RPM 50 sec into 82 second data capture
Condition 08 - Pump on prime completed starting condition medium speed 2400 RPM reduce medium-speed RPM in 50 RPM increments to 2250 RPM 50 sec into 82 second data capture
Condition 09 - Pump on prime completed starting condition low speed 1400 RPM reduce low-speed RPM in 50 RPM increments to 1300 RPM 50 sec into 82 second data capture
Condition 10 – Pump running in low-speed mode 1 minute before 5 minute auto-priming timer expires
Condition 11 - Pump off 1 min before programmable high-speed start scheduled and executed 2 min data capture
Condition 12 - Pump on 1 before programmable high-speed  completes 2 min data capture 
Current Result set
01 - Pump off no buttons pushed 82 second data capture
        Local control sends message AB, 00, 00, 00, 00, 00 and the motor responds 56, 00, 00, 00, 00, 00
        Every 9th message Local control sends message AA, 00, 00, 00, 00, 00 and the motor responds 56, 00, 00, 00, 00, 00

02 - Pump on high-speed 3100 RPM after prime  no buttons pushed 82 second data capture
        Local control sends message AB, 00, 00, 00, 00, 00 and the motor responds 56, 00, 00, 00, 00, 00
        Every 9th message Local control sends message AA, 82, 92, 00, 00, 14 and the motor responds 56, 82, 92, xx, xx, xx
        xx vary in undefined way possibly reporting motor conditions other than RPM

03 - Pump on 5 min high speed priming time out complete starting condition push stop 50 sec into 82 second data capture
        Before pushing stop, same as condition 02
        Pushing stop, local control sends AB, 85, BC, 11, C7, 19, AB, 00, 00, 00, 00, 00 motor responds 56, 00, 00, 00, 00, 00
        After stop same as condition 01

04 - Pump off prime completed starting condition pushed high-speed start 3100 RPM  50 sec into 82 second data capture
        Before pushing high-speed start same as condition 01
        Pushing start local control sends AB, 43, 46, 00, 01, CA, AB, 82, 7A, 00, 02, FE the motor returns 56, 00, 02, 00, 01, 03
        After pushing start same as condition 02

john1158way, 2021-12-01:
I believe I understand the basic get status, start and stop protocols and I would like to build a prototype to test that my data collection and data analysis process is sound.   @Shaun I would like your thoughts on prototype approaches.   I can build a pi Zero that is a simple replacement app and integrates to AqualinkD via my home automation that uses the AqualinkD API in the future. or I can use my Pi 3 to run AqualinkD and the local control.   To this set up I could add RS 485 Jandy or Pentair VSP emulation into the controller app in the future.   Or I could possibly just replace connect to the controller with a TTL to RS485 adapter and use your defined process in the sub-forum for adding new equipment.

john1158way, 2021-12-10:
I have looked on the website without success.   Where can I find the Jandy or Pentair VSP motor control protocol definitions?  i am moving ahead with the design that has the Raspberry Pi 3 perform motor control via serial communication with the motor control package and emulates a Jandy or Pentair motor on a separate 485 bus.

john1158way, 2021-12-20:
I have a python program on the Pi that can send the appropriate serial commands to the pump speed control package using the GPIO pins of the Pi.  This had some added complication because the controller protocol is inverted from normal and I had to find a program that created a configurable UART port to make it work.  Soft_UART did the trick.   I am a bit undecided on implementing the integration with AqualinkD as my idea requires understanding of the Jandy or Pentair protocols that I don't have.  Since my Pi has wifi access and is a separate device with a separate ip address from my aqualinkD host it will be much more straightforward to just have a pump mini-page

john1158way, 2021-12-20:
I also wonder if my board REV F will support the Jandy or Pentair VSP protocols?

sfeakes, 2021-12-20:
I guess this depends on what you are trying to do.  AqualinkD webui can support other devices served up over any protocol a browser can support.  If you look below, the bottom line of icons are served up from the AqualinkD UI but AqualinkD itself can't control them, the AqualinkD UI sends an http API request to a different server to turn them on/off/open/close.  It's undocumented on how to do this, it's kind-a like a plugin, I can post some examples. In this case you could simply control your python program from the AqualinkD UI, but no real integration.
So in this case, you could create an API in your python program that controls the pump, then do the javascript coding in AqualinkD.
[image]

If you want full integration to Jandy control panel, then that mean you need to create a bridge between the two protocols in your pi & python program.  It can be done, I've done it with a SWG but will take a serious amount of time.

If you want to receive messages from AqualinkD then change something on the pump to some event. Example receive "Cleaner turned on", so you can set pump to Max in your python program.  Then subscribing to MQTT events in your python program, that AqualinkD posts to the MQTT server would be the best option for that.

sfeakes, 2021-12-20:
> I also wonder if my board REV F will support the Jandy or Pentair VSP protocols?

No is doesn't.   VSP came in rev O or P if I remember correctly.

sfeakes, 2021-12-20:
In `/var/www/aqualinkd/config.js` add the below line to the end of the file
```
document.writeln("<script type='text/javascript' src='extra/extra.js'></script>");
```

Then create a directory `/var/www/aqualinkd/extra`, and a file called `extra.js` in that directory.   Now that file will get loaded and set the the browser every time you open the webui.

Below is the content of that file, that produces the UI a few posts above. (obviously you need to add the icons in that directory as well).
```
GroupSwitchType = -99;
GroupLightType = -98;
GroupAudioType = -97;

setTimeout(function() {
  //extra_domoticz('get', 'type=devices&rid=12');  // Absolute radio poller
  //extra_domoticz('get', 'type=devices&rid=281'); // Audiotron Switch
  //extra_domoticz('get', 'type=devices&rid=279'); // Audiotron Selector
  //extra_domoticz('get', 'type=scenes', 8, GroupAudioType);
  extra_domoticz('get', 'type=devices&rid=332'); // Patio audio virtual switch
  extra_domoticz('get', 'type=devices&rid=147');
  extra_domoticz('get', 'type=devices&rid=148');
  //extra_domoticz('get', 'type=devices&rid=21'); // Patio String Lights
  //extra_domoticz('get', 'type=devices&rid=55'); // Patio String Lights #2
  //extra_domoticz('get', 'type=devices&rid=193');
  extra_domoticz('get', 'type=scenes', 2, GroupLightType);
  extra_domoticz('get', 'type=scenes', 6, GroupLightType);
}, 1000);

function switchTileState_extra(id, idx, type) {
  // console.log("Click "+id+" "+idx);
  state = (document.getElementById(id).getAttribute('status') == 'off') ? 'on' : 'off';
  var switchval;
  // console.log("Status "+state);

  if (type == 19) // Reverse on off for doors
    switchval = ((state == 'on') ? 'Off' : 'On');
  else
    switchval = ((state == 'on') ? 'On' : 'Off');

  if (type == GroupSwitchType || type == GroupLightType || type == GroupAudioType)
    extra_domoticz('set', 'type=command&param=switchscene&idx=' + idx + '&switchcmd=' + switchval);
  else if (type == 18) // selectorswitch
    //http://trident/json.htm?type=command&param=switchlight&idx=279&switchcmd=Set%20Level&level=10
    extra_domoticz('set', 'type=command&param=switchlight&idx=' + idx + '&switchcmd=Set%20Level&level=' + ((switchval=='On') ? 10 : 0) );
  else
    extra_domoticz('set', 'type=command&param=switchlight&idx=' + idx + '&switchcmd=' + switchval);

  setTileOn(id, state);
}

function getLevelName(levelNames, levelIndex) {
  var names = decode64(levelNames).split("|");
  return names[levelIndex / 10];
}

function extra_updateDevice(data) {
  //console.log(data);

  var tile
  // console.log(data.Name+" - hw "+data.HardwareTypeVal+" - sw "+data.SwitchTypeVal);
  if ((tile = document.getElementById(data.ID)) == null) {
    // console.log(data.Name+" - create");
    switch (data.SwitchTypeVal) {
    case 0:  // switch
    case GroupSwitchType: // Group switch
      if ( data.idx == "332" ) // If music switch change icon
        add_tile(data.ID, data.Name, data.Status, 'switch', 'switch', 'extra/switch_8-off.png', 'extra/switch_8-on.png');
      else
        add_tile(data.ID, data.Name, data.Status, 'switch', 'switch', 'extra/switch_0-off.png', 'extra/switch_0-on.png');
      break;
[... 134 lines cut ...]
      chr2 = ((enc2 & 15) << 4) | (enc3 >> 2);
      chr3 = ((enc3 & 3) << 6) | enc4;
      output = output + String.fromCharCode(chr1);

      if (enc3 != 64) {
          output = output + String.fromCharCode(chr2);
      }
      if (enc4 != 64) {
          output = output + String.fromCharCode(chr3);
      }
      chr1 = chr2 = chr3 = "";
      enc1 = enc2 = enc3 = enc4 = "";
  } while (i < input.length);

  return decodeURI(output);
}
```

The above is using Domoticz API, and written in a generic way to see what type of device Domoticz returns and create the tiles appropriately.  So a lot more complicated than you would need.  Import parts are.

setTimeout function.  This is what would poll your python program for information. in this example it calls extra_domoticz function.  That function all the Domoticz api, and passes the return.  it calls extra_updateDevice which will either update or create the tile if it's not their.  
To add a tile  it uses the add_tile function ( part of AqualinkD ) to add a tile, then uses setAttribute on the tile to overwrite the onclick event, to call a function in this file.  (switchTileState_extra)
To update a tile it uses setTileOn function (part of AqualinkD) 

That's the flow, you shouldn't need to modify any AqualinkD code.

john1158way, 2021-12-24:
thank you very much for this.   I have family in for the holidays and will try in a week or so.    Merry Christmas Shaun!