aqualinkd/AqualinkD #153: Clarity re Aqualink M8 and PDA+RS

woodgieboodgie, 2022-01-03:
I thought Sprinklerd was good, this is next level - awesome work!

Regarding the documented wiki re PDA only installs, do you recommend the M8 be purchased with an RS hard wired panel? Will that alleviate the issues if I hard wire the USB-RS485 straight into a RS wall mounted unit? or does the 'going to sleep asap' design discussed in the wiki for the PDA still apply if RS is present too?

sfeakes, 2022-01-07:
Sorry what do you mean M8?
The sleep mode is because only one PDA is supported by PDA only panels, and since AqualinkD mimics a PDA (in PDA mode) it has to sleep as to not conflict with the actual jandy PDA device. There is nothing that can be done except to overcome this except upgrade the control panel. If you don’t use the jandy PDA or have a panel that supports non PDA devices as well, this is not a problem.

woodgieboodgie, 2022-01-07:
I understand the Aqualink M8 is the new panel/model.  The wiki refers to upgrading the panel to support non PDA, but I am thinking of buying new and want to ensure i get the right M8 (assuming ther eis a PDA and non PDA version?)
Your comment "_have a panel that supports non PDA devices as well, this is not a problem_".. so if the M8 supports hard wired wall mounted control panels (in addition to PDA) is it better to piggy back the hard wired panel..?

AUPoolUser, 2022-03-12:
I believe the Aqualink M8 is an Australian branding of the Aqualink.