aqualinkd/AqualinkD #163: Can't turn buttons off in config.js

SethL, 2022-04-04:
I commented out the buttons I don't want to see (I don't have a heater or a spa) but they keep showing up on the webui.  Am I doing something wrong?

    var devices = [
        "Filter_Pump",
        //"Spa_Mode",
        "Aux_1",
        //"Aux_2",
        //"Aux_3",
//        "Aux_4",
//        "Aux_5",
//        "Aux_6",
//        "Aux_7",
//        "Aux_B1",
//        "Aux_B2",
//        "Aux_B3",
//        "Aux_B4",
//        "Aux_B5",
//        "Aux_B6",
//        "Aux_B7",
 //       "Aux_B8",
        //"Pool_Heater",
        //"Spa_Heater",
        "SWG",
        //"SWG/Percent",
        "SWG/PPM",
        //"SWG/Boost",
        "Temperature/Air",
        "Temperature/Pool",
        //"Temperature/Spa",
        "Pool_Water",
        // "Spa_Water",
        "Freeze_Protect",
        "CHEM/pH",
        "CHEM/ORP"
        //"Solar_Heater",

[image]

sfeakes, 2022-04-04:
Aqualinkd tells the browser to cache heavily, so your browser probably hasn’t requested the new config.   Clear the cache / try a different browser.

SethL, 2022-04-04:
I just tried on my mobile browser, I haven't ever visited the site before on this device. 
[image]

sfeakes, 2022-04-04:
You edited the file in /var/www/AqualinkD/…….  And not the file from where you download the github repo? (/home/……) usually.

SethL, 2022-04-04:
yeah I was in the wrong place :-)  Everything looks good now.  I love this setup thank you very much!
question: can I have VSP on a button by itself?  I have turned my pump off twice because I forgot to tap and hold :-(

sfeakes, 2022-04-05:
Interesting observation.  VSP support has not always been their, so many are used to the tap being on/off and were pleased to finally get VSP, so making it like the other buttons where long press gives options and tap for on/off was natural.  But that may not be true for new users.  There isn’t a clean way to do what you want, but I think if you have a circuit empty you don’t use, you could assign the VSP to that button and not the actual pump circuit, but you would need to also turn it on and off for the UI to work and give you the RPM/GPM options. So kinda hack.   I’ll need some more time to think that one through and see if I can come up with a better way.

SethL, 2022-04-05:
honestly what would help the most would be to have the option to longpress to turn items on/off.  I turned off the pump and the SWG 3 times yesterday lol.  Im not sure how that factors in with the variable options but it would be nice to have the options.