aqualinkd/AqualinkD #172: Saltwater cell - 0%-> programmed % -> 0% (bouncing) when AqualinkD service is started

igorek24, 2022-04-23:
### Discussed in https://github.com/sfeakes/AqualinkD/discussions/171

<div type='discussions-op-text'>

<sup>Originally posted by **igorek24** April 23, 2022</sup>
My Jendy AquaPure SWG keeps turning on and off when AqualinkD service is running and the only way to make it work is ether disconnect it from RS285 or shutdown AqualinkD. I tried to set the read_`RS485_swg` to yes/no as well as `force_SWG` yes/no/comment it out and nothing helps.
Here is my output from the logger:

`AqualinkD serial_logger V1.4
Notice:  RS Serial: Logging serial information!
Please wait.|

Notice:  RS Serial: RS485 interface received 2000 packets in 42 seconds (~47.62 Msg/Sec)
Notice:  RS Serial: Jandy Control Panel Model   :  RS-8 Combo
Notice:  RS Serial: Jandy Control Panel Version :  REV T.2   
Notice:  RS Serial: Jandy ID's found
Notice:  RS Serial: ID 0x33 is in use  <-- Aqualink (iAqualink / Touch)
Notice:  RS Serial: ID 0xa3 is in use  <-- Unknown Device
Notice:  RS Serial: ID 0x50 is in use  <-- Salt Water Generator (Aquarite mode)
Notice:  RS Serial: ID 0x60 is not used  <-- can use for Aqualinkd (PDA mode only)
Notice:  RS Serial: ID 0x08 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x0a is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x0b is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x40 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x41 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x42 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x43 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x30 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x31 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x32 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x48 is not used  <-- can use for Aqualinkd (RSSA ID)
Notice:  RS Serial: ID 0x09 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: 

Notice:  RS Serial: Pentair ID's found
Notice:  RS Serial: ID 0x10 is in use  <-- Pentair Master (Probably Jandy RS Control Panel)
Notice:  RS Serial: ID 0x60 is in use  <-- Pentair VSP
Notice:  RS Serial: `

and Here is my config:

`web_directory=/var/www/aqualinkd/

log_file=/var/log/aqualinkd.log

log_level=DEBUG

display_warnings_in_web=true

socket_port=8081

serial_port=/dev/ttyS2

device_id=0x0a

rssa_device_id=0x48

extended_device_id=0x31

extended_device_id_programming = yes

read_RS485_swg = no
read_RS485_ePump = no
read_RS485_vsfPump = yes

keep_paneltime_synced = yes

override_freeze_protect = no

convert_mqtt_temp_to_c = no
convert_dz_temp_to_c = no

report_zero_spa_temp = yes

report_zero_pool_temp = yes

mqtt_address = <redacted>
mqtt_user = <redacted>
mqtt_passwd = <redacted>
mqtt_aq_topic = <redacted>

force_SWG = yes

light_programming_mode=0.6

light_programming_initial_on=15

light_programming_initial_off=12

use_panel_aux_labels=no

button_01_label=Filter Pump
button_01_pumpID=0x60
button_01_pumpIndex=1

button_02_label=Spa Mode

button_03_label=Spa Jets

button_04_label=Waterfall

button_05_label=Spa Light

button_06_label=Pool Light
#button_05_lightMode=0

button_07_label=Cleaner
#button_05_lightMode=0

button_08_label=NONE

button_09_label=NONE

button_10_label=Pool Heater

button_11_label=Spa Heater

button_12_label=Solar Heater`</div>

sfeakes, 2022-04-25:
SWG like VSP require a constant connection to the control panel for information about what they should be doing. When SWG looses that connection it will go to 0%.  There are other reasons is will bounce to zero.  But your issue kind-a sounds like a connection issue (also the most common).  You can check by looking at physical SWG panel itself, and monitoring the LED's or LCD depending on what you have / displayed.  If you see is bouncing between "controlled by panel", "beeping", or LCD displaying 0, then this is the case.

If you do see that, it's due to wiring or incorrectly terminated wiring. The RS485 connector takes power when it sends commands, and this is what causes the SWG to looks "being controlled by the panel".  These issues are not picked up by aqualinkd's serial logger, as that is in read only.

niharmehta, 2023-06-03:
Similar to a long standing (and intermittent) issue I have had.  For me it is usually it is triggered after a process restart or power cycle of the computer running aqualinkd.  What I have found, and I have no idea why this works, is basically keep aqualinkd process shut down for at least a full day.  Then start the process. For some reason in my case, it will sync back up correctly and no longer bounce the SWG or the readings from the pump (Watts&speed) .  I can recreate this pretty easily nowadays. 
It makes me wonder if there is some sort of timing mechanism with the RS which is sensitive to new talkers on the bus and if not done correctly, the SWG bouncing is a symptom.  After some period of time, it resets something and works normally.

sfeakes, 2023-06-04:
@niharmehta @igorek24 

I have added some information to the wiki that may help understand the problem.
https://github.com/sfeakes/AqualinkD/wiki#Common-Problems

There is also a new version posted v3.2.0f that has a few options that may help.  Again the details are in that section of the wiki.  Quickly
* Check your usb2rs485 device supports low latency mode
* Maybe try the priorities-ack option in the config.

johnwillyn, 2023-09-10:
I have been experiencing this as well, and reported it a few months back.  I have had aqualinkd running now for a few months, and monitoring the system for this issue via a notification framework (of my own design).  It alerts me on my phone when I get repeated SWG->0% msgs.

I am going to try some of the workarounds in the latest version, and will report back.

I can provide the following details:

I have a relatively old Pentair VS pump.
This issue only happens when my pump first turns on (either by schedule or pool service mode), but it does not happen on every pump start.   It happens about 50% of the days.
Stopping aqualinkd and restarting it after a minute or so seems to correct the issue, but sometimes it takes a few stop/start cycles.

sfeakes, 2024-03-30:
@niharmehta   @igorek24   @johnwillyn 

I have just updated AqualinkD to version 2.3.3 and think this may have been fixed.  If you pull the latest version and add the below to aqualinkd.conf ```rs485_frame_delay = 4``` see if that helps.

johnwillyn, 2024-03-30:
Installing and configuring now.  I will let you know if this fixes the issue.

I have been limping along with the issue and wired up a notification to alert me on my phone when this error occurs.  It is very intermittent, but does happen about 20% of the days on startup.

Thanks for chasing this and letting us know.

JohnL

sfeakes, 2024-03-30:
The `rs485_frame_delay = 4` is telling AqualinkD to wait 4 milliseconds before replying to the control panel, you may need to play with the 4, somewhere between 1 and 20 I think/hope will fix the problem.

johnwillyn, 2024-03-31:
Thanks again!

Testing with value of 4 for now, and will tweak if I encounter the error.  So far, so good.  Will report results.

johnwillyn, 2024-05-08:
Reporting back after some testing.

I am afraid I still experience the issue, and I have tried values of 4, 8, 12, 16 and 20 for this setting.

The frequency is certainly less (maybe about half) and it often is corrected with just one stop/start cycle.

I will go back and do some longer term testing with the various values to try to get a sense of which one might work best (with my pump).

Hope this helps for now.

sfeakes, 2024-06-11:
@johnwillyn 
I believe this issue is fixed in 2.3.7

It's been an issue that plagued very specific setups for a while now. (Pentair VSP and SWG, with specific board Rev#). I have always known it was an issue with Jandy firmware, but never known what AqualinkD was doing to make the Jandy firmware hit that part of the code with the issue.  So I broke down and bought a Pentair VSP, I got the problem right away, and have been able to change AqualinkD so hopefully we don't see this again.  At least on my setup it's been rock solid for over a weeks's worth of testing.
The `rs485_frame_delay` setting was only making it happen a little less frequently in my testing, not actually solving it. So you can set that back to 4 or so, once you upgrade to 2.3.7

johnwillyn, 2024-06-11:
Awesome.  I will try this right away (and report back).

It has been happening less frequently, and it usually took just a single restart of AqualinkD to “fix”.

Sorry you had to go to such efforts (and expense).  I hope you can repurpose the Pentair pump somehow.  I have had mine a long time, and I can tell you it has been rock solid for 15+ years.

On the other hand, I am on my 4th Jandy at our vacation rental home (where I hope to install AqualinkD this summer).

Thanks again,
JohnL

sfeakes, 2024-06-11:
The new Jandy pump is now my main filter pump, and the old is a backup. since my setup needs 3 pumps in total, having a backup is always good that way I can be lazy about rebuilding any of the pumps.  So no real expense. I just hope I’ve finally got to the bottom of this issue.

niharmehta, 2024-06-12:
Just upgraded to 2.3.7 from my previous (several months old .. pre- rs485_frame_delay) .  Worked perfectly the first time.  Excited that its very possible the Pentair bounce issue is finally resolved!  Thank you @sfeakes! 

Is there a decoder ring for the Pentair mode and status values for the Pentair VSP ?

sfeakes, 2024-06-12:
> Just upgraded to 2.3.7 from my previous (several months old .. pre- rs485_frame_delay) . Worked perfectly the first time. Excited that its very possible the Pentair bounce issue is finally resolved! [...]

Really glad to hear it’s looking good so far.

On the stuff I’ve captured so far, are the values of mode / status / pressure curve & drive state, and post those to MQTT.  But at the moment not decoded what all the values actually mean.  Also captured both GPM and RPM for vs, vf and vsf pumps. Jandy only reports rpm for vs /. vsf and gpm for vf.

Status 0 = ok, is all I know for sure, I know some of the others are overcurrent, blocked, overtemp, communication failure. But I’ll have to force some errors on the pump to figure out what numbers it reported for those states. (I believe their is some cleaning status as well)

Mode is simply remote controlled or local controlled. 1=remote controlled

drive state, no idea as of yet.

pressure curve is described in their manual.

what I’m really interested in is if I can detect the filter basket getting full from amps/gpm/pressure curve.

I’ll add to documentation and make updates as I find out more / get some time to play.

johnwillyn, 2024-06-25:
Just wanted to report that this release has fixed the Pentair salt cell bounce for me as well!

Thanks so much for all of your extra effort on this sfeakes!

JohnL

sfeakes, 2024-06-28:
@johnwillyn Thanks very much for confirming that.