aqualinkd/AqualinkD #194: Increase syslog level limit

niharmehta, 2023-02-16:
I am trying to send all my logs to a remote syslog server to reduce wear on the SDCard as well as to centralize my logs to a server. 
In some cases, I would like to have higher intensity logging sent over to my syslog server, however it looks like syslog is hardcoded to NOTICE . Is there a specific reason that this limit exists limiting to send Debug, Debug_Serial, and Info ? I can understand that when not used carefully, this can cause an issue where somebody quickly fills up their log file, but not sure if it can cause other issues as well.   

```
  // struct lws_context_creation_info info;
  // Log only NOTICE messages and above. Debug and info messages
  // will not be logged to syslog.
  setlogmask(LOG_UPTO(LOG_NOTICE));
```

ballle98, 2023-02-17:
not any issues I am aware of.  Only @sfeakes can speak to the original intent but I would guess it wearing out the flash was also a concern as well as performance on a Pi Zero.  In your case where you are streaming to a syslog server wear that would be less of a concern.  If you turn on debug serial it can be 10+ logs per second.

I would comment out the line.  make; sudo make install

sfeakes, 2023-02-18:
@ballle98 Hit the nail on the head.  Debug and Debug_serial spit out a ton stuff that really has no relevance unless you are trying to fix a problem. There is also extra CPU usage running in these modes, and the intent for this to run on Pi Zero means that it really shouldn't be used unless necessary. Also with rsyslog you have all the extra network overhead to account for as well.
It should work, but can't say I've really tested it long term, and if you have a busy RS485 buss with lots of devices I'd really warn against setting debug_serial long term.
BTW, When I use debug_serial I always send to file that's on a ramdisk (no syslog), that way the extra IO isn't effecting anything.

niharmehta, 2023-02-21:
Thanks for the feedback. All makes sense. Actually wanted to use this to troubleshoot a very long standing intermittent issue with the SWG setting bouncing and the watts jumping like crazy. Some interaction with the SWG, iAqualink, or the RS.  Usually can be stable for many, many months, but after some process restart or the iaqualink, does something it will pop up until I  restart the pi  and/or restart the RS. 
After I destroyed my original pi-zero by encouraging it to send out magic smoke with a few extra volts,  I ended up getting a BananaPi since any Raspberry's have been long not available.  The BananaPI actually works pretty well and I have not had any new issues. Just the same ones that sometimes crop up. 
I made the changes and set my logging to informational, will see how much this impacts the system with the network overhead of using UDP syslog. (hopefully minor).  Thanks for the guidance.

closed 2023-05-28 by sfeakes