aqualinkd/AqualinkD #210: Read RS485 bus from ESP32

maniscalcojohn, 2023-05-08:
Is it possible to install the Aqualinkd on a server and then use an esp32 at the pool panel to read/relay commands to the aqualinkd service?

ballle98, 2023-05-08:
This is similar to #203 it would be much easier to use a Pi Zero W or run cat5 from your panel to the server.  RS485 can be several 100 ft.

sfeakes, 2023-05-08:
You can install on server, but no using an esp (or anything else) at the RS485 to use as an IP bridge for the serial connection.  It would introduce way too much latency.

I’ve played with AqualinkD on ESP devices, and if you strip out all functionality that requires multiple threads it works well, but at that point AqualinkD looses all the functionality people like and it simply becomes a proxy for on/off commands.   Cost of pi zero W (or equiv) is only a few $ more than a capable ESP device that it doesn’t seem worth the effort to peruse.  And for one off it’s cheaper and easier to use a pi zero as no usb programmer or development knowledge is needed to get a pi up and running.

maniscalcojohn, 2023-05-09:
Ok. Understood. I’ve got an esp32 at my pool controller for reading ph/orp and dosing. Thought maybe I could incorporate this, but as you mentioned, probably not worth the effort.

sfeakes, 2023-05-09:
Be best to simply add a pi zero and move the esp functionality to that if you only want one device.
But I’m very interested in your whole esp32 setup for reading ph/orp, can you post details or link?

maniscalcojohn, 2023-05-10:
Sure. Here is a writeup I did on it a while back [https://community.home-assistant.io/t/swimming-pool-chemical-dispenser/205561/112?u=jm123](url). It has changed a bit since--I dont use the Atlas Scientific pumps anymore (they didnt last). I use this probe from Atlas Scientific [https://atlas-scientific.com/kits/ph-kit/](url) for pH and this probe for ORP [https://atlas-scientific.com/kits/orp-kit/](url). They communicate on the i2c bus with an esp32 currently running tasmota which sends the data to Home Assistant via MQTT (planning to switch over to ESP Home eventually). The pH probe works great, very accurate. I calibrate it about once a year at the beginning of pool season. The ORP probe is not perfect--it is mostly useful for telling me when my water has a good level of chlorine, and the readings will start to drop off when the chlorine levels get low (usually because my chlorine tank runs empty). I am told the ORP probe also wont work well if you have high CYA. I keep mine around 40-50.

I also have a couple of stenner peristaltic pumps which I use to dose acid and chlorine. These pumps are 24v attached to a relay, also run with the esp32. For acid, I bought the pentair intellichem acid tank already with the pump. I'm sure there is a cheaper way to do this one, but it works well, and has a way to vent the tank away from your pad so you dont have any corrosion. For the chlorine, I bought the same pump on the intellichem but put it in a seperate housing and it feeds from 2.5G chlorine tanks I buy from the pool store,

maniscalcojohn, 2023-05-23:
Any experience with these RS485 to wifi adapters?

[](https://www.amazon.com/ARAIERD-Converter-Adapter-Support-Protocol/dp/B0B764B3LY/ref=sr_1_3?crid=35BMUA16B7HPF)

[](https://www.amazon.com/gp/product/B07DNWM62H/ref=ox_sc_saved_title_1?smid=A2LNZEPI7X9Y44&psc=1)