aqualinkd/AqualinkD #217: Build broken by commit 77c8e29

ballle98, 2023-06-07:
In commit 
https://github.com/sfeakes/AqualinkD/commit/77c8e290d5e366a3b4656cd43f34190ae65ddb3d

mongoose.c and mongoose.h are replaced by symlinks to files that have not been added to the repository

./mongoose/mongoose-v6.18.h
./mongoose/mongoose-v6.18.c

```
C:\git\AqualinkD>make CC=arm-linux-gnueabihf-gcc
cat: /etc/os-release: No such file or directory
cut: the delimiter must be a single character
Try 'cut --help' for more information.
OS:
GLIBC build with: ldd (cygwin) 3.4.6
GLIBC Prefered  : 2.24-11+deb9u1 2.24
arm-linux-gnueabihf-gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c aqualinkd.c  -o aqualinkd.o
In file included from aqualinkd.c:32:
mongoose.h:1:1: error: expected identifier or '(' before '.' token
    1 | ./mongoose/mongoose-v6.18.h
      | ^
mongoose.h:1:23: error: too many decimal points in number
    1 | ./mongoose/mongoose-v6.18.h
      |                       ^~~~~
make: *** [aqualinkd.o] Error 1

```

ballle98, 2023-06-07:
Fails on pi as well

```
pi@raspberrypi:~/git/AqualinkD $ make
OS: 11 (bullseye)
GLIBC build with: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u5) 2.31
GLIBC Prefered  : 2.24-11+deb9u1 2.24
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c aqualinkd.c  -o aqualinkd.o
aqualinkd.c:32:10: fatal error: mongoose.h: No such file or directory
   32 | #include "mongoose.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:184: aqualinkd.o] Error 1

```

ballle98, 2023-06-07:
```
pi@raspberrypi:~/git/AqualinkD $ ls -l mon*
lrwxrwxrwx 1 pi pi 27 Jun  7 13:57 mongoose.c -> ./mongoose/mongoose-v6.18.c
lrwxrwxrwx 1 pi pi 27 Jun  7 13:57 mongoose.h -> ./mongoose/mongoose-v6.18.h

```

sfeakes, 2023-06-07:
https://github.com/sfeakes/AqualinkD/commit/77c8e290d5e366a3b4656cd43f34190ae65ddb3d#r116945664

closed 2023-06-07 by sfeakes