aqualinkd/AqualinkD #220: Not seeing Pentair VS Pump speed

mcoutsoft, 2023-06-19:
Hi, Thanks for this awesome project.

Wanted to see if you could give me a hand. Have a couple of issues.

Using Pi Zero WH + the USB connector cable you specified in the Wiki.

1. I'm not seeing the pump speed in the web interface for our Pentair VS Pump. Tried w/ current release + latest build.

2. AqualinkD gets stuck at connecting to control panel at least 50% of the time.

Any help would be greatly appreciated.

I've included the output from the serial logger below:

[image]

+ output when I run AqualinkD

Notice:  AqualinkD: Aqualink Daemon v2.3.0f
Notice:  AqualinkD: Panel set to RS-8 Combo Pool/Spa
Notice:  AqualinkD: Config log_level         = 5
Notice:  AqualinkD: Config device_id         = 0x0a
Notice:  AqualinkD: Config rssa_device_id    = 0x48
Notice:  AqualinkD: Config extra_device_id   = 0x41
Notice:  AqualinkD: Config extra_device_prog = NO
Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
Notice:  AqualinkD: Config socket_port       = 80
Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
Notice:  AqualinkD: Config use_aux_labels    = NO
Notice:  AqualinkD: Config override frz prot = NO
Notice:  AqualinkD: Config mqtt_server       = (null)
Notice:  AqualinkD: Config mqtt_dz_sub_topic = (null)
Notice:  AqualinkD: Config mqtt_dz_pub_topic = (null)
Notice:  AqualinkD: Config mqtt_aq_topic     = (null)
Notice:  AqualinkD: Config mqtt_user         = (null)
Notice:  AqualinkD: Config mqtt_passwd       = (null)
Notice:  AqualinkD: Config mqtt_ID           = aqualinkd_b827ebf4
Notice:  AqualinkD: Config force SWG         = YES
Notice:  AqualinkD: Config force PS setpoint = NO
Notice:  AqualinkD: Config deamonize         = NO
Notice:  AqualinkD: Config log_file          = (null)
Notice:  AqualinkD: Config enable scheduler  = YES
Notice:  AqualinkD: Config light_pgm_mode    = 0.00
Notice:  AqualinkD: Debug RS485 protocol     = NO
Notice:  AqualinkD: Debug RS485 protocol raw = NO
Notice:  AqualinkD: Display warnings in web  = NO
Notice:  AqualinkD: Keep panel time in sync  = YES
Notice:  AqualinkD: Read SWG direct          = NO
Notice:  AqualinkD: Read ePump direct        = NO
Notice:  AqualinkD: Read vsfPump direct      = YES
Notice:  AqualinkD: Serial FTDI low latency  = YES
Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | VSP ID 0x60 | PumpID 1 |
Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa Mode        | 
Notice:  AqualinkD: Config BTN Aux_1         = label Spa Booster     | 
Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
Notice:  AqualinkD: Config BTN Aux_3         = label Spillover       | 
Notice:  AqualinkD: Config BTN Aux_4         = label Spa Light       | 
Notice:  AqualinkD: Config BTN Aux_5         = label Spa Jets High   | 
Notice:  AqualinkD: Config BTN Aux_6         = label Pool Light      | 
Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heater     | 
Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heater      | 
Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
Notice:  NetService:Starting network services thread
Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Notice:  AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Notice:  AqualinkD: Waiting for Control Panel probe
Notice:  NetService:Starting web server on port 80
Notice:  NetService:Starting MQTT client to (null)
Notice:  AqualinkD: Got probe on '0x48' RS SerialAdapter Protocol
Notice:  AqualinkD: Got probe on '0x0a' Standard Protocol
Notice:  AqualinkD: Got probe on '0x41' Extended Protocol
Notice:  AqualinkD: Starting communication with Control Panel
Notice:  Panl Prog: Starting programming thread 'Get SerialAdapter setpoints'

sfeakes, 2023-06-19:
> AqualinkD gets stuck at connecting to control panel at least 50% of the time.

This might be the problem, can you provide some more details on what's happening.  Usually this happens when there are lots of serial errors. These happen usually from wiring issue (or having multiple instances or AqualinkD running).  But you should see the errors in the logs.

Also if you post more of the log a full would be helpful.  (you can log to file, or just post a lot more of the output above).  Setting `log_level=INFO` or maybe `log_level=DEBUG` in the config will increase information.

mcoutsoft, 2023-06-19:
root@aqualinkd:/var/log# sudo systemctl stop aqualinkd
root@aqualinkd:/var/log# sudo aqualinkd -d -c /etc/aqualinkd.conf
Notice:  AqualinkD: Aqualink Daemon v2.3.0f
Notice:  AqualinkD: Panel set to RS-8 Combo Pool/Spa
Notice:  AqualinkD: Config log_level         = 7
Notice:  AqualinkD: Config device_id         = 0x0a
Notice:  AqualinkD: Config rssa_device_id    = 0x48
Notice:  AqualinkD: Config extra_device_id   = 0x41
Notice:  AqualinkD: Config extra_device_prog = YES
Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
Notice:  AqualinkD: Config socket_port       = 80
Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
Notice:  AqualinkD: Config use_aux_labels    = NO
Notice:  AqualinkD: Config override frz prot = NO
Notice:  AqualinkD: Config mqtt_server       = (null)
Notice:  AqualinkD: Config mqtt_dz_sub_topic = (null)
Notice:  AqualinkD: Config mqtt_dz_pub_topic = (null)
Notice:  AqualinkD: Config mqtt_aq_topic     = (null)
Notice:  AqualinkD: Config mqtt_user         = (null)
Notice:  AqualinkD: Config mqtt_passwd       = (null)
Notice:  AqualinkD: Config mqtt_ID           = aqualinkd_b827ebf4
Notice:  AqualinkD: Config force SWG         = YES
Notice:  AqualinkD: Config force PS setpoint = NO
Notice:  AqualinkD: Config deamonize         = NO
Notice:  AqualinkD: Config log_file          = /var/log/aqualinkd.log
Notice:  AqualinkD: Config enable scheduler  = YES
Notice:  AqualinkD: Config light_pgm_mode    = 0.00
Notice:  AqualinkD: Debug RS485 protocol     = NO
Notice:  AqualinkD: Debug RS485 protocol raw = NO
Notice:  AqualinkD: Display warnings in web  = NO
Notice:  AqualinkD: Keep panel time in sync  = YES
Notice:  AqualinkD: Read SWG direct          = NO
Notice:  AqualinkD: Read ePump direct        = NO
Notice:  AqualinkD: Read vsfPump direct      = YES
Notice:  AqualinkD: Serial FTDI low latency  = YES
Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | VSP ID 0x60 | PumpID 1 |
Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa Mode        | 
Notice:  AqualinkD: Config BTN Aux_1         = label Spa Booster     | 
Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
Notice:  AqualinkD: Config BTN Aux_3         = label Spillover       | 
Notice:  AqualinkD: Config BTN Aux_4         = label Spa Light       | 
Notice:  AqualinkD: Config BTN Aux_5         = label Spa Jets High   | 
Notice:  AqualinkD: Config BTN Aux_6         = label Pool Light      | 
Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heater     | 
Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heater      | 
Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
Notice:  NetService:Starting network services thread
Debug:   RS Serial: Openeded serial port /dev/ttyUSB0
Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Info:    RS Serial: Set serial port /dev/ttyUSB0 I/O blocking attributes
Notice:  AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Notice:  AqualinkD: AqualinkD is using use ONETOUCH mode for programming (where supported)
Notice:  AqualinkD: Waiting for Control Panel probe
Notice:  NetService:Starting web server on port 80
Notice:  NetService:Starting MQTT client to (null)
Notice:  AqualinkD: Got probe on '0x48' RS SerialAdapter Protocol
Notice:  AqualinkD: Got probe on '0x0a' Standard Protocol
Notice:  AqualinkD: Got probe on '0x41' Extended Protocol
Notice:  AqualinkD: Starting communication with Control Panel
Debug:   RS SAdptr: Probe received, will queue device update shortly
Info:    RS SAdptr: Panel Model 0x19|0x78|0x01
Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 328
Info:    RS SAdptr: Queue device update requests
Info:    Panl Prog: Initial setup call from RSSA with RSSA=enabled ONETouch=enabled IAQTouch=disabled ExtendedProgramming=enabled
Notice:  Panl Prog: Starting programming thread 'Get SerialAdapter setpoints'
Debug:   RS SAdptr: Added to message queue, position 0 0x00|0x01|0x0a|0x05
Debug:   RS SAdptr: Added to message queue, position 1 0x00|0x01|0x05|0x05
Debug:   RS SAdptr: Added to message queue, position 2 0x00|0x01|0x07|0x05
Debug:   RS SAdptr: Pull from message queue, length 3 0x00|0x01|0x0a|0x05
[... 41 lines cut ...]
Info:    NetService:URI request: '/hk/background.jpg '
Debug:   NetService:Served WEB request
Debug:   NetService:++ Websocket joined
Debug:   NetService:WS: Message - Key 'uri' Value 'devices'
Debug:   NetService:WS: URI Request 'devices': value -999.00
Debug:   NetService:JSON: web used 2882 of 5120
Debug:   NetService:WS: Message - Key 'uri' Value 'status'
Debug:   NetService:WS: URI Request 'status': value -999.00
Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 330
Info:    RS SAdptr: Queue device update requests
Debug:   RS SAdptr: Added to message queue, position 0 0x00|0x01|0x05|0x05
Debug:   RS SAdptr: Added to message queue, position 1 0x00|0x01|0x07|0x05
Debug:   RS SAdptr: Pull from message queue, length 2 0x00|0x01|0x05|0x05
Debug:   RS SAdptr: Remove from message queue, length 1
Info:    RS SAdptr: Pool SP is 82
Debug:   RS SAdptr: Pull from message queue, length 1 0x00|0x01|0x07|0x05
Debug:   RS SAdptr: Remove from message queue, length 0
Info:    RS SAdptr: Spa SP is 99
Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 332
Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 330
Warning: AqualinkD: Stopping!
Info:    RS Serial: Forcing close of blocking serial port, ignore following read errors
Warning: RS Serial: Read error: 9 - Bad file descriptor
Error:   AqualinkD: Nothing read on blocking serial port
Notice:  AqualinkD: Exit!

mcoutsoft, 2023-06-19:
Just realized I'd been running the process as root with sudo command. Dropped sudo and things seem to be working properly.

mcoutsoft, 2023-06-19:
Looks like I may have spoken too soon.. After a Pi reboot, things seem to connect properly, then after a while I get a red stripe on the UI that says no connection to RS Control panel, but all of the buttons still work and still activate their respective functions. Anything else I can check to troubleshoot?

sfeakes, 2023-06-19:
Can you log to a file, and post that.

mcoutsoft, 2023-06-19:
Hi @sfeakes, Let me change the config and log to a file for the next couple of hours and I'll post it.

mcoutsoft, 2023-06-19:
@sfeakes Was able to reproduce the issue pretty quickly.. I turned the filter pump off via the web UI and then got the following screen + logged error

Jun-19-23 16:21:44 PM Warning: RS Serial: Serial read bad Jandy checksum, ignoring
Jun-19-23 16:21:44 PM Warning: RS Serial:  Read BAD PACKET To 0x2b of type            Probe | HEX: 0x10|0x02|0x2b|0x00|0x3d|0x10|0xfb|0x10|0x02|0x0a|0x02|0x00|0x00|0x00|0x00|0x00|0x1e|0x10|0x03|
Jun-19-23 16:21:44 PM Error:   AqualinkD: Nothing read on blocking serial port
Jun-19-23 16:21:44 PM Error:   AqualinkD: Aqualink daemon looks like serial error, resetting.
Jun-19-23 16:21:44 PM Debug:   RS Serial: Openeded serial port /dev/ttyUSB0
Jun-19-23 16:21:44 PM Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Jun-19-23 16:21:44 PM Info:    RS Serial: Set serial port /dev/ttyUSB0 I/O blocking attributes

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11793921/aqualinkd.log)
[image]

sfeakes, 2023-06-20:
This below line shows that AqualinkD (and OS) is detecting a FTDI chip and setting everything according.  But that doesn’t prove it’s not a clone.  But you should be good, I doubt it’s the adapter.
> “Jun-19-23 16:21:44 PM Notice: RS Serial: Port /dev/ttyUSB0 low latency mode is set“

I would check the wiring to the RS bus, the checksum errors are 99.999% of the time due to that.  There is a section in the wiki about that.

If you really think it’s the adapter being a clone, you could try the below setting in aqualinkd.conf, this will turn off specific FTDI stuff that can cause issues on cloned chips.
> ftdi_low_latency = no

sfeakes, 2023-06-20:
In your log, I can see you turned the pump off and it worked, then tried to turned in on and it failed and got a serial error.   So from that you should check wiring.

One thing that’s very odd, is AqualinkD is not going into programming mode on startup so it can get heater set points / freeze protect etc etc.   Can you please post you aqualinkd.conf, I need to check into that further.  But my guess is, you would see a lot more checksum errors if AqualinkD went into programming mode.   You could also try an earlier release like V2.2.2a and see it that does into programming on startup.

mcoutsoft, 2023-06-20:
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11795319/aqualinkd.log)

Aqualink Conf attached as Aqualinkd_conf.log

[aqualinkd_conf.log](https://github.com/sfeakes/AqualinkD/files/11795333/aqualinkd_conf.log)

sfeakes, 2023-06-20:
Thanks,   I’ve never seen that specific panel REV before, and it gives a very slightly different connection message to other panels on initial connection.  I will need to post an update to the code.
> Jun-19-23 16:38:30 PM Info:    RS Allbtn: RS Message :- 'E0260801 REV. O.2'

mcoutsoft, 2023-06-20:
Thank you again for the support. This is my first Pi project.

- I just noticed the leads for the RS485 to USB cable are sharing a terminal block with the two wires from the Pentair VS pump on the Aqualink RS PCB. It looks like i've may have a couple of other RS485 terminal blocks I can use on the multiplexer to isolate them, just in case that's causing an issue.

- I also swapped out the Pi power adapter for something a bit more robust, in the event it's a power related issue.

mcoutsoft, 2023-06-20:
Also, @sfeakes please let me know if I can help with any diagnostic info/logging/troubleshooting for the O.2 firmware rev.

sfeakes, 2023-06-20:
I just posted an update, that should fix AqualinkD not going into programming mode on startup.  if you used the `aqua.sh` script to install, you'll need to run it again with `force_latest` to get the version I just posted.

If you think there might be a power issue with the Pi. their is a script in `/extras/pi_health.sh` that will let you know if you have a power issue that's causing a problem.

On the cabling, you can also play with RS485 terminating on the USB adapter.  The one you have supports that and it may help.  But read the wiki & internet on terminating RS485 so you understand that and can transpose it to how your Pool equipment is wired.  Sometimes connecting the negative wire (NOT THE positive) can help.

mcoutsoft, 2023-06-20:
Thank you. Was trying to figure out how to terminate and read the docs on your wiki + the FTDI website.

As it stands right now, just the two data wires are connected, no ground and no termination.

As I understand it, terminating involves taking one of the two termination wires and connecting it somewhere.

Where, I’m not quite certain.

Our current setup is the RS Aqualink Combo unit connected to an indoor LCD One-touch screen inside the house.

I used the git method to download the app, can I still use the install script?

closed 2025-11-19 by github-actions[bot]