aqualinkd/AqualinkD #221: make clean broken on windows

ballle98, 2023-06-20:
https://github.com/sfeakes/AqualinkD/blob/52d336d2a445b4a85d37404f99ba6e4a5b1b4cc3/Makefile#L51

```
21:15:27 **** Clean-only build of configuration Release for project AqualinkD ****
make CC=arm-linux-gnueabihf-gcc -j clean 
del /Q *.o *~ ./release/aqualinkd    ./release/aqualinkd-debug
Invalid switch - "release".
make: *** [clean] Error 1
"make CC=arm-linux-gnueabihf-gcc -j clean" terminated with exit code 2. Build might be incomplete.

21:15:27 Build Failed. 1 errors, 0 warnings. (took 454ms)
```

are trying to build something for windows native?  using "RM = del /Q" makes no sense without using FixPath() everywhere RM is used.

RM is also packaged with the SysGCC tollkit for doing cros compiles 

```
C:\git\AqualinkD>which rm
/cygdrive/c/SysGCC/raspberry64/bin/rm
```

sfeakes, 2023-06-20:
I'm not trying to build on Windows. I was trying to give you a place in the Makefile for windows specific stuff since that's how you compile/test.   I didn't mean to post the updated Makefile as I haven't finished it.  Just pushed a small change.  But change to make it work for you, at least this way you don't get the GLIBC errors.

ballle98, 2023-06-20:
fixed by https://github.com/sfeakes/AqualinkD/commit/5aab2be4151569270dd6892be2f3a196a6abe6f7

ballle98, 2023-06-22:
Still

ballle98, 2023-06-22:
related to sfeakes/AqualinkD#127 windows build - cat: /etc/os-release: No such file or directory

sfeakes, 2023-06-22:
Windows will need to be changed but shouldn't be hard.  I had to move to a build directory since serial_logger needs to be compiled with different flags for an upcoming release.
quick a dirty fix might be to add something like below 
```
ifeq ($(OS),Windows_NT)
    mkdir .\build
    mkdir .\build\slog
    mkdir .\build\debug
else
```

That way the tests for the director existing should pass and it won't go into the linux mkdir rules.

ballle98, 2023-06-22:
Got a fix in the process of git rebase on my fork   I'll share shortly.

PR #229 (2023-06-23): sfeakes/AqualinkD#221 windows build fix for MKDIR
sfeakes/AqualinkD#127 windows build - cat: /etc/os-release: No such file or directory

fix #221 
fix #127 

Note: there is significant whitespace changes around trailing spaces and tabs.  The real change is on line 180 wrapping FixPath around paramerters to mkdir.  Verified that this works on both Windows and Linux (Pi).

$(MKDIR) $(call FixPath,$@)
closed 2023-06-23 by sfeakes