aqualinkd/AqualinkD #222: make clean make fails on Pi

ballle98, 2023-06-20:
Not sure how the Makefile got messed up but it's no longer linking against the pthread lib

```
pi@raspberrypi:~/git/AqualinkD $ make
OS: 11 (bullseye)
GLIBC build with: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u5) 2.31
GLIBC Prefered  : 2.24-11+deb9u1 2.24
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_ENABLE_HTTP_SSI=0 -D MG_ENABLE_DIRECTORY_LISTING=0 -D MG_ENABLE_HTTP_CGI=0  -l pthread -l m -o release/aqualinkd build/aqualinkd.o build/utils.o build/config.o build/aq_serial.o build/aq_panel.o build/aq_programmer.o build/net_services.o build/json_messages.o build/rs_msg_utils.o build/devices_jandy.o build/packetLogger.o build/devices_pentair.o build/color_lights.o build/serialadapter.o build/aq_timer.o build/aq_scheduler.o build/web_config.o build/mongoose.o build/pda.o build/pda_menu.o build/pda_aq_programmer.o build/onetouch.o build/onetouch_aq_programmer.o build/iaqtouch.o build/iaqtouch_aq_programmer.o
/usr/bin/ld: build/aq_programmer.o: in function `_aq_programmer':
aq_programmer.c:(.text+0xc40): undefined reference to `pthread_detach'
/usr/bin/ld: aq_programmer.c:(.text+0xc54): undefined reference to `pthread_create'
/usr/bin/ld: aq_programmer.c:(.text+0xc70): undefined reference to `pthread_create'
/usr/bin/ld: build/net_services.o: in function `start_net_services':
net_services.c:(.text+0x55d4): undefined reference to `pthread_create'
/usr/bin/ld: net_services.c:(.text+0x55e4): undefined reference to `pthread_detach'
/usr/bin/ld: build/aq_timer.o: in function `start_timer':
aq_timer.c:(.text+0x404): undefined reference to `pthread_create'
/usr/bin/ld: aq_timer.c:(.text+0x44c): undefined reference to `pthread_detach'
collect2: error: ld returned 1 exit status
make: *** [Makefile:171: release/aqualinkd] Error 1

```

sfeakes, 2023-06-20:
Not sure why, but changing the order of where "-l pthread" is placed in the command seems to solve the issue with ldd 2.31.  I'll need to figure out exactly why, as it worked fine on all other ldd versions I tested.   Posting an update now.

ballle98, 2023-06-22:
retested with https://github.com/sfeakes/AqualinkD/commit/57978e4419ca87294ebd0288aa5036b0395a2132 and it is building now

closed 2023-06-22 by ballle98