aqualinkd/AqualinkD #223: aqualinkd.c:1772:8: error: too few arguments to function ‘serial_logger’

ballle98, 2023-06-20:
build broken by https://github.com/sfeakes/AqualinkD/commit/d86e25cd354f879ea96aa9fcc1045e643663ff4d

https://github.com/sfeakes/AqualinkD/blame/d86e25cd354f879ea96aa9fcc1045e643663ff4d/aqualinkd.c#L1772

```
pi@raspberrypi:~/git/AqualinkD $ make
OS: 11 (bullseye)
GLIBC build with: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u5) 2.31
GLIBC Prefered  : 2.24-11+deb9u1 2.24
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_ENABLE_HTTP_SSI=0 -D MG_ENABLE_DIRECTORY_LISTING=0 -D MG_ENABLE_HTTP_CGI=0  -c -o build/aqualinkd.o aqualinkd.c
aqualinkd.c: In function ‘main_loop’:
aqualinkd.c:1771:38: error: ‘CONNECTION_RUNNING_SLOG’ undeclared (first use in this function)
 1771 |        broadcast_aqualinkstate_error(CONNECTION_RUNNING_SLOG);
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~
aqualinkd.c:1771:38: note: each undeclared identifier is reported only once for each function it appears in
aqualinkd.c:1772:8: error: too few arguments to function ‘serial_logger’
 1772 |        serial_logger(rs_fd, _aqconfig_.serial_port, getSystemLogLevel());
      |        ^~~~~~~~~~~~~
In file included from aqualinkd.c:54:
serial_logger.h:14:5: note: declared here
   14 | int serial_logger(int rs_fd, char *port_name, int logPackets, int logLevel, bool panleProbe, bool rsSerialSpeedTest, bool errorMonitor);
      |     ^~~~~~~~~~~~~
make: *** [Makefile:159: build/aqualinkd.o] Error 1

```

sfeakes, 2023-06-20:
Posted an update.  Thanks for catching, this is all to do with the Makefile i inadvertently posted.

ballle98, 2023-06-20:
fixed aqualinkd.c:1771:38: error: ‘CONNECTION_RUNNING_SLOG’ undeclared with https://github.com/sfeakes/AqualinkD/commit/77b24688aac3f406d8a21d6b36cb0e8f6862b405

still seeing the 2nd issue aqualinkd.c:1772:8: error: too few arguments to function ‘serial_logger’

```
pi@raspberrypi:~/git/AqualinkD $ make
OS: 11 (bullseye)
GLIBC build with: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u5) 2.31
GLIBC Prefered  : 2.24-11+deb9u1 2.24
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_ENABLE_HTTP_SSI=0 -D MG_ENABLE_DIRECTORY_LISTING=0 -D MG_ENABLE_HTTP_CGI=0  -c -o build/aqualinkd.o aqualinkd.c
aqualinkd.c: In function ‘main_loop’:
aqualinkd.c:1772:8: error: too few arguments to function ‘serial_logger’
 1772 |        serial_logger(rs_fd, _aqconfig_.serial_port, getSystemLogLevel());
      |        ^~~~~~~~~~~~~
In file included from aqualinkd.c:54:
serial_logger.h:14:5: note: declared here
   14 | int serial_logger(int rs_fd, char *port_name, int logPackets, int logLevel, bool panleProbe, bool rsSerialSpeedTest, bool errorMonitor);
      |     ^~~~~~~~~~~~~
make: *** [Makefile:159: build/aqualinkd.o] Error 1

```

sfeakes, 2023-06-21:
Try again.

ballle98, 2023-06-21:
That is fixed no seeing another issue

```
pi@raspberrypi:~/git/AqualinkD $ make
OS: 11 (bullseye)
GLIBC build with: ldd (Debian GLIBC 2.31-13+rpt2+rpi1+deb11u5) 2.31
GLIBC Prefered  : 2.24-11+deb9u1 2.24
gcc -Wall -O3  -D AQ_PDA -D AQ_ONETOUCH -D AQ_IAQTOUCH -D AQ_RS16 -D MG_ENABLE_HTTP_SSI=0 -D MG_ENABLE_DIRECTORY_LISTING=0 -D MG_ENABLE_HTTP_CGI=0  -o release/aqualinkd build/aqualinkd.o build/utils.o build/config.o build/aq_serial.o build/aq_panel.o build/aq_programmer.o build/net_services.o build/json_messages.o build/rs_msg_utils.o build/devices_jandy.o build/packetLogger.o build/devices_pentair.o build/color_lights.o build/serialadapter.o build/aq_timer.o build/aq_scheduler.o build/web_config.o build/serial_logger.o build/mongoose.o build/pda.o build/pda_menu.o build/pda_aq_programmer.o build/onetouch.o build/onetouch_aq_programmer.o build/iaqtouch.o build/iaqtouch_aq_programmer.o -l pthread -l m
/usr/bin/ld: build/serial_logger.o:(.bss+0x40): multiple definition of `_aqconfig_'; build/config.o:(.bss+0x4): first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:169: release/aqualinkd] Error 1

```

ballle98, 2023-06-22:
Fixed with https://github.com/sfeakes/AqualinkD/commit/57978e4419ca87294ebd0288aa5036b0395a2132

```
./release/aqualinkd has been compiled
./release/serial_logger has been compiled
pi@raspberrypi:~/git/AqualinkD $

```

closed 2023-06-22 by ballle98