aqualinkd/AqualinkD #224: High salt warning error

mcoutsoft, 2023-06-20:
Build 2.3.1 

Getting a banner that says Check Aquapure High Salt, but salt level is ok and I am not seeing any errors/notices on Aquapure panel or Onetouch Panel. Please see screenshot.

[image]

sfeakes, 2023-06-20:
“Check aquapure” Messages is directly from control panel.  The other message that was in a different image you posted “high salt” in the SWG icon, that comes directly from the SWG not the panel.

the latter you can sometimes see odd messages when the SWG is turned on or off depending on model.  If you don’t want this, set the config to not read SWG messages directly.

The former since it comes from the panel directly you should also see it in a keypad/one touch.  Maybe it’s clearing quickly so it’s being hidden / masked?  

Also note that a SWG is usually the device that’s most effected by RS485 bus issues so if you are still seeing checksum and serial reset errors, it’s most likely due to that

mcoutsoft, 2023-06-20:
Hmm.. keeps popping up, but no RS485 bus issues today. Will disable reading SWG directly and see if that helps, thank you!

sfeakes, 2023-06-20:
Please do. I just thought, It could also be something specific to the panel version# you have as well.

mcoutsoft, 2023-06-20:
Changed the config file (disabled read from SWG) and still getting the high salt warning that started when I installed the 2.3.1 build.  Strangely... the filter pump is off. So not sure how it's even reading from the SWG.

[image]

mcoutsoft, 2023-06-21:
Logged for a few hours and the salt level never went into 'high salt' level.  Very strange we didn't see this in the prior build. Going to see if I have a spare Aquapure sensor to swap out.

sfeakes, 2023-06-21:
Can you send the log, of when you see that message, I'd like to understand what's going on.

mcoutsoft, 2023-06-22:
Relocated the RS485 serial cable to the multiplexer using a dedicated connector. This seems to have helped the high salt event issue. Haven't seen this error since the cable was moved, will continue to monitor and report back.

mcoutsoft, 2023-06-22:
High salt is back. Swapped the sensor out and still getting the issue randomly.

sfeakes, 2023-06-22:
> Is there a way to clear the High Salt message? It looks like the SWG may send a sporadic high salt event which triggers the banner message, but then it never disappears (even with salt levels not being high)

The Aquapure message in the blue banner is from the control panel, and when the message disappears from the control panel, it should also disappear from the banner.  I'm guessing if you look in the logs, you will see that message being repeated quite frequently.  The clearing comes from a complete cycle in the panel messages without it being repeated. (a cycle is usually about 1 minute)

Keep monitoring your logs, my guess is you are seeing some "checksum" or "bad packet" errors, that's more than lightly what's causing the SWG errors you are seeing.  This will be down to wiring and/or termination.

>Is there a way to log the raw events coming from the SWG (ID 0x50) while the AqualinkD service is running and write it to a log vs running via command line (eg ./serial_logger /dev/ttyUSB0 -i 0x50)

Not without a 2nd USB2RS485 adapter and running the serial_logger on that.  BUT, you can log everything to a file, and then filter that out.  
This section of the wiki explains the options
https://github.com/sfeakes/AqualinkD/wiki#AqualinkD-debugging
ONE Below will do what you want, you don't need all 3.  My guess is `debug_RSProtocol_packets` is the one you are after.

```
debug_RSProtocol_packets=<yes/no>
debug_RSProtocol_bytes<yes/no>
debug_log_mask = 512
```

mcoutsoft, 2023-06-22:
@sfeakes Strange thing is.. neither the Aquapure control panel or my RS control panel indicate any errors / high salt, but the Aqualinkd high salt banner is showing.

sfeakes, 2023-06-22:
My guess is the all button keypad is, that is the protocol that AqualinkD get's its messages from.  These are the lines in the log with `RS Allbtn`. 

```
Info:    RS Allbtn: RS Message :- 'Check AQUAPURE High Salt General Fault'
```

You don't need to get down to debug messages to see them, `log_level=INFO` is enough or `debug_log_mask = 4`

If you monitor this, and post the debug logs from when you see it happening, I can give some more information and check it's not a issue with code.

mcoutsoft, 2023-06-22:
Hi @sfeakes I've got a Onetouch interface >> 

https://www.jandy.com/en/products/controls/system-components/interfaces/aqualink-onetouch

mcoutsoft, 2023-06-23:
Found some instances where high salt would come up for a second on the panel and then clear,  but the UI still shows high salt.

**Example 1** Salt is 3400 PPM = high salt warning.

Jun-20-23 09:20:49 AM Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 333
Jun-20-23 09:20:51 AM Info:    RS Allbtn: RS Message :- 'AIR TEMP 50 F'
Jun-20-23 09:20:53 AM Info:    RS Allbtn: RS Message :- 'POOL TEMP  73 F'
Jun-20-23 09:20:55 AM Info:    RS Allbtn: RS Message :- 'AquaPure 35%'
Jun-20-23 09:20:57 AM Info:    RS Allbtn: RS Message :- 'SALT 3400 PPM'
Jun-20-23 09:20:59 AM Info:    RS Allbtn: RS Message :- '06/20/23 TUE'
Jun-20-23 09:21:00 AM Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 331
Jun-20-23 09:21:01 AM Info:    RS Allbtn: RS Message :- '9:20 AM'
Jun-20-23 09:21:03 AM Info:    RS Allbtn: RS Message :- 'AIR TEMP 50 F'
Jun-20-23 09:21:05 AM Info:    RS Allbtn: RS Message :- 'POOL TEMP  73 F'
Jun-20-23 09:21:07 AM Info:    RS Allbtn: RS Message :- 'AquaPure 35%'
Jun-20-23 09:21:09 AM Info:    RS Allbtn: RS Message :- 'SALT 3400 PPM'
Jun-20-23 09:21:11 AM Info:    RS Allbtn: RS Message :- '06/20/23 TUE'
Jun-20-23 09:21:11 AM Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 332
Jun-20-23 09:21:13 AM Info:    RS Allbtn: RS Message :- '9:20 AM'
Jun-20-23 09:21:15 AM Info:    RS Allbtn: RS Message :- 'AIR TEMP 50 F'
Jun-20-23 09:21:17 AM Info:    RS Allbtn: RS Message :- 'POOL TEMP  73 F'
Jun-20-23 09:21:19 AM Info:    RS Allbtn: RS Message :- 'AquaPure 35%'
**Jun-20-23 09:21:21 AM Info:    RS Allbtn: RS Message :- 'SALT 3400 PPM'
Jun-20-23 09:21:22 AM Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 330
Jun-20-23 09:21:23 AM Info:    RS Allbtn: RS Message :- 'CHECK AquaPure HIGH SALT'**
Jun-20-23 09:21:23 AM Info:    RS Allbtn: RS Message :- '06/20/23 TUE'
Jun-20-23 09:21:25 AM Info:    RS Allbtn: RS Message :- '9:20 AM'
Jun-20-23 09:21:27 AM Info:    RS Allbtn: RS Message :- 'JANDY AquaLinkRS'
Jun-20-23 09:21:29 AM Info:    RS Allbtn: RS Message :- 'AIR TEMP 50 F'
Jun-20-23 09:21:31 AM Info:    RS Allbtn: RS Message :- 'POOL TEMP  73 F'
Jun-20-23 09:21:33 AM Info:    RS Allbtn: RS Message :- 'AquaPure 35%'

**Example 2** Salt PPM is 4200 for 2-4 seconds, shows in panel for <5 seconds and clears, but never clears from WebUI

Jun-20-23 12:03:47 PM Info:    RS Allbtn: RS Message :- 'AIR TEMP 58 F'
Jun-20-23 12:03:48 PM Info:    RS Allbtn: RS Message :- 'POOL TEMP  77 F'
Jun-20-23 12:03:50 PM Info:    RS Allbtn: RS Message :- 'AquaPure 35%'
Jun-20-23 12:03:52 PM Info:    RS Allbtn: RS Message :- 'SALT 3400 PPM'
Jun-20-23 12:03:53 PM Info:    PentaDvce: Pentair Pump Status message = RPM 1750 | WATTS 329
Jun-20-23 12:03:54 PM Info:    RS Allbtn: RS Message :- '06/20/23 TUE'
Jun-20-23 12:03:56 PM Info:    RS Allbtn: RS Message :- '12:03 PM'
Jun-20-23 12:03:59 PM Info:    RS Allbtn: RS Message :- 'JANDY AquaLinkRS'
Jun-20-23 12:04:00 PM Info:    RS Allbtn: RS Message :- 'AIR TEMP 58 F'
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 0 = EQUIPMENT STATUS
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 1 = 
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 2 =   AquaPure 35%  
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 3 =  SALT 4200 PPM  
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 4 =  CHECK AquaPure 
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 5 =   (High Salt)   
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 6 =   FILTER PUMP   
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 7 = Intelliflo VS 1 
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 8 =     RPM: 1750
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 9 =   WATTS: 329
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 10 = 
Jun-20-23 12:04:02 PM Info:    One Touch: OneTouch Menu Line 11 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 0 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 1 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 2 = SPA MODE     OFF
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 3 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 4 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 5 = CLEAN MODE   OFF
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 6 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 7 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 8 = ALL OFF         
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 9 = 
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 10 =       NEXT
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu Line 11 =      SYSTEM
Jun-20-23 12:04:03 PM Info:    One Touch: OneTouch Menu highlighted line = 10 =       NEXT
Jun-20-23 12:04:03 PM Info:    RS Allbtn: RS Message :- 'POOL TEMP  77 F'
Jun-20-23 12:04:04 PM Info:    RS Allbtn: RS Message :- 'AquaPure 35%'
[... 80 lines cut ...]
Jun-20-23 12:18:40 PM Info:    One Touch: OneTouch Menu Line 0 = EQUIPMENT STATUS
Jun-20-23 12:18:40 PM Info:    One Touch: OneTouch Menu Line 1 = 
Jun-20-23 12:18:40 PM Info:    One Touch: OneTouch Menu Line 2 =   AquaPure 35%  
Jun-20-23 12:18:40 PM Info:    One Touch: OneTouch Menu Line 3 =  SALT 3400 PPM  
Jun-20-23 12:18:40 PM Info:    One Touch: OneTouch Menu Line 4 =   FILTER PUMP   
[... cut ...]
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu Line 6 = 
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu Line 7 = 
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu Line 8 = ALL OFF         
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu Line 9 = 
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu Line 10 =       NEXT
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu Line 11 =      SYSTEM
Jun-20-23 12:18:42 PM Info:    One Touch: OneTouch Menu highlighted line = 10 =       NEXT

mcoutsoft, 2023-06-26:
Hi @sfeakes Would it be possible to put in similar logic for High Salt warning as you had for SWG zero values? I.e Ignore unless there are X in a row? Thanks!

closed 2025-11-19 by github-actions[bot]