aqualinkd/AqualinkD #225: Pentair VF pump speed Error:  One Touch: Can't set Pump RPM/GPM until type is known

mcoutsoft, 2023-06-22:
Trying to change the speed of the Pentair VF pump and the pump is not responding to the change.

Ran debug and I'm seeing the following.

Jun-21-23 17:18:56 PM Debug:   NetService:WS: Message - Key 'parameter' Value 'Filter_Pump'
Jun-21-23 17:18:56 PM Debug:   NetService:WS: Message - Key 'value' Value '1515'
Jun-21-23 17:18:56 PM Debug:   NetService:WS: Message - Key 'uri' Value 'Filter_Pump/RPM/set'
Jun-21-23 17:18:56 PM Debug:   NetService:WS: URI Request 'Filter_Pump/RPM/set': value 1515.00
Jun-21-23 17:18:56 PM Notice:  NetService:WS: request to change pump 1 RPM to 1515
Jun-21-23 17:18:56 PM Debug:   AqualinkD: Actioning delayed request
Jun-21-23 17:18:56 PM Notice:  Panl Prog: Starting programming thread 'Set OneTouch Pump RPM'
**Jun-21-23 17:18:56 PM Error:   One Touch: Can't set Pump RPM/GPM until type is known**

Any suggestions on what I can check? Thanks!

sfeakes, 2023-06-22:
Have you set the pump type in the config?

mcoutsoft, 2023-06-22:
Yes. I can see GPM/RPM, but it still gets this error.

sfeakes, 2023-06-22:
Can you send complete debug logs from startup to changing RPM/GPM, making sure you see the RPM/GPM in the UI after startup.

mcoutsoft, 2023-06-22:
Will do.

mcoutsoft, 2023-06-23:
@sfeakes Log attached

Screenshot 1- Web UI Showing RPM before change
Screenshot 2- Web UI Showing RPM after change
Controller PIC- Onetouch control panel after change

[image]
[image]
[image]
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11842020/aqualinkd.log)

sfeakes, 2023-06-23:
Thanks for the debug logs.  I see the problem.   The panel version you have `REV. O.2` is displaying the VSP information very differently to all the other panel versions I've come across on the OneTouch protocol.  And that's the protocol that AqualinkD is configured to use for VSP.  Looking over the notes I have this is the first panel version that supported VSP on OneTouch. (Typical Jandy things are different between this version and the later versions of the protocol AqualinkD is programmed to use).
The other option is to use the Touch Protocol for VSP, but your panel doesn't support that. 

I will have to modify the code to get VSP working for you.   I'll post an update, but can't give a time/date yet.

mcoutsoft, 2023-06-23:
Thanks for taking a look. Appreciate all of your help and work on this project!

sfeakes, 2023-06-23:
I just pushed Version 2.3.2 that has support for this panel# and VSP.

I have no way of testing this, so coded against your logs, let's hope this works.  Below are the messages to look for in the logs if everything has worked.
```
One Touch: Setting early version for OneTouch
One Touch: Matched OneTouch Pump to 'Filter_Pump', Index 1, RPM 2250, Watts 100, GPM 50
One Touch: OneTouch Pump index 0 set PumpType to 0
```

If this works, the next problem might be the programming of the RPM/GPM.  It quite lightly If the status menu is different on your panel version that the programming is as well.

mcoutsoft, 2023-06-23:
Thanks for the update. Doesn't look like it is generating the log events we'd hoped for. Please see attached.
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11853608/aqualinkd.log)

sfeakes, 2023-06-24:
Stupid error on my behalf, Just updates 2.3.2, so please try again.   Since it's same version# make sure you force the latest.

mcoutsoft, 2023-06-24:
Did a fresh install of 2.3.2 
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11854472/aqualinkd.log)
and not seeing any of the following

One Touch: Setting early version for OneTouch
One Touch: Matched OneTouch Pump to 'Filter_Pump', Index 1, RPM 2250, Watts 100, GPM 50
One Touch: OneTouch Pump index 0 set PumpType to 0

sfeakes, 2023-06-24:
OK,  pushed a change.  please try again.

mcoutsoft, 2023-06-26:
@sfeakes Was able to test this morning..looks like some progress! The RPM value is now showing in the button to start /stop filter pump.

When I go to change the setting (i.e. to 1800 RPM) it says programming pump and then the RPM label always switches to 2750, no matter what value I am selecting on the slider.

Attached the logs, hope this helps!
[aqualinkd_0626.log](https://github.com/sfeakes/AqualinkD/files/11871086/aqualinkd_0626.log)

sfeakes, 2023-06-26:
Glad we got a bit further, can you post debug log (attached is Info) of when you're trying to change RPM, that way I can see what's going on.

mcoutsoft, 2023-06-26:
Restarted in debug mode and the filter pump button no longer shows RPM.  Log attached
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11872799/aqualinkd.log)

mcoutsoft, 2023-07-07:
Hi @sfeakes Hope you're doing well. Just wanted to check in and see if you needed any additional logging from me. Thanks!

sfeakes, 2023-07-07:
From your last set of logs, AqualinkD did not connect to the panel using the OneTouch protocol.  Only the Allbutton and SerialAdapter protocols.  Since OneTouch is the protocol that gives VSP information, that's why it didn't show up.  But I can see from your logs, it's not a configuration error, the panel never tried to start the one touch protocol.

Can you try restarting AqualinkD, and see if you get notices like in the logs.
```
Debug: One Touch:   .....
```

closed 2025-11-19 by github-actions[bot]