aqualinkd/AqualinkD #227: Jandy ePump on PDA not collecting correct RPMs

BobClav, 2023-06-22:
Im having an issue where my PDA-6 system is not presenting the correct RPM value on the filter pump. Will it cycle the watt usage as well?   In the debug logs, it seems to show the correct value, but not mapped correctly.   

From aqualinkd.conf:
button_01_label=FILTER PUMP
button_01_pumpIndex=1
button_01_pumpID=0x78

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11840692/aqualinkd.log)

Any help, advice, or setting changes would be appreciated.

sfeakes, 2023-06-23:
In the debug logs you sent, it never for to the "equipment status" menu which is where the RPM is shown, it only get's to the "equipment" menu.   Leave it running for longer, and please post the logs.
At the moment, AqualinkD can only read the pump RPM / GPM directly from Pentair pumps, and you have a Jandy pump, so AqualinkD has to wait for the message to be shown in the equipment status. 

Even if this works, please do post a longer set of logs, and even better if you can post with changing the RPM (on the PDA or AqualinkD).  That way I can figure out the Jandy ePump protocol and make AqualinkD faster at reading those RPM values.

As got GPM, AqualinkD will only display the value that can be changed, with a ePump, I believe that's RPM and not GPM.

BobClav, 2023-06-23:
Thanks for the quick response.   Attached is another debug log that shows the ePump values that would appear on the screen.   Also note that it displays power usage (Watts).   It would be awesome to have the icon toggle back and forth the RPMs and then WATTS.  

Back to the issue, the pump motor icon shows RPM as 520, which is in the logfile, just the wrong value selected.   Is it possibly the naming I am using for the ePump?   

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11853669/aqualinkd.log)

sfeakes, 2023-06-24:
Thank.  I see the problem, it'll take a bit to fix though.  In the mean time, if you get chance, can you post exactly the same log, but with a different RPM/GPM on the VSP?
Also do you know the version# of your panel?

BobClav, 2023-06-24:
PDA: 7.1.0
J-Box: 4.0
Handheld: 3.0

I ran through 3 different speeds SPEED4: @3450, SWEEP: @2900 and back to POOL: @2520 .  Note that looking at the debug log, the RPMs do not show when the PDA is active, but WATTS do.   I shut off the PDA and the RPMs show up then.

Let me know if you need to to try anything else.

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/11857170/aqualinkd.log)

sfeakes, 2023-06-24:
Thanks so much, very helpful.  I'm more interested in the below type messages when you have set the speed on the PDA, vs what the PDA states it's set to. And I can see all that in the log.    The HEX is what the control panel is sending the pump, and what the pump is returning.  This allows me to try to decode the ePump protocol.   
```
Debug:   JandyDvce: To   ePump:  Read To 0x78 of type   Unknown '0x44' | HEX:0x10|0x02|0x78|0x44|0x00|0x60|0x27|0x55|0x10|0x03|
Debug:   JandyDvce: From ePump:  Read To 0x00 of type   Unknown '0x1f' | HEX:0x10|0x02|0x00|0x1f|0x44|0x00|0x60|0x27|0x00|0xfc|0x10|0x03|
Debug:   JandyDvce: To   ePump:  Read To 0x78 of type   Unknown '0x41' | HEX:0x10|0x02|0x78|0x41|0xcb|0x10|0x03|
Debug:   JandyDvce: From ePump:  Read To 0x00 of type              Ack | HEX:0x10|0x02|0x00|0x01|0x41|0x00|0x54|0x10|0x03|
```

sfeakes, 2023-06-24:
@BobClav I just posted an update that should display VSP info from the PDA menu.   Still working on decoding protocol so I can read vPump into directly.

BobClav, 2023-06-24:
If you need/want access, let me know. Think you have direct contact info in github.

sfeakes, 2023-06-24:
Thanks very much, I hope that shouldn't be necessary. Give me a few days to see if I can decode the information you have already logged for me.