aqualinkd/AqualinkD #230: Can't change name of Aux 1 on web UI

42Cyborg, 2023-06-26:
Hello, I'm sure I am missing something obvious but my Aux 1 button always says "Cleaner" no matter what I change in the `~/software/AqualinkD/release/aqualinkd.conf`

I want it to say "Spa Blower"
all of the buttons that could possibly be the Aux 1 button are set to spa blower in the aqualinkd.conf and yet it still says cleaner

`button_01_label=Spa Blower

button_02_label=Spa Blower

button_03_label=Spa Blower

button_04_label=Spa Blower`

[image]

I am running the prebuilt version on a raspberry pi 4. Let me know if there's any other information that would help diagnose this. Thank you

BobClav, 2023-06-26:
The file you need to edit is /etc/aqualinkd.conf .  Try that one instead.

42Cyborg, 2023-06-26:
That worked, thank you so much! I knew it was something simple

closed 2023-06-26 by 42Cyborg