aqualinkd/AqualinkD #231: Max spa temp is currently 96

42Cyborg, 2023-06-26:
Hello, I am trying to adjust the spa temp from the web UI, and while it is able to control the heater temp and turn it on and off, it will not set the temperature above 96º f. I can slide it all the way to 104º, it says "programming heater" then sets it to 96º. it has no issue programming any value I want that is lower than 96

I am running prebuilt on raspberry pi 4
Thank you

(tangentially related, is there a way to set the range of values on the web UI slider to only go down to 70º? it would just make it easier for me to use)

sfeakes, 2023-06-27:
Two things will help.
Can you post your /etc/aqualinkd.conf file
And can you set logging to debug, try to set the temp, wait for the “programming” message to clear, and post the log.

at the moment their is no way to set the slider range.  But I’ll take a look for future release.

42Cyborg, 2023-06-27:
once I set the log to debug and went to pull it, I found the error: `Jun-26-23 21:01:29 PM Debug:   Panl Prog: found message 'MUST BE SET' in 'Temp2 must be set lower than Temp1... set Temp1 first'
`
This seems to be a setting on the panel itself, so not an aqualinkd issue, sorry for not checking that first.

On the topic of adjustable slider bounds, is there a way I should be making a note of that in another github ticket, and closing this one? or just close this ticket?

sfeakes, 2023-06-27:
Yup Temp1 / Temp2 is some strange limitation Jandy put on a single panel (non pool/spa) for some reason.  But AqualinkD should have caught that and given you the warning.  Makes me think you might have the panel type configured incorrectly.
What do you have for panel_type in /etc/aqualinkd.conf.  You should have something with `Only` at the end, not `Dual` or `Combo` ie `panel_type = RS-8 Only`

Slider bounds are already done, was a simply 10 min thing to implement.  It will be in next release.

42Cyborg, 2023-06-28:
I see, yep, mine was set to `combo`, not `only`, so that explains it! Thank you so much! and amazing to hear about the slider bounds

closed 2023-06-28 by 42Cyborg

sfeakes, 2023-07-08:
Release 2.3.2 has the option to set slider max/min values.   So if you get the latest, your should be good to go.
The install script will NOT overwrite your '/var/www/aqualinkd/config.js', so either copy over that manually from the latest, or add the below lines to your config.js (around line 100)
```
      var heater_slider_min = 36;
      var heater_slider_max = 104;
```