aqualinkd/AqualinkD #236: AqualinkD Preventing Aqualink from Setting Correct Light Color

maniscalcojohn, 2023-07-17:
I have AqualinkD set up with my Jandy LEDs using the Aqualink panel to set the color (lightMode = 0), however, Aqualink is unable to cycle to the correct color unless I shutdown the aqualinkd service. Watching the lights cycle through their colors, when AqualinkD is active, eventually one or more of the lights return to white part of the way through their cycle and then some of them are completely out of sync and the incorrect color.

sfeakes, 2023-07-18:
lightmode=0 means AqualinkD will set the light color, not the control panel.  Sounds like you want the control panel to set the light mode (I assume that's what you mean by Aqualink), in which case you should be using a light mode other than 0 in AqualinkD configuration.

But in either case, AqualinkD will not interfere with the control panel setting the light mode. So having to shutdown AqualinkD for the control panel to set the light mode, probably means something else is going on. Or I am understanding you problem incorrectly.

The reason you need to set lightmode correctly is so AqualinkD can program the light.  It needs to know how the control panel has it configured. So that setting will ONLY effect AqualinkD setting the light, it has no effect  (or interference) on how the control panel controls sets the light.

sfeakes, 2023-07-18:
If you configure AqualinkD to Debug log, and log to file.  Then start AqualinkD, give it a minute or two to set itself with control panel.  Then try to program the light from the control panel / iAqualink / keypad etc, stop AqualinkD and post the log.  I can take a look and see if I can tell when the problem is.

maniscalcojohn, 2023-07-19:
Thanks. Here's the log. Something is causing the lights to occasionally take too long to power cycle, which results in the lights resetting to white part of the way though the cycle.

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/12090599/aqualinkd.log)

sfeakes, 2023-07-19:
I went over the logs, and I can’t see anything where AqualinkD is interfering with the control panel programming the light, in-fact AqualinkD has not even seen the light change state in those logs.
What device are you using the change the light state? Ie keypad / OneTouch / iAqualink etc?

maniscalcojohn, 2023-07-19:
Same. I read through the log before I posted it looking for something regarding the lights but could not find anything. 

I did this test with the Jandy iAqualink app, but fairly certain it did the same with AqualinkD. 

Could it be related to MQTT?

sfeakes, 2023-07-20:
Since AqualinkD did not see the light even chance, it’s not related to MQTT.

Just to confirm the below is what you are experiencing.
AqualinkD not running, changing the light mode with iAqualink app, Light changes mode correctly.
AqualinkD running, changing the light mode with iAqualink app, Light does not cycle mode correctly.

maniscalcojohn, 2023-07-20:
Correct. At a point in the cycle, the lights will stay off for too long and reset to white.

maniscalcojohn, 2023-07-22:
Here's the full Serial Debug log if that helps...

[aqualinkd2.log](https://github.com/sfeakes/AqualinkD/files/12136710/aqualinkd2.log)

I also disabled MQTT and still having the same issue.

maniscalcojohn, 2023-07-22:
And same problem using the web interface of Aqualinkd

sfeakes, 2023-07-24:
I noticed on log 3 & 4 you used AqualinkD to set the light and not iAqualink.  Both times it was set to “Disco Tech” and the control panel thought the light was already on “Disco Tech”, so it looked like it just turned it on rather than actually programming it.

I really can not see where AqualinkD is interfering with the light at any point, One thing to try would be to take AqualinkD out of extended programming, by commenting out or deleting this line in your config 

“extra_device_id   = 0x31”.

Make sure to restart aqualinkd once you take that line out.  You should keep using the control panel to program the light and not AqualinkD, get that working first.
Other things to try.
1) update AqualinkD to the latest version, you are a few behind.
2) If the above makes no change, it would be worth knowing if the control panel can program the light with serial logger running rather than aqualinkd.

Stop aqualinkd, run serial logger, use iAqualink to program the light.

maniscalcojohn, 2023-09-11:
Finally got around to working on this again...

Updating to the latest version did not fix the issue--see log 6

Removing "extended_device_id=0x31" and "extended_device_id_programming = yes" **fixed** the problem--see log 7

Unfortunately, as your documentation points out, this prevents me from setting my pump rpm, which is pretty much the primary reason I set up AqualinkD... 

[aqualinkd6.log](https://github.com/sfeakes/AqualinkD/files/12571602/aqualinkd6.log)

[aqualinkd7.log](https://github.com/sfeakes/AqualinkD/files/12571591/aqualinkd7.log)

maniscalcojohn, 2023-09-11:
I am able to re-enable "extended_device_id=0x31" only and the lights cycle properly. Also have access to my pump RPMs, so this may work. Not sure what I am missing by not having extended_device_id_programming.

maniscalcojohn, 2023-09-11:
Nvm. Its not working with extended_device_id enabled. Same problem.

fd-80, 2025-01-21:
I'm running an Aqualink Rev T.2 and I'm also encountering the above.  Along with the Aqualink RS8 I have an iAqualink 3.0 connected to it.  When trying to change lights with the Aqualinkd service started it often exhibits similar behavior as described by @John5373 - (ie the pause for a change is too long and it resets to the initial White mode).  If I shut down the Aqualinkd service and change the light color in the iAqualink app it works fine.... but once I have the AqualinkD with the extended_device_id it does not change the light colour correctly - but it works?

sfeakes, 2025-01-21:
Can you post your aqualinkd.conf file.  Also did you try the `rs485_frame_delay` posted above?

If that doesn't work,  can you try  using `extended_device_id` with a one Touch ID and not Aqualink Touch ID (so use one of 0x40, 0x41, 0x42, 0x43 and not 0x30, 0x31, 0x32, 0x33).

fd-80, 2025-01-22:
Sorry just thought I should also clarify I have a Jxi heater and the Jandi VSP pump which works well in Aqualinkd showing the RPM etc.

I tried the `rs485_frame_delay ` and the light problem is still there (I even tried pushing that up higher to 20)

I've had to comment out the #enable_iaqualink=yes as when I enable it - it shuts down my iAqualink 3.0 (but I also am unsure whether I should have it enabled or not)

However when I change the  `extended_device_id` to a Aqualink Touch ID the light colour change problem does not occur (but I lose the ability to use the OneTouch Buttons/Virtual buttons to turn the spa on 😢 ) so its definitely only a problem when using an Aqualink Touch ID

I've also tested using by disconnecting the iAqualink 3.0 from the 485 Bus - when I disconnect it and use an Aqualink Touch ID - the light colour change works correctly!

aqualinkd.conf here:
[aqualinkd.conf.txt](https://github.com/user-attachments/files/18499402/aqualinkd.conf.txt)

sfeakes, 2025-01-22:
That confirms what I thought it might be. It's a problem with the panel running two iAqualink Touch protocols + iAqualink (2.0 or 3.0), it probably can't handle all the messages while it's programming a light.  One thing to try would be to set AqualinkD `extended_device_id` to One touch (0x4?), fire up AqualinkD iAqualink Touch  simulator, and set the light color through iAqualink 3.0 or iAqualink Touch panel / web.  If that works, then I can look at reducing the number of messages AqualinkD uses on the Aqualink touch protocol.  ( The Simulator will use less messages on the iAqualink Touch protocol)
Problem is either Panel firmware bug, (there are plenty of those I've had to code around) or the CPU isn't powerful enough.

It's been on my todo list to get virtual buttons working using the OneTouch protocol, so we could also get around it that way, it'll just be a little slower to turn them on/off.

fd-80, 2025-01-23:
Apologies I just realised I had a typo in my 4th paragraph but I believe you realised what I meant that I tried with the OneTouch protocol (`extended_device_id=0x40`) !  

I just tried setting the 0x40 again and firing up the `One Touch simulator ` and also the iAqualink app - using either and setting the light to say USA works correctly using both methods with this setup.  

When returning to use a  0x3x` extended_device_id `it definitely causes the lighting issues (and to be fair also makes the iAqualink app work slowert to get updated)

sfeakes, 2025-01-29:
I just posted a development update if you want to try that.  This should reduce the load on the control panel using ID's 0x3?, and I believe (or hope) this will fix the issue.

Since it's a dev update, you'll need to force latest if using the aqua.sh script to install.

davearias, 2025-07-28:
Just got my AqualinkD back up and running (love the new HAT!), and I am seeing this issue persist on version 2.6.9. Light program changes appear to work correctly when using an extended_device_id of 0x40, but not when using 0x30. I'm using the iAqualink app to set the light programs.

Including config and log download, please let me know if I can provide more info! Assuming you can parse it out, first I set the light programs on my three lighting circuits with the extended_device_id set to 0x40, and then I turned them off, updated the config to extended_device_id of 0x30, and set the lighting programs again.

[aqualinkd (4).log](https://github.com/user-attachments/files/21474491/aqualinkd.4.log)
I grabbed 15k lines of the log, around 13:27 is where Ithe lights set correctly, around 13:36 is where the lights have an issue because the time between off/on cycles is too long.

[aqualinkd.conf.txt](https://github.com/user-attachments/files/21474193/aqualinkd.conf.txt)

sfeakes, 2025-07-28:
What I see in your logs (For when you had extended is = 0x30)

1. 13:30:04 - start up
2. 13:32:32 - All light set to Fast Splash
3. 13:34:44 - All lights off
4. 13:34:57 - Bubbler light sets to Emerald Green
5. 13:35:45 - All light off
6. 13:35:41 - Pool light set to Slow Splash
7. 13:35:57 - All lights set to slow plash.

Which of those worked as expected / what where you setting from the app?

Also, to debug this further I will need some more logs.  Can you post the output of serial_logger and I can provide some more instructions.

davearias, 2025-07-29:
Here's some new logs with better timestamps:
[aqualinkd-0x40 tests.log](https://github.com/user-attachments/files/21478699/aqualinkd-0x40.tests.log)
Configured as 0x40:
20:26:50 - Restarted AqualinkD (approx)
20:31:30 - Set all circuits to Magenta, OK
20:30:30 - Turned all circuits OFF, OK
20:33:00 - Set all circuits to Slow Color Splash, OK
20:34:30 - Turned all cirucits OFF, OK
20:35:00 - Set all circuits to Fat Tuesday, OK
20:36:30 - Turned all circuits off, OK

[aqualinkd-0x30 tests.log](https://github.com/user-attachments/files/21478732/aqualinkd-0x30.tests.log)
Configured as 0x30:
20:38:30 - Restarted Aqualinkd (approx)
20:41:30 - Set all circuits to Slow Color Splash, OK
20:43:30 - Turned all circuits OFF, OK
20:44:00 - Set all circuits to Fat Tuesday, OK
20:45:30 - Turned all circuits OFF, OK
20:46:00 - Set all circuits to Magenta, This did not set correctly
20:47:10 - Turned all circuits OFF, OK
20:47:30 - Set all circuits to Slow Color Splash, OK
20:49:00 - Turned all circuits OFF, OK
20:49:15 - Set to all circuits to Cobalt, Spa and Pool OK, Bubblers not ok (reset to white)
Last change I turned off the Bubblers circuit and then set it to Cobalt again, and it set correctly

Seems it's intermittent when the timing of the on/off cycles is within tolerance or not.

Log file with serial-logger data:
[serial-logger.log](https://github.com/user-attachments/files/21478735/serial-logger.log)

FYI I may not be able to do any more testing for a few days, but I can certainly try remotely.

davearias, 2025-08-06:
New log file with the debug mask as requested:
[aqualinkd-0x30-debugmask.log](https://github.com/user-attachments/files/21630123/aqualinkd-0x30-debugmask.log)

14:28 - Restarted AqualinkD
14:30 - Set all circuits to Fast Color Splash, lost panel comms via iAqualink
14:31:15 - Comms restored, panel started setting light program and failed
14:32:15 - All circuits off
14:32:30 - Set all circuits to Fast Color Splash, Failed
<turned off all circuits>
14:33:40 - Set all circuits to Fast Color Splash, Failed - This time I took note of when the extended delay occured. The cycle delay was after the 6th cycle (reset, white, Sky Blue, Cobalt Blue, Caribbean Blue). The panel continued cycling after that, but of course stopped on the wrong program.

brianmaas, 2026-03-01:
Thanks @sfeakes - this works for me! I upgraded to dev6, set extended_device_id=0x3, and set light_programming_iaqualink_delay=YES

I did a bunch of different light color suggestions from home assistant, iaqualink, and aqualinkd, and they all worked. Nice work.