aqualinkd/AqualinkD #239: Service failed "no such file or directory

originalprime, 2023-08-02:
Hello!

I've been running this delightful software successfully since May. However yesterday we had brief power outage. After the power was restored, I noticed that no data was present on the web gui for aqualinkd. I hit the terminal and came to find out that the aqualinkd.service wasn't running! Weird.

I told the service to start, but I am now met with:

`● aqualinkd.service - Aqualink RS daemon
     Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-08-02 19:21:05 EDT; 12s ago
    Process: 5130 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=203/EXEC)
        CPU: 2ms

Aug 02 19:21:05 raspberrypi systemd[5130]: aqualinkd.service: Failed at step EXEC spawning /usr/local/bin/aqualinkd: No such file or directory
Aug 02 19:21:05 raspberrypi systemd[1]: Starting Aqualink RS daemon...
Aug 02 19:21:05 raspberrypi systemd[1]: aqualinkd.service: Control process exited, code=exited, status=203/EXEC
Aug 02 19:21:05 raspberrypi systemd[1]: aqualinkd.service: Failed with result 'exit-code'.
Aug 02 19:21:05 raspberrypi systemd[1]: Failed to start Aqualink RS daemon.`

I'm not sure what to do about this. I have verified that the USB-to485 adaptor is still recognized, still chatting, and still reporting as USB0. All are true. And for what it's worth, the adaptor's blinkenlights are going nuts, as usual.

Is there something silly going on here, or did the power outage do something serious?

I'm running on a Pi 4 with a few other lightweight services, all of which continue to report correctly, which implies that the Pi itself is fine.

Again, it's been running swimmingly well for months. Any thoughts on the matter would be appreciated.

sfeakes, 2023-08-03:
Are you using an SD card?  If so looks like the SD card is missing some files that were running at the time of the outage. Ie  AqualinkD.    Quick way round is to reinstall AqualinkD.   But you may want to check the sd card as corruption is quit common.
If USB drive, then same is true, reinstall AqualinkD, but the corruption is probably less of a problem.

brownjd, 2023-08-05:
> Another way to check the file is to run AqualinkD manually. `/usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf`
> 
> The only other time I can think of where you would get a similar error is if you have a 64bit OS, 32bit binary and no 32bit runtime installed/supported. But since you said it was just a power outage and no OS update, this probably isn't the case.

I'm having the same problem since upgrading to 2.3.2 from 2.2.x. I can run manually, but the systemctl command just hangs. ps doesn't show the process running.

It worked swimmingly before the upgrade.

originalprime, 2023-08-05:
Okie dokie. I've had a few days to do some work on this.

Firstly, thank you @sfeakes for taking your time to reply.

I am running on an SD card, and I recognize that they can be finicky. I ran the fsck command on the boot partition and the data partition, and it indicated the partitions were fine. However, since I was feeling spicy, I reformatted the SD card anyway.

It's running the May 2023 variant of Raspberry Pi OS Lite (64-bit), as before, and on a fresh install using the "Simple Install; Trust Everything" method, I am continuing to experience errors. The `journalctl -xe` command throws up the following. I have removed my username for privacy, and the system is called "scrypted" because I intend it to run Scrypted as well, as it was doing before the Aqualinkd error(s) occured:

`Aug 05 13:51:41 scrypted systemd[1]: Started Session 4 of user USERNAME.
░░ Subject: A start job for unit session-4.scope has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit session-4.scope has finished successfully.
░░ 
░░ The job identifier is 1028.
Aug 05 13:52:19 scrypted sudo[1068]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME ; USER=root ; COMMAND=/usr/bin/cp aqualinkd.con /etc/
Aug 05 13:52:19 scrypted sudo[1068]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
Aug 05 13:52:19 scrypted sudo[1068]: pam_unix(sudo:session): session closed for user root
Aug 05 13:52:23 scrypted sudo[1070]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME ; USER=root ; COMMAND=/usr/bin/cp aqualinkd.con /etc
Aug 05 13:52:23 scrypted sudo[1070]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
Aug 05 13:52:23 scrypted sudo[1070]: pam_unix(sudo:session): session closed for user root
Aug 05 13:52:38 scrypted sudo[1080]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME ; USER=root ; COMMAND=/usr/bin/cp aqualinkd.conf /etc/
Aug 05 13:52:38 scrypted sudo[1080]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
Aug 05 13:52:38 scrypted sudo[1080]: pam_unix(sudo:session): session closed for user root
Aug 05 13:52:52 scrypted sshd[1042]: pam_unix(sshd:session): session closed for user USERNAME
Aug 05 13:52:52 scrypted systemd[1]: session-4.scope: Succeeded.
░░ Subject: Unit succeeded
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit session-4.scope has successfully entered the 'dead' state.
Aug 05 13:52:52 scrypted systemd-logind[463]: Session 4 logged out. Waiting for processes to exit.
Aug 05 13:52:52 scrypted systemd-logind[463]: Removed session 4.
░░ Subject: Session 4 has been terminated
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ Documentation: sd-login(3)
░░ 
░░ A session with the ID 4 has been terminated.
Aug 05 13:53:01 scrypted systemd[1]: Configuration file /etc/systemd/system/aqualinkd.service is marked executable. Please remove executable>
Aug 05 13:53:03 scrypted polkitd(authority=local)[433]: Registered Authentication Agent for unix-process:1083:39501 (system bus name :1.19 [>
Aug 05 13:53:06 scrypted polkit-agent-helper-1[1094]: pam_unix(polkit-1:auth): authentication failure; logname= uid=1000 euid=0 tty= ruser=h>
Aug 05 13:53:08 scrypted polkitd(authority=local)[433]: Operator of unix-process:1083:39501 FAILED to authenticate to gain authorization for>
Aug 05 13:53:08 scrypted polkitd(authority=local)[433]: Unregistered Authentication Agent for unix-process:1083:39501 (system bus name :1.19>
Aug 05 13:53:09 scrypted dhcpcd[697]: eth0: Router Advertisement from fe80::18c4:5f35:228c:1d59
Aug 05 13:53:15 scrypted dhcpcd[697]: eth0: Router Advertisement from fe80::18c4:5f35:228c:1d59
Aug 05 13:56:49 scrypted sudo[1146]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME/software/AqualinkD-2.3.2 ; USER=root ; COMMAND=/usr/bin/nano a>
Aug 05 13:56:49 scrypted sudo[1146]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
Aug 05 13:56:55 scrypted sudo[1146]: pam_unix(sudo:session): session closed for user root
Aug 05 13:57:21 scrypted sudo[1159]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME/software/AqualinkD-2.3.2/release ; USER=root ; COMMAND=/usr/bi>
Aug 05 13:57:21 scrypted sudo[1159]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
Aug 05 13:57:58 scrypted sudo[1159]: pam_unix(sudo:session): session closed for user root
Aug 05 13:58:14 scrypted sudo[1175]:  USERNAME : TTY=pts/0 ; PWD=/ ; USER=root ; COMMAND=/usr/sbin/service aqualinkd start
Aug 05 13:58:14 scrypted sudo[1175]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
Aug 05 13:58:14 scrypted systemd[1]: Configuration file /etc/systemd/system/aqualinkd.service is marked executable. Please remove executable>
Aug 05 13:58:16 scrypted systemd[1]: Starting Aqualink RS daemon...
░░ Subject: A start job for unit aqualinkd.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit aqualinkd.service has begun execution.
░░ 
░░ The job identifier is 1097.
Aug 05 13:58:16 scrypted systemd[1183]: aqualinkd.service: Failed to execute /usr/local/bin/aqualinkd: No such file or directory
Aug 05 13:58:16 scrypted systemd[1183]: aqualinkd.service: Failed at step EXEC spawning /usr/local/bin/aqualinkd: No such file or directory
░░ Subject: Process /usr/local/bin/aqualinkd could not be executed
░░ Defined-By: systemd
[... cut ...]
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit systemd-tmpfiles-clean.service has finished successfully.
░░ 
░░ The job identifier is 1163.`

I tried to install Aqualinkd first with nothing other than the host OS running, and that is what happened. Interestingly, Scrypted reinstalled just fine and is up and running again. 

I see @brownjd commenting about the newer version of Aqualinkd, so I am going to try to do a manual install / pull of the last 2.2.x version next.

originalprime, 2023-08-05:
Quick update before I have to stop tinkering for the evening:

I found a forked repo at [https://github.com/ralton66/AqualinkD](url) on version 2.2.x so I gave that a go after uninstalling the 2.3.2 version from @sfeakes repo. Upon startup, the `journalctl -xe` output looks much cleaner:

> Aug 05 16:21:02 scrypted sudo[7609]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME ; USER=root ; COMMAND=/usr/bin/cp aqualinkd.conf /etc/
> Aug 05 16:21:02 scrypted sudo[7609]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
> Aug 05 16:21:02 scrypted sudo[7609]: pam_unix(sudo:session): session closed for user root
> Aug 05 16:21:14 scrypted sudo[7611]:  USERNAME : TTY=pts/0 ; PWD=/home/USERNAME ; USER=root ; COMMAND=/usr/sbin/service aqualinkd start
> Aug 05 16:21:14 scrypted sudo[7611]: pam_unix(sudo:session): session opened for user root(uid=0) by USERNAME(uid=1000)
> Aug 05 16:21:14 scrypted systemd[1]: Configuration file /etc/systemd/system/aqualinkd.service is marked executable. Please remove executable permission bits. Proceeding anyway.
> Aug 05 16:21:16 scrypted systemd[1]: Starting Aqualink RS daemon...
> ░░ Subject: A start job for unit aqualinkd.service has begun execution
> ░░ Defined-By: systemd
> ░░ Support: https://www.debian.org/support
> ░░ 
> ░░ A start job for unit aqualinkd.service has begun execution.
> ░░ 
> ░░ The job identifier is 1590.
> Aug 05 16:21:16 scrypted aqualinkd[7619]: Error:   AqualinkD: Unknown config parameter 'enable_scheduler = yes'
> Aug 05 16:21:16 scrypted aqualinkd[7619]: Error:   AqualinkD: Unknown config parameter 'force_PS_setpoints = yes'
> Aug 05 16:21:16 scrypted aqualinkd[7619]: Error:   AqualinkD: Unknown config parameter 'force_Frzprotect_setpoints = no'
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Aqualink Daemon v2.2.2a
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Panel set to RS-8 Combo Pool/Spa
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config log_level         = 5
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config device_id         = 0x0a
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config rssa_device_id    = 0x48
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config extra_device_id   = 0x32
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config extra_device_prog = YES
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config serial_port       = /dev/ttyUSB0
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config socket_port       = 8080
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config web_directory     = /var/www/aqualinkd/
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config use_aux_labels    = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config override frz prot = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_server       = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_dz_sub_topic = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_dz_pub_topic = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_aq_topic     = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_user         = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_passwd       = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config mqtt_ID           = aqualinkd_e45f0143
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config idx water temp    = -999
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config idx pool temp     = -999
> Aug 05 16:21:16 scrypted systemd[1]: Started Aqualink RS daemon.
> ░░ Subject: A start job for unit aqualinkd.service has finished successfully
> ░░ Defined-By: systemd
> ░░ Support: https://www.debian.org/support
> ░░ 
> ░░ A start job for unit aqualinkd.service has finished successfully.
> ░░ 
> ░░ The job identifier is 1590.
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config idx spa temp      = -999
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config idx SWG Percent   = 0
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config idx SWG PPM       = 0
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config force SWG         = YES
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config deamonize         = YES
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config log_file          = (null)
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config light_pgm_mode    = 0.00
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Debug RS485 protocol     = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Display warnings in web  = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Keep panel time in sync  = YES
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Read SWG direct          = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Read ePump direct        = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Read vsfPump direct      = NO
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Thread Network Services  = YES
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | VSP ID 0x78 | PumpID 1 |
> Aug 05 16:21:16 scrypted aqualinkd[7619]: AqualinkD: Config BTN Spa_Mode      = label Spa Mode        |
[... cut ...]
...which indicates that it's happier, however, when in the web interface I'm still stuck with "Connecting to control panel" as the status on the top of the page. No data loads in the web GUI, and clicking on the buttons does not action anything.

So much for that idea, but I thought it was a good one.

Even though I have other services up and running just fine, and fsck indicated my SD card was fine, I've ordered a T7 to use as a boot drive instead. Hopefully I'll have that in hand in a few days and I can try again.

brownjd, 2023-08-05:
> I found a forked repo at https://github.com/ralton66/AqualinkD on version 2.2.x so I gave that a go after uninstalling the 2.3.2 version from @sfeakes repo. Upon startup, the journalctl -xe output looks much cleaner:

You are getting further than I am. It doesn’t even seem to start the process.

Jason

sfeakes, 2023-08-06:
@originalprime @brownjd   I just updated the latest 2.3.2 binary, I did see a problem with how it was compiled that may cause problems on some systems.  You may want to update.
But neither of the problems you are describing will be fixed by this, unless their is something missing in descriptions.

@originalprime ”connecting to control panel” indicates that it has not yet established a connection to the control panel, usually this is the RS485 connection.
But it does look like you are using a 2.3.x config file with a 2.2.x binary.  I doubt that would cause this issue.

@brownjd   If AqualinkD command line works, but the service doesn’t, can you post the logs / output from systemd

brownjd, 2023-08-06:
> @brownjd If AqualinkD command line works, but the service doesn’t, can you post the logs / output from systemd

I downloaded both the new 2.3.2 (using aqua.sh) and also the development release. Both yielded the same results.

Here's the result from sudo journalctl -n 200 -f

g 06 15:14:07 aqualink sudo[1339]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/journalctl -n 200 -f
Aug 06 15:14:07 aqualink sudo[1339]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Aug 06 15:14:11 aqualink sudo[1341]:       pi : TTY=pts/1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/systemctl start aqualinkd
Aug 06 15:14:11 aqualink sudo[1341]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)

Here's the result from sudo journalctl -u aqualinkd.service -n 200

<empty>

If you have a suggestion for log configuration, happy to run it.

originalprime, 2023-08-13:
I've moved my Pi over to USB boot and have it booting from a brand new, out of the box Samsung T7. On a fresh OS install, the Aqualinkd "simple" install method failed with all the prior gobbledygook listed in this thread.

To be sure there wasn't something else going on, I wiped the T7 and reinstalled Raspberry Pi OS and used the "apt-get" install method. This worked, or at least got me back to a working web instance. However, I am still encountering the "Connecting to control panel" error on the web GUI. As @sfeakes already stated, this seems to point to a serial communication issue.

Even though this seems odd to me given that serial logging correctly detects all of my devices and the `sudo -s eval 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0'` command correctly produces a prodigious amount of jibberish.

However, to be as thorough as I can be, I have ordered another RS-485 cable. I will try a cable swap as soon as I have the time, and I will report back on my findings.

originalprime, 2023-08-14:
Ahoy!

I can now report that a swap of the RS-485-to-USB cable did the trick. The service is operational and the data is flowing again. For the record, I did attempt to reseat the cabling of the former RS-485 adaptor in case it had come loose somehow, but that didn't work either. Replacement worked.

Despite the former cable having blinkenlights and despite plenty of data showing up, somehow it was still damaged. So be it.

It seems like what compounded the problem however, is that, at least for me, the 2.3.2 "simple" installation method is broken. That's what I was chasing after through the early part of this thread. I have had success twice now using the "apt-get" install method. I hope @sfeakes is able to figure out what's going on here.

At the end of the day I ended up upgrading unnecessarily to an external Samsung SSD, but I'm alright with that because it's favorable for the long run. And replacing the RS-485 cable wasn't the most fun I've ever had, but it wasn't the end of the world either.