aqualinkd/AqualinkD #240: AqualinkD Installation Risks to Current Home Setup

EliotKhachi, 2023-08-06:
Before I attempt this project, I would like to understand in more detail what the risk is of compromising my current Jandy Aqualink setup/configuration. I plan to connect my raspberry pi to the Aqualink control panel in my house since the WiFi signal is stronger there than by the pool. My questions/concerns are the following:

1. What happens when I disconnect the RS485 interface (the 4-pin connector) from the in-house controller's circuit board?
2. If I reconnect it, will everything be as it was before? Will the previous state persist?

Thanks!

sfeakes, 2023-08-06:
Nothing will happen disconnecting the keypad/interface and once reconnected and booted, everything will be as before.  2 of those 4 wires are power, so be careful not to short them. If you do that will damage the control panel.   Personally I disconnect the RS485 wires at the control panel before messing with them in the house, that way their area not powered.

EliotKhachi, 2023-08-06:
Awesome, that answers my question. Thank you!

EliotKhachi, 2023-08-06:
I actually have one more question.

I measured the power wire on the panel to be 10V. Can I power the raspberry pi with this? Does the RS485 cable or the micro-usb port regulate the voltage from 10V to 5V, or would I need to add a voltage regulator (MAX603) to the circuit?

sfeakes, 2023-08-07:
You can power a Raspbery PI zero (w) from the RS485 power, BUT it will need a stepdown transformer to get you to 5v. (The wiki lists one that works for the zero).  You should not “back power” a PI from the USB port since it’s not fused / regulated, and there are no USB2RS485 adapters that I know of that will backpower the Pi.
So of the 4 RS485 wires, 2 go to the adapter, and 2 go to the step down transformer.

I have not tried powering other pi’s, but I don’t think you’ll get enough amps to power anything more that an Raspbery Pi A+.     Remember booting is one thing, keeping it from undervolt and there for throttling is a bit harder.

If you do decide to try to power a larger Pi from the RS485, you should see if you can find spec’s of the RS485 side of the panel, as I am unsure of any consequence of pulling too much power from that.