aqualinkd/AqualinkD #242: MQTT setup for aqualinkd

Grand-Tee, 2023-08-13:
I’m new to mqtt and having trouble getting aqualinkd connected to Home Assistant via mqtt. I have a mqtt broker addon running on Home Assistant, and have followed the aqualinkd wiki configuration instructions, but I’m unable to get the connection to work. Do I need to install an mqtt server/broker (e.g. Mosquitto) on the RPi running aqualinkd also?

sfeakes, 2023-08-13:
No you don’t need to install mqtt server / broker on the RPI running aqualinkd. But you do need to have a MQTT server / broker running on a machine on your network.
Below link, has some details on Home Assistant and MQTT broker.
https://www.home-assistant.io/integrations/mqtt/#broker-configuration

Grand-Tee, 2023-08-13:
I do have the Mosquitto MQTT Addon installed and running in Home Assistant. It is working correctly with another MQTT device on my network. But its not connecting to the aqualinkd client. The aqualink `/var/log` file is continually outputting line pairs like the following:
```
Aug 13 16:41:17 aqualinkd aqualinkd: NetService:Starting MQTT client to localhost:1883
Aug 13 16:41:17 aqualinkd aqualinkd: NetService:MQTT Connection closed
```
I'm not able to determine why the connection is continually being closed by the aqualinkd RPi.

sfeakes, 2023-08-14:
You have MQTT set to localhost for AqualinkD.  From your post above, that doesn’t sound right. 

User and password are configured by/on the MQTT server, so only use those if Mosquitto is configured that way.  AqualinkD works with / without them.

Grand-Tee, 2023-08-14:
OK. I've resolved the issues. It was not clear in the aqualinkd wiki that the mqtt stuff refers to the MQTT *server* setup, and not the aqualinkd mqtt client. Second, at least on my system, the mqtt_address must be an IPv4 address -- "homeassistant.local:1883" does not work, even though I can ping this name from the aqualinkd RPi with no issues. Now on to straightening out my Aux_n mappings. Thanks!

closed 2023-08-14 by Grand-Tee