aqualinkd/AqualinkD #243: WebUI not rendering Watts

jbronikowski, 2023-08-14:
I noticed within 2.3.2, the watts are not showing up within the UI. When I debug the logs, i can see the following which indicates its finding the correct watts. RPM are showing and speeds are updating when changing. Thoughts?

```
Aug-13-23 20:42:31 PM Debug:   JandyDvce: ControlPanel request Pump ID 0x78 get watts
Aug-13-23 20:42:31 PM Debug:   JandyDvce: To   ePump:  Read To 0x78 of type  ePump get Watts | HEX: 0x10|0x02|0x78|0x45|0x00|0x05|0xd4|0x10|0x03|

Aug-13-23 20:42:31 PM Debug:   JandyDvce: Last panel info RPM:1800 GPM:-999 WATTS:-999
Aug-13-23 20:42:31 PM Info:    JandyDvce: Jandy Pump Status message = WATTS 346
Aug-13-23 20:42:31 PM Debug:   JandyDvce: From ePump:  Read To 0x00 of type      ePump Watts | HEX: 0x10|0x02|0x00|0x1f|0x45|0x00|0x05|0x5a|0x01|0xd6|0x10|0x03|

Aug-13-23 20:42:31 PM Debug:   JandyDvce: To   ePump:  Read To 0x78 of type   Unknown '0x43' | HEX: 0x10|0x02|0x78|0x43|0xcd|0x10|0x03|

Aug-13-23 20:42:31 PM Debug:   JandyDvce: Last panel info RPM:1800 GPM:-999 WATTS:346
Aug-13-23 20:42:31 PM Debug:   JandyDvce: From ePump:  Read To 0x00 of type  ePump (unknown) | HEX: 0x10|0x02|0x00|0x1f|0x43|0x0b|0x00|0x00|0x00|0x7f|0x10|0x03|

Aug-13-23 20:42:31 PM Debug:   JandyDvce: To   ePump:  Read To 0x78 of type   Unknown '0x46' | HEX: 0x10|0x02|0x78|0x46|0x00|0x00|0x03|0xd3|0x10|0x03|

Aug-13-23 20:42:31 PM Debug:   JandyDvce: Last panel info RPM:1800 GPM:-999 WATTS:346
Aug-13-23 20:42:31 PM Debug:   JandyDvce: From ePump:  Read To 0x00 of type   Unknown '0x20' | HEX: 0x10|0x02|0x00|0x20|0x46|0x00|0x00|0x03|0x31|0x30|0x34|0x30|0x00|0x00|0x40|0x10|0x03|
```

sfeakes, 2023-08-14:
There is a setting in the webui config.js that will set this for you.  Remember to clear web browser cache if you change that file.

sfeakes, 2023-08-14:
show_vsp_gpm=true

Use the above line in config.js.  Sorry, looks like I only addded it to the release notes, and not in the config.js file.

jbronikowski, 2023-08-14:
@sfeakes I changed 
```
      // By default all Variable Speed Pumps will show RPM.
      // this will show GPM on VSP's that you can only set GPM (ie Jandy VF pumps)
      var show_vsp_gpm=true;
 ```
[image]

I can see it the updated var within chrome however still same outcome. Only RPM not Watts

sfeakes, 2023-08-14:
Sorry, I was thinking GPM, not Watts since you said it changed in the release.     AqualinkD has never displayed watts in the webUI.

jbronikowski, 2023-08-14:
I believe if we can fix the mqtt in the discussion thread, i can pull that data from there and no need to have it in the UI. We can close this as it is not a bug at this time