aqualinkd/AqualinkD #246: systemctl start fails. Command line works.

brownjd, 2023-08-21:
> @brownjd If AqualinkD command line works, but the service doesn’t, can you post the logs / output from systemd

I downloaded both the new 2.3.2 (using aqua.sh) and also the development release. Both yielded the same results.

Here's the result from sudo journalctl -n 200 -f

g 06 15:14:07 aqualink sudo[1339]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/journalctl -n 200 -f
Aug 06 15:14:07 aqualink sudo[1339]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Aug 06 15:14:11 aqualink sudo[1341]:       pi : TTY=pts/1 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/systemctl start aqualinkd
Aug 06 15:14:11 aqualink sudo[1341]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)

Here's the result from sudo journalctl -u aqualinkd.service -n 200

<empty>

If you have a suggestion for log configuration, happy to run it.

_Originally posted by @brownjd in https://github.com/sfeakes/AqualinkD/issues/239#issuecomment-1666960813_

sfeakes, 2023-08-21:
What command line are you using?   

The below is what the service should run
`sudo /usr/local/bin/aqualinkd -c /etc/aqualinkd.conf`

That run's aqualinkd as a daemon, (ie background process) so you will get a command prompt back, but AqualinkD should be running.
You should then be able to look in /var/log/messages (or /var/log/syslog depending on config), and you should see something about AqualinkD starting or some error.

brownjd, 2023-08-21:
Yes, from the command line it runs fine. I’m talking about invoking it as systemctl start aqualinkd. That never even starts the process. It used to work fine.

Jason

sfeakes, 2023-08-21:
Can you confirm that's the exact command line you are using?

brownjd, 2023-08-22:
this does not work (and used to): sudo systemctl start aqualinkd
this command works: sudo aqualinkd -d -c /etc/aqualinkd.conf
 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
Model		: Raspberry Pi Model B Plus Rev 1.2

brownjd, 2023-08-22:
I kept at it and found this by enabling debug for systemctl:

 aqualinkd.service - Aqualink RS daemon
     Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

Aug 21 20:50:13 aqualink systemd[1]: aqualinkd.service: Reinstalled deserialized job aqualinkd.service/start as 1906
Aug 21 20:50:18 aqualink systemd[1]: aqualinkd.service: starting held back, waiting for: multi-user.target
Aug 21 20:51:25 aqualink systemd[1]: aqualinkd.service: Trying to enqueue job aqualinkd.service/start/replace
Aug 21 20:51:25 aqualink systemd[1]: aqualinkd.service: Merged aqualinkd.service/start into installed job aqualinkd.service/start as 1906
Aug 21 20:51:25 aqualink systemd[1]: aqualinkd.service: Enqueued job aqualinkd.service/start as 1906
Aug 21 20:51:25 aqualink systemd[1]: aqualinkd.service: starting held back, waiting for: multi-user.target

sfeakes, 2023-08-22:
‘`aqualinkd.service: starting held back, waiting for: multi-user.target`’

You’re not booted in multi user mode.

brownjd, 2023-08-22:
The issue is a bug with running a raspberrypi in headless mode with bullseye and ssh keys only. It will prompt you to create a user. That prevents the multi-user target from firing, holding aqualinkd back.

https://github.com/RPi-Distro/userconf-pi/issues/5

This solved the problem:

systemctl disable userconfig

I must have update the raspberrypi at the same time that I updated the aqualinkd. Sorry for the omission - that would be a pretty big change to the environment!

closed 2023-08-22 by brownjd