aqualinkd/AqualinkD #247: Pool Light Dimmer Web UI Confusing

brownjd, 2023-08-22:
Using lightMode=6 and a dimmer requires an extra step to turn on a light. You have to long press the tile to get the popup. This presents a user with set of dim settings and a runtime slider below that defaults to 0 minutes. 

The light tile/button won't work in the default state unless the user manually moves the slider beyond 0 seconds. This is suboptimal from a UI perspective. If you just want to turn on the light to 50%, you are forced to also pick a duration. 

This also doesn't translate well if you are using home bridge to link to apple HomeKit. Switches don't normally have their own timers. You use automation for that.

sfeakes, 2023-08-25:
I've been away for a few weeks and was not able to test this.  But I just been able to and can't replicate the problem.  Can you please post your `/etc/aqualinkd.conf` file and browser version you are using.

Short press turns light on or off.
Long press brings up options where :-

- Toggle on/off works to turn on / off
- Select % will turn on and select %
- Timer will set timer.
In any of the above, you do need to select "close" before AqualinkD will program the light to any settings you have selected.

ie, You do NOT need to select a duration to turn light on. 

Homekit does not support associated timers on lights, so this functionality only works in WebUI / MQTT / Automation hub.

brownjd, 2023-08-25:
Sure, I will test it this evening. WRT to your comment about HomeKit. I don’t think it needs to support the timer. If you could just support the dimmer and leave the timer set to infinity, then it does not need to be exposed. Simply allowing the use to turn on the light and set the dim level is enough through HomeKit.

Jason

brownjd, 2023-08-29:
[aqualinkd.conf.txt](https://github.com/sfeakes/AqualinkD/files/12459860/aqualinkd.conf.txt)

Attached aqualinkd.conf. I had to rename the extension to txt to upload it. 

Using Safari Version 16.6 (18615.3.12.11.2). I also tested with Chrome Version 116.0.5845.110 (Official Build) (x86_64).

Short click causes the Pool Light tile to light up momentarily and then after 2 seconds it turns back off. The lights never come on.

Long press brings up the selection dialog. Selecting a percentage causes the on/off to toggle and the light turns on at the appropriate dim setting. You don't have to move the timer past zero (I was incorrect in my original post). 

If you then try a short press when it is off, it will revert to the original behavior. Tile turns on, no lights, then tile turns off.

Based on the fact that the timer is not required to be > 0 with long press means that HomeKit could be supported without timer functionality. If commands came through the home bridge plugin, then timer would always just be zero and it would work as expected.

brownjd, 2025-04-13:
Did this workaround not fit the bill?

sfeakes, 2025-04-13:
Light mode 10 will limit to 0/25/50/75/100 (ie 25% increments)
Light mode 11 will allow full range (1% increments), and is a lot better for things like HA or Homebridge.  This mode is dependent on having a panel that supports RS Serial Connection, ie `rssa_device_id=0x48` in aqualinkd.conf

brownjd, 2025-04-13:
What if you exposed 1% increments in Lightmode 10 in Homekit, but they snapped to the closest increment? 

Pick 5% and it goes to 20. Pick 25, and it goes to 50, etc?

sfeakes, 2025-04-13:
That works ok for small increments like 5% (SWG will do this). But doesn't work well in HomeKit for large increments like 25%. It's a limitation of Apple's HomeKit app since it doesn't support resolution.  
I will test other options in the future, but if you use light mode 11 you'll simply get full range. The Jandy Dimmable light seems to support full range, they just don't expose that through their devices.  You can only set 25,50,75,100% in Jandy's iAqualink app, but if AqualinkD sets 62%, iAqualink will display the light is 62%.

brownjd, 2025-04-13:
You are correct. Lightmode=11 works like it should. Thanks for implementing this. It is nice to be able to control the dimmable non-colored lights through Home.app. 

I'm really impressed you figured this one out without the hardware. Thanks again.

closed 2025-04-13 by brownjd

sfeakes, 2025-04-13:
> I'm really impressed you figured this one out without the hardware. Thanks again.

I've got enough info on their protocol that I can usually write some simulator to trick the panel into thinking it's got different devices attached, and a few panels people have donated over the years to help decode this stuff.  It just a time suck.