aqualinkd/AqualinkD #262: VSP control for Aqualinkd

parchmaf, 2023-12-31:
I know that you have had VSP support listed as a to-do for a future version.  Any idea when this will be available?  It's the only component of my pool that cannot be controlled by Aqualinkd and once I have it, I won't have to use the Aqualink app for control, just occasional program changes.  I am currently using your latest version and it works great.  It's much faster and more responsive than the aqualink app.  It also is much more responsive and reliable with Homekit.

Also, I wanted to share a slight modification for the solution you have posted using pi zero.  In my case, my Aqualink controller is very far from the house and wifi is weak.  Also, there is no good way to get an ethernet cable to the controller.  I got around this by putting the pi zero and serial device inside the Aqualink panel inside my home.  I used a power voltage adapter to get the correct power for the pi zero.  It was a little tight, but it does fit and now works great because wifi inside my home is strong.

This is a great solution.  Thank you!

sfeakes, 2024-01-01:
VSP support was added in version 1.3.2c, and improvements in releases since then.  Current version is 2.3.2.  If you upgrade VSP will work, but you'll need to modify your config.

closed 2024-01-01 by sfeakes

parchmaf, 2024-01-02:
To be clear, when I use the web interface, the pump button shows the current VSP speed but there is no ability to change the speed.  In HomeKit, I do not see the VSP at all.  Pie in the sky, it would be really nice to be able to view and set VSP in HomeKit.  Is this possible with your current version.  I’m attaching my AqualinkD.conf for reference inn case you have any suggestions.  Thanks in advance for your help.

￼
Fred Parchman
(973) 886-1200 (m)
***@***.***

> 
> 
> Closed #262 <https://github.com/sfeakes/AqualinkD/issues/262> as completed.

# aqualinkd.conf
#

# The directory where the web files are stored

web_directory=/var/www/aqualinkd/

# Log to file, comment out if you do not want to log to file
#log_file=/var/log/aqualinkd.log

# The log level. [DEBUG_DERIAL, DEBUG, INFO, NOTICE, WARNING, ERROR]
# Pick the highest level, and all levels below will be sent to syslog.
# your syslog settings may be set to only display messages above a certian level
# in which case make sure you use the log_file settings to capture everything
# you want when debugging
# so, NOTICE also prints WARNING & ERROR
#     DEBUG_SERIAL would print everything possible

#log_level=DEBUG
#log_level=INFO
log_level=NOTICE
#log_level=WARNING

# Display any ERROR & Warning messages in web interface. 
display_warnings_in_web=true

# The socket port that the daemon listens to
# If you change this from 80, remember to update aqualink.service.avahi
socket_port=80

# The serial port the daemon access to read the Aqualink RS8
serial_port=/dev/ttyUSB0

# Your RS panel size. ie 4, 6, 8, 12 or 16 relates to RS4, RS6, RS8, RS12 or RS16.
# VERY important that you select 12 or 16, if you have either of those size panels.
# Also don't think setting a 12 when you have a 8 will give you 4 more accessories to control, it won't the
# panel information is needed as different panels use different bits within the RS protocol for status and key
# presses.
# serial_logger will get the panel type string if you don't know it, below are examples.
# Must be in format `XX-N ????`  (XX=RS or PD, N=Circuits, ????=Combo or Only or Dual)
panel_type = RS-8 Combo
#panel_type = PD-8 Combo
#panel_type = RS-16 Combo
#panel_type = RS-2/14 Dual
#panel_type = RS-4 Combo
#panel_type = RS-8 Only
#
# If serial_logger doesn't give you a type string in the format above, you can use the next options to set the specifics.
# 
# panel_type_size = (6, 8, 10, 12, 14 or 16)  (Number of supported accessories / buttons)
# panel_type_combo = (yes or no)              (combo panels support BOTH pool & spa)
# panel_type_dual = (yes or no)               (dual circuit panel)
# panel_type_pda = (yes or no)                (PDA panel. only set this if you have to. Panel ONLY supports the PDA protocol)
# panel_type_rs = (yes or no)                 (RS panel. Panel Supports all protocols)

# The ID of the Aqualink terminal device. Devices probed by RS8 master are:
# 08-0b, 10-13, 18-1b, 20-23, 28-2b, 30-33, 38-3b, 40-43
# Working RS ID's are 0x0a 0x0b 0x09 0x08    <- 0x08 is usually taken
# If your panel is a PDA only model, then PDA device ID's are 0x60, 0x61, 0x62, 0x63.
[... 192 lines cut ...]
button_01_pumpIndex=1

button_02_label=Spa Mode

button_03_label=Cleaner

button_04_label=Waterfall

button_05_label=Spa Blower

button_06_label=Pool Light
#button_05_lightMode=0

button_07_label=Spa Light
#button_05_lightMode=0

button_08_label=NONE

button_09_label=NONE

button_10_label=Pool Heater

button_11_label=Spa Heater

button_12_label=NONE

sfeakes, 2024-01-03:
in the webui, what happens when you do a long press on the icon showing the pump RPM/GPM?  It should bring up a dialog where you can change the RPM/GPM.
You can't change RPM/GPM from HomeKit as that's not supported, there is no nice way to do that with HomeKit. Closest device that's supported would be a dimmable light, but then you could only change by a % not by RPM/GPM.