aqualinkd/AqualinkD #264: No probe on 0x48 (rssa_device_id)

EliotKhachi, 2024-02-18:
Hi,

The Serial Logger is missing my Jandy Control Panel Model and Version. I can figure that `panel_type=RS-8 Combo` based on the fact that RS-8 is written on the panel and I have a pool *and* spa. However, I feel that the serial bugger is missing some critical information I need, specifically regarding the `rssa_device_id`. Currently, I'm setting `rssa_device_id=0x48`.

## sudo aqualinkd -d -c /etc/aqualinkd.conf
`Error:   AqualinkD: No probe on '0x48', giving up! (please check config)`

## sudo ./release/serial_logger /dev/ttyUSB0
```
AqualinkD serial_logger V2.0
Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Notice:  RS Serial: Logging serial information!
Please wait.|

Notice:  RS Serial: RS485 interface received 600 packets in 24 seconds (~25.00 Msg/Sec)
Notice:  RS Serial: Jandy Control Panel Model   : 
Notice:  RS Serial: Jandy Control Panel Version : 
Notice:  RS Serial: Jandy ID's found
Notice:  RS Serial: ID 0x09 is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x31 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x41 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x08 is in use    <-- RS Keypad
Notice:  RS Serial: ID 0x42 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x43 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: ID 0x0a is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x0b is not used  <-- can use for Aqualinkd
Notice:  RS Serial: ID 0x30 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x32 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x33 is not used  <-- can use for Aqualinkd (Prefered Extended Device ID)
Notice:  RS Serial: ID 0x40 is not used  <-- can use for Aqualinkd (Extended Device ID)
Notice:  RS Serial: 

Notice:  RS Serial: Suggested aqualinkd.conf values
Notice:  RS Serial: -------------------------
Notice:  RS Serial: device_id = 0x09
Notice:  RS Serial: extended_device_id = 0x31
Notice:  RS Serial: -------------------------
```

Eliot

EliotKhachi, 2024-02-18:
Commenting out rssa_device_id in the configuration, I now get stuck at this error message: `Error:   Panl Prog: Did not receive expected reply from last RS SEND command, resending '0x1d'`
## $ sudo aqualinkd -d -c /etc/aqualinkd.conf
```
Notice:  AqualinkD: Aqualink Daemon v2.3.2
Notice:  AqualinkD: Panel set to RS-8 Combo Pool/Spa
Notice:  AqualinkD: Config log_level         = 5
Notice:  AqualinkD: Config device_id         = 0x09
Notice:  AqualinkD: Config rssa_device_id    = 0x00
Notice:  AqualinkD: Config extra_device_id   = 0x31
Notice:  AqualinkD: Config extra_device_prog = NO
Notice:  AqualinkD: Config serial_port       = /dev/ttyUSB0
Notice:  AqualinkD: Config socket_port       = 80
Notice:  AqualinkD: Config web_directory     = /var/www/aqualinkd/
Notice:  AqualinkD: Config use_aux_labels    = NO
Notice:  AqualinkD: Config override frz prot = NO
Notice:  AqualinkD: Config mqtt_server       = (null)
Notice:  AqualinkD: Config mqtt_dz_sub_topic = (null)
Notice:  AqualinkD: Config mqtt_dz_pub_topic = (null)
Notice:  AqualinkD: Config mqtt_aq_topic     = (null)
Notice:  AqualinkD: Config mqtt_user         = (null)
Notice:  AqualinkD: Config mqtt_passwd       = (null)
Notice:  AqualinkD: Config mqtt_ID           = aqualinkd_b827eb0f
Notice:  AqualinkD: Config force SWG         = NO
Notice:  AqualinkD: Config force PS setpoint = NO
Notice:  AqualinkD: Config force Freeze Prot = NO
Notice:  AqualinkD: Config deamonize         = NO
Notice:  AqualinkD: Config log_file          = (null)
Notice:  AqualinkD: Config enable scheduler  = YES
Notice:  AqualinkD: Config light_pgm_mode    = 0.00
Notice:  AqualinkD: Debug RS485 protocol     = NO
Notice:  AqualinkD: Debug RS485 protocol raw = NO
Notice:  AqualinkD: Display warnings in web  = NO
Notice:  AqualinkD: Keep panel time in sync  = YES
Notice:  AqualinkD: Read SWG direct          = YES
Notice:  AqualinkD: Read ePump direct        = YES
Notice:  AqualinkD: Read vsfPump direct      = YES
Notice:  AqualinkD: Serial FTDI low latency  = YES
Notice:  AqualinkD: Config BTN Filter_Pump   = label Filter Pump     | 
Notice:  AqualinkD: Config BTN Spa_Mode      = label Spa Mode        | 
Notice:  AqualinkD: Config BTN Aux_1         = label Cleaner         | 
Notice:  AqualinkD: Config BTN Aux_2         = label Waterfall       | 
Notice:  AqualinkD: Config BTN Aux_3         = label Spa Blower      | 
Notice:  AqualinkD: Config BTN Aux_4         = label Pool Light      | 
Notice:  AqualinkD: Config BTN Aux_5         = label Spa Light       | 
Notice:  AqualinkD: Config BTN Aux_6         = label NONE            | 
Notice:  AqualinkD: Config BTN Aux_7         = label NONE            | 
Notice:  AqualinkD: Config BTN Pool_Heater   = label Pool Heater     | 
Notice:  AqualinkD: Config BTN Spa_Heater    = label Spa Heater      | 
Notice:  AqualinkD: Config BTN Solar_Heater  = label Solar Heater    | 
Notice:  NetService:Starting network services thread
Notice:  RS Serial: Port /dev/ttyUSB0 low latency mode is set
Notice:  NetService:Starting web server on port 80
Notice:  AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Notice:  AqualinkD: Waiting for Control Panel probe
Notice:  NetService:Starting MQTT client to (null)
Notice:  AqualinkD: Got message but no probe on '0x09', did we start too soon? (waiting for probe)
Notice:  AqualinkD: Got probe on '0x31' Extended Protocol
Notice:  AqualinkD: Got probe on '0x09' Standard Protocol
Notice:  AqualinkD: Starting communication with Control Panel
Notice:  RS Allbtn: Control Panel version 6520 REV HH
Notice:  RS Allbtn: Control Panel revision HH
Notice:  Panl Prog: Starting programming thread 'Get Heater setpoints'
Notice:  Panl Prog: Starting programming thread 'Get Freeze protect'
Error:   Panl Prog: Did not receive expected reply from last RS SEND command, resending '0x1d'
```
Starting the service anyway yields
## $ sudo systemctl start aqualinkd
## $ sudo systemctl status aqualinkd

```
● aqualinkd.service - Aqualink RS daemon
     Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2024-02-17 21:34:22 PST; 3s ago
    Process: 5496 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
   Main PID: 5497 (aqualinkd)
      Tasks: 2 (limit: 414)
        CPU: 74ms
     CGroup: /system.slice/aqualinkd.service
             └─5497 /usr/local/bin/aqualinkd -c /etc/aqualinkd.conf

Feb 17 21:34:22 pizero aqualinkd[5496]: AqualinkD: Config BTN Pool_Heater   = label Pool Heater     |
Feb 17 21:34:22 pizero aqualinkd[5496]: AqualinkD: Config BTN Spa_Heater    = label Spa Heater      |
Feb 17 21:34:22 pizero aqualinkd[5496]: AqualinkD: Config BTN Solar_Heater  = label Solar Heater    |
Feb 17 21:34:22 pizero aqualinkd[5497]: NetService:Starting network services thread
Feb 17 21:34:22 pizero aqualinkd[5497]: NetService:Starting web server on port 80
Feb 17 21:34:22 pizero aqualinkd[5497]: NetService:Starting MQTT client to (null)
Feb 17 21:34:22 pizero aqualinkd[5497]: RS Serial: Port /dev/ttyUSB0 low latency mode is set
Feb 17 21:34:22 pizero aqualinkd[5497]: AqualinkD: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Feb 17 21:34:22 pizero aqualinkd[5497]: AqualinkD: Waiting for Control Panel probe
Feb 17 21:34:22 pizero aqualinkd[5497]: AqualinkD: Got message but no probe on '0x09', did we start too soon? (waiting for probe)

```

Also, I'm not sure how the AqualinkD web app is being served so I'm not sure where to look or what IP address to `curl` to see if it's running. The wiki says to use http://aqualink.ip.address/ or http://aqualink.ip.address/simple.html to access the web UI, but doesn't actually provide an ip address. Do I set it somewhere?

sfeakes, 2024-02-18:
> Commenting out rssa_device_id in the configuration, I now get stuck at this error message: `Error: Panl Prog: Did not receive expected reply from last RS SEND command, resending '0x1d'`

I don't think it's getting stuck there, that's just an error that happened so it was printed to the log.  There are probably no other "notice" messages to be printed.  You can change the log level to debug to check by editing the config file.

> 
> I can access the web app, but the controls seem misconfigured. Turning on the pool or spa lights does not do anything but instead it triggers some noise. Perhaps it's the pump priming.

It sounds like you name not configured the keys/buttons correctly.  In the wiki, lookout the section on confgiuring buttons
https://github.com/sfeakes/AqualinkD/wiki#Configuration

chrisdaun, 2024-02-18:
Hey @EliotKhachi ,

I recently setup AqualinkD and I had a similar issue where the serial logger would return blank for the make and model of my control panel.

Mine is an RS-6 combo and I configured it as such in the aqualinkd.config and all seemed to work after I got my uart to rs485 serial board wired up.

EliotKhachi, 2024-02-18:
Thanks for the feedback guys. I played around with the buttons and set the labels myself in the config. Not sure if I just have a weird RS-8 Combo panel or none at all.

For reference, here is my panel and config:
[image]
```
# Below are settings for my actual panel
button_01_label=Filter Pump                                                                                                                                            
button_02_label=Spa Mode
button_03_label=Cleaner                                     
button_04_label=Pool Light                                       
button_05_label=Spa Light                                         
button_06_label=Waterfall                                                          
#button_05_lightMode=0                                         
button_07_label=Jet Pump                                                           
#button_05_lightMode=0                                        
button_08_label=Not Used 1                                     
button_09_label=Not Used 2                                      
button_10_label=Pool Heater                                                        
button_11_label=Spa Heater                                                         
button_12_label=Solar Heater   
```

closed 2024-02-18 by EliotKhachi