aqualinkd/AqualinkD #269: Getting wrong Temperature via MQTT  but ok in web interface

Bigfellas, 2024-03-06:
AS the subject says. I am getting the correct display in the we interface 27C but via MQTT I am getting -2.78 I can't see any seetiing that  may be effecting this. 

```
# aqualinkd.conf
#

# The directory where the web files are stored

web_directory=/var/www/aqualinkd/

# Log to file, comment out if you do not want to log to file
#log_file=/var/log/aqualinkd.log

# The log level. [DEBUG_DERIAL, DEBUG, INFO, NOTICE, WARNING, ERROR]
# Pick the highest level, and all levels below will be sent to syslog.
# your syslog settings may be set to only display messages above a certian level
# in which case make sure you use the log_file settings to capture everything
# you want when debugging
# so, NOTICE also prints WARNING & ERROR
#     DEBUG_SERIAL would print everything possible

log_level=DEBUG
#log_level=INFO
#log_level=NOTICE
#log_level=WARNING

# Display any ERROR & Warning messages in web interface. 
display_warnings_in_web=true

# The socket port that the daemon listens to
# If you change this from 80, remember to update aqualink.service.avahi
socket_port=80

# The serial port the daemon access to read the Aqualink RS8
serial_port=/dev/ttyUSB0

# Your RS panel size. ie 4, 6, 8, 12 or 16 relates to RS4, RS6, RS8, RS12 or RS16.
# VERY important that you select 12 or 16, if you have either of those size panels.
# Also don't think setting a 12 when you have a 8 will give you 4 more accessories to control, it won't the
# panel information is needed as different panels use different bits within the RS protocol for status and key
# presses.
# serial_logger will get the panel type string if you don't know it, below are examples.
# Must be in format `XX-N ????`  (XX=RS or PD, N=Circuits, ????=Combo or Only or Dual)
#panel_type = RS-8 Combo
#panel_type = PD-8 Combo
#panel_type = RS-16 Combo
#panel_type = RS-2/14 Dual
#panel_type = RS-4 Combo
#panel_type = RS-8 Only
panel_type = PD-P8 Only
#
# If serial_logger doesn't give you a type string in the format above, you can use the next options to set the specifics.
# 
# panel_type_size = (6, 8, 10, 12, 14 or 16)  (Number of supported accessories / buttons)
# panel_type_combo = (yes or no)              (combo panels support BOTH pool & spa)
# panel_type_dual = (yes or no)               (dual circuit panel)
# panel_type_pda = (yes or no)                (PDA panel. only set this if you have to. Panel ONLY supports the PDA protocol)
# panel_type_rs = (yes or no)                 (RS panel. Panel Supports all protocols)

# The ID of the Aqualink terminal device. Devices probed by RS8 master are:
# 08-0b, 10-13, 18-1b, 20-23, 28-2b, 30-33, 38-3b, 40-43
# Working RS ID's are 0x0a 0x0b 0x09 0x08    <- 0x08 is usually taken
# If your panel is a PDA only model, then PDA device ID's are 0x60, 0x61, 0x62, 0x63.
# (These are NOT recomended to use unless you absolutly have no other option)
device_id=0x63

# The ID of Jandy SerialInterface device. These is only one usable ID, if serial_logger
# picks up your panel supports this, uncomment, as it will speed up heater setpoints & RS16 panels.
#rssa_device_id=0x48

[... 192 lines cut ...]
button_03_PDA_label=TEMP2

button_04_label=Solar Heater
button_04_PDA_label=SOLAR HEAT

button_05_label=Cleaner
button_05_PDA_label=CLEANER

#button_06_label=Aux  2
#button_05_lightMode=0

#button_07_label=Aux 3
#button_05_lightMode=0

#button_08_label=NONE

#button_09_label=NONE

#button_10_label=Pool Heater

#button_11_label=Spa Heater

#button_12_label=Solar Heater

```

sfeakes, 2024-03-06:
convert_mqtt_temp_to_c = yesThe above converts deg f to dec c

Bigfellas, 2024-03-06:
I'm a fool. I think last time I just commented this out. But I changed the value to no and that worked. Thanks.

closed 2024-03-06 by Bigfellas