aqualinkd/AqualinkD #274: Serial Packet Too Large - Read Bad Packet to 0x33

parchmaf, 2024-04-21:
[image]

Installed 2.3.3 and it seems to be working ok.  Any idea what this warning message is about or how to fix or supress?  TY!

sfeakes, 2024-04-22:
you can suppress it by setting the log_level in aqualinkd.conf, but that's a serial error and probably indicative of something more serious.  Can you please post the full log?

parchmaf, 2024-04-22:
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/15063532/aqualinkd.log)

sfeakes, 2024-04-22:
Thanks for the log, everything looks good in what you posted.  But it didn’t have the Warnings you posted in the screen shot.   I’d really like to see the complete warning with the hex/bytes to figure out what’s going on.  (It could just be a message I have not come across before, or a bad RS connection, until I see the complete message I won’t know for sure).
The message is being sent to an iAqalinkTouch device ID, so I assume you have one of those connected?

parchmaf, 2024-04-22:
I used the management console and changed the debug level, 10000 lines, and let it run longer before downloading the log.  Is this better?  I appreciate the help.
[aqualinkd-2.log](https://github.com/sfeakes/AqualinkD/files/15064082/aqualinkd-2.log)

closed 2025-11-19 by github-actions[bot]